microsoft/Qcodes #1782: Bug in DelegateParameter: get_latest and set_cache are not forwarded to source parameter

astafan8, 2019-10-23:
See the following tests of `DelegateParameter` that are marked as `xfail`: https://github.com/QCoDeS/Qcodes/pull/1757/commits/b04266b94ff362d0829cdbd8d3215ca557009e2d

Expectation: `DelegateParameter`s `get_latest` and `set_cache` should in principle simply forward the call to `get_latest` and `set_cache` of the source parameter.

Dominik-Vogel, 2019-11-08:
I think it is more involved than that. At the example of the `scale` of a parameter, consider the following:
[image]
[2:31 PM] Dominik Vogel (Adecco)
It seems that set_cached needs calculate the raw value, and forward that to set_cached of the original parameter.
`get_latest` of the delegate parameter should convert its raw value, which is received by calling into get_latest of the source, to the non-raw value.
I will need to look into #1757 again to refresh my knowledge of the new caching mechanism. Then I can implement the cache for the `DelegateParameter` next week.

Dominik-Vogel, 2019-11-13:
Plan of actions agreed upon with @astafan8 
1) Merge PR #1791, removing all uses of self._latetst ouside the actual getters/setters
2) Repair Delegate Parameter
3) Create new PR to move ._latest into seperate class to make clear that the scope of the private attribute is only for getters and setters

PR #1827 (2019-11-22): Implement cache for DelegateParameter to mirror cache of the source parameter
This PR implements cache of `DelegateParameter` to exactly mirror the cache of the source parameter. Depends on #1832.

For now, some dirty stuff with respect to accessing `_value` of cache is needed, but this wouldn't be necessary when `snapshot_base` improvements will be worked on in #1833.

Fixes #1782.
closed 2019-11-22 by Dominik-Vogel