microsoft/Qcodes #1882: Improve implementation of instrument instance registry

astafan8, 2019-12-19:
This issue flags the importance of improving how `Instrument._all_instruments` and `Instrument._instances` are implemented. 

There are cases where closing instrument is important not to forget, otherwise some warning like the ones mentioned in this PR https://github.com/QCoDeS/Qcodes/pull/1875 can start showing up for no good reason.

Anyway, `Insrtument._all_instruments` is intended to hold weak references to instrument instances in order to prevent multiple connections to the same hardware instrument. It is now implemented via simple `dict` which requires methods like `Instrument.exists` and `Instrument.find_instance` to account for the fact that some of those weak references can be `None` because instruments may have gotten closed in the meantime. And this seems to be causing weird unnecessary bugs and dirty code.

My suggestion is to consider using the right tools for the job - use `weakref.WeakValueDictionary` for `_all_instruments` which will automatically take care of the case where a weakref is "dead". For a usage example see `DllWrapper` for `Alazar` driver https://github.com/QCoDeS/Qcodes/blob/master/qcodes/instrument_drivers/AlazarTech/dll_wrapper.py#L95.

jenshnielsen, 2019-12-19:
This sounds good. For me the main userfaceing bug to solve here would the fact that an instrument is registered even when the setup fails e.g.

```python
inst = MyInstrument(name='foo', totally_the_wrong_args)
# now the name foo is reserved even if the instrument never got constructed
# and this line below fails
inst = MyInstrument(name='foo', the_right_args)
```

astafan8, 2019-12-19:
this is an important bug, let's see if this weakdict stuff fixes it. Otherwise I'll make a separate issue for this bug.

jenshnielsen, 2021-12-17:
Another related potential improvement would be to use [weakref.finalize](https://docs.python.org/3/library/weakref.html#weakref.finalize) rather than `__del__` to close an instrument when it goes out of scope

jenshnielsen, 2022-01-03:
This will be fully fixed by https://github.com/QCoDeS/Qcodes/pull/3699

commit a00867dc (2022-01-03): Merge #3696

3696: Only record instrument instances when successfully created r=jenshnielsen a=spauka

This commit fixes the issue of an instrument instance being registered in the instance dictionary even if the instrument creation failed. This will allow users to retry instrument connection when the first attempt fails without having to restart the interpreter or modify private variables in the Instrument class.

It certainly resolves a source of frustration when trying to debug failing instrument connections 😆 

This partially closes #1882, and should also allow us to be smarter about loggi [... cut]
PR #3696 (2022-01-03): Only record instrument instances when successfully created
This commit fixes the issue of an instrument instance being registered in the instance dictionary even if the instrument creation failed. This will allow users to retry instrument connection when the first attempt fails without having to restart the interpreter or modify private variables in the Instrument class.

It certainly resolves a source of frustration when trying to debug failing instrument connections 😆 

This partially closes #1882, and should also allow us to be smarter about logging successful connections or cleaning up gra [... cut]
closed 2022-01-04 by jenshnielsen