microsoft/Qcodes #1909: Aliasing parameters in station config file behaves differently since v0.9.0

Jajiko, 2020-01-28:
We have set aliases for *channel* parameters in a station config file. Before v0.9.0, the aliases were assigned to the main instrument, now they are assigned to the specific channel instead. 

In the example below, the alias `temp_alias` becomes an attribute of the channel `A`, not `DummyChannelInstrument` as before.

I don't know which behavior is the intended one, but I am reporting it since it breaks code (at least it broke ours) and it wasn't documented in the changelog. Maybe this was the intention of commit de8f90e04886b2871613856d881cfeab04b02faa ?  @Dominik-Vogel ?

### Steps to reproduce
1. Write the following in a station configuration file:
```
instruments:
    dummy:
        driver: qcodes.tests.instrument_mocks
        type: DummyChannelInstrument
        parameters:
            A.temperature:
                alias: temp_alias
                initial_value: 273.15
 ```
2. run
```
d = station.load_dummy()
print("d has temp_alias:",  hasattr(d, "temp_alias"))
print("d.A has temp_alias:",  hasattr(d.A, "temp_alias"))
```

### Output before v0.9.0
```
d has temp_alias: True
d.A has temp_alias: False
```
### Output after v0.9.0
```
d has temp_alias: False
d.A has temp_alias: True
```
### System
Qcodes master @ 00ec65352f2bd71cce607274a64c334c40b51142

fblanchetNaN, 2022-01-16:
Hi. No news about that? Is this behavior expected?

mgunyho, 2023-05-02:
Hi, I came across this as well. For reference, here's a MWE code snippet tested with qcodes 0.38.1:

```python
import qcodes as qc

station = qc.Station()
station.load_config("""
instruments:
    dummy:
        type: qcodes.tests.instrument_mocks.DummyChannelInstrument
        parameters:
            A.temperature:
                alias: temp_alias
                initial_value: 273.15
""")

d = station.load_dummy()

print(d.A.temp_alias())  # works
print(d.temp_alias())  # doesn't work

```

The example Station configuration in [the documentation](https://qcodes.github.io/Qcodes/examples/Station.html#Example-of-YAML-Station-configuration) has this:
```yaml
...
    parameters:
      # Each parameter is specified by its name from the
      # instrument driver class.
      # Note that "dot: notation can be used to specify
      # parameters in (sub)channels and submodules.
      ch01.v:
        # If an alias is specified, the paramater becomes
        # accessible under another name, so that you can write
        # `qdac.cutter_gate(0.2)` instead of `qdac.ch01.v(0.2)`.
        # Note that the parameter instance does not get copied,
        # so that `(qdac.ch01.v is qdac.cutter_gate) == True`.
        alias: cutter_gate
```

So the documentation is inconsistent with the behavior. Which one is intended? (I personally would like to have the behavior from pre v0.9 / the one that is shown in the documentation, so that I can give semantic names to the submodule parameters and don't have to think about the specific channel numbers.)

Can e.g. @astafan8 or @jenshnielsen comment?

astafan8, 2023-05-03:
sorry for not responding earlier. I think i finally udnerstand the issue:

you'd like to use `alias` to create an alias to a parameter on the root instrument: e.g. `qdac.ch01.v` --> `qdac.cutter_gate`. And apparently that's how it worked in v0.9 (that's a looong time ago :) ), and now it's not possible because the alias is created at the same level as the original parameter e.g. `qdac.ch01.v` --> `qdac.ch01.cutter_gate`.

first of all, it's very unfortunate that this behavior changed without noticing and thank you for filing the issue back then!

now, i don't think it's fair to change the behavior again, it's better to fix the documentation and keep `alias` for aliasing on the same level, and add another option called `alias_at_root` (or with a better name) which would create an alias at the root instrument. how do you like this approach? @mgunyho @Jajiko @fblanchetNaN

mgunyho, 2023-05-03:
Yes, exactly. We have a device with many channels, and we would like to name certain parameters from those channels, like the `qdac.ch01.v` -> `qdac.cutter_gate` example. A solution like the `alias_at_root` could be good.

astafan8, 2023-05-03:
awesome!

would anyone be willing to start a PR? :)