microsoft/Qcodes #1993: Invalid syntax in qcodes file

Danielgb23, 2020-05-01:
I tried running qcodes on the jupyter notebook for the 15 minutes intro and it ran into a problem where it couldn't even import the code:

`Traceback (most recent call last):

  File "/usr/local/lib/python3.5/dist-packages/IPython/core/interactiveshell.py", line 3326, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)

  File "<ipython-input-5-2e79a95c9d0e>", line 7, in <module>
    import qcodes as qc

  File "/usr/local/lib/python3.5/dist-packages/qcodes/__init__.py", line 12
    config: qcconfig.Config = qcconfig.Config()
          ^
SyntaxError: invalid syntax
`
Since my OS is ubuntu 16.04 I had to downgrade panda to 0.24 so it worked along python 3.5.2

**qcodes branch**
Last stable version

jenshnielsen, 2020-05-04:
@Danielgb23 We do not support python 3.5. Earliest supported python version is 3.6, If you want to run on an old ubuntu installation I suggest that you install a newer version of python using either anaconda or pyenv

closed 2020-05-04 by jenshnielsen

Danielgb23, 2020-05-05:
I installed python 3.6 and it worked. I just had to reinstall some thing for it though.