microsoft/Qcodes #2003: Yokogawa GS210 output mode cache always set to 'VOLT' upon initializing

abargerbos, 2020-05-14:
### Steps to reproduce
1. load Yokogawa GS210 instrument  
2. set yoko.source_mode('CURR')
3. close connection, reload instrument
4. try calling yoko.current_range()

### Expected behaviour
What should happen is that the current range is returned.

### Actual behaviour
What happens instead is a value error:
('Cannot get/set CURR settings while in VOLT mode', 'getting yoko_current_range')
This is because the driver checks the mode of the command (a command like current_range can only be used in the 'CURR' mode) not against the actual source mode (gotten from yoko.source_mode()) but against a cached value yoko._cached_mode. However, in the init yoko._cached_mode is always set to 'VOLT', even when yoko.source_mode is 'CURR'. 

This is probably fixed by changing line 198 in the driver to self._cached_mode = self.source_mode(), but I was not 100% sure that this is the desired solution? 

### System
**operating system**
Windows 10
**qcodes version**
0.13.0 (pip installed)

astafan8, 2020-05-15:
@abargerbos i think indeed the fix that you are suggesting is a step in the right direction - could you submit a PR and then we can take it from there?

WilliamHPNielsen, 2020-09-04:
@Akshita07 or @FarBo, this might be an issue for you to look into?

FarBo, 2020-09-04:
> @Akshita07 or @FarBo, this might be an issue for you to look into?

Definitely.

FarBo, 2020-09-07:
@abargerbos 
I am working on some fixes. I will test those fixes and make a PR. But, the fix is something like what you proposed.

FarBo, 2020-09-08:
@abargerbos
Have you seen this notebook:
https://qcodes.github.io/Qcodes/examples/driver_examples/Qcodes%20example%20with%20Yokogawa%20GS2xx.html?highlight=source_mode
With the current driver, you should be able to switch between modes as shown in the notebook. Could you try to follow the notebook to see if you can successfully switch to the current mode and run yoko.current_range() without any issue?
The driver can be improved, but I would like to know if the current driver is capable of switching between modes.

closed 2020-09-23 by jenshnielsen