microsoft/Qcodes #2071: OperationalError: locking protocol on trying to access a database which is being written

davguy, 2020-07-03:
If you encounter a bug use the following template.
If you have a feature request feel free to freestyle.
### Steps to reproduce
1. Create a sqlite3 database for a qcodes experiment in jupyter lab
2. run 'load_or_create_experiment'
3. start a new measurement
4. run 'load_by_run_spec' on this, or an older measurement in another jupyter lab

### Expected behaviour
One should be able to read a database which is being written. For example, for live plotting

### Actual behaviour
I get a "OperationalError: locking protocol" error. The sqlite database is not letting me in.

### System
**operating system**
Windows subsystem for linux: Ubuntu

**qcodes branch**
master

**qcodes commit**
Merge pull request #2066 from WilliamHPNielsen/remove_slave_master_refs

jenshnielsen, 2020-07-03:
I cannot reproduce this issue. For example https://github.com/QCoDeS/Qcodes/pull/2062 containst 2 notebooks that I have just run simultaiously. One writes to the database and the other one reads. 

It seems likely that your issue is related to WSL see for example https://github.com/microsoft/WSL/issues/2395 
Is there any particular reason you are using WSL?

jenshnielsen, 2020-09-25:
It seems like this is specific to wsl and there is not a lot we can do to fix that

closed 2020-09-25 by jenshnielsen