microsoft/Qcodes #2093: Bug: Keysight N5249A PNA-X network analyzer trace error

bugusan, 2020-07-22:
Regardless of which qcodes release(tried with 4 different qcodes version)  I use whenever I  want to have trace data of network analyzer, N5249A, I get could not convert string to float: 'S21' error. My code is simple. I first set the required parameter for the network analyzer and then try to get trace data: `data=qc.Measure(vna.trace).run(name='test')` 

my code is:
```
vna.sweep_mode('CONT')
vna.start(150.1e6)
vna.stop(350.1e6)
vna.power(-23)
vna.points(1601)# 1601
vna.if_bandwidth(1000)
vna.trace("S21")
data=qc.Measure(vna.trace).run(name='test') 
```

and it returns this error:

```
ValueError                                Traceback (most recent call last)
<ipython-input-8-48831a7c8d52> in <module>
----> 1 data=qc.Measure(vna.trace).run(name='test')
      2 
      3 # plot
      4 #plot = MatPlot(data.vna_magnitude, data.vna_phase)
      5 #plot.save()

~\anaconda3\envs\qcodes10\lib\site-packages\qcodes\measure.py in run(self, use_threads, quiet, station, **kwargs)
     83 
     84         # run the measurement as if it were a Loop
---> 85         self._dummyLoop.run(use_threads=use_threads,
     86                             station=station, quiet=True)
     87 

~\anaconda3\envs\qcodes10\lib\site-packages\qcodes\loops.py in run(self, use_threads, quiet, station, progress_interval, set_active, *args, **kwargs)
    745             if not quiet:
    746                 print(datetime.now().strftime('Started at %Y-%m-%d %H:%M:%S'))
--> 747             self._run_wrapper()
    748             ds = self.data_set
    749         finally:

~\anaconda3\envs\qcodes10\lib\site-packages\qcodes\loops.py in _run_wrapper(self, *args, **kwargs)
    794     def _run_wrapper(self, *args, **kwargs):
    795         try:
--> 796             self._run_loop(*args, **kwargs)
    797         finally:
    798             if hasattr(self, 'data_set'):

~\anaconda3\envs\qcodes10\lib\site-packages\qcodes\loops.py in _run_loop(self, first_delay, action_indices, loop_indices, current_values, **ignore_kwargs)
    882                     # log.debug('Going through callables at this sweep step.'
    883                     #           ' Calling {}'.format(f))
--> 884                     f(first_delay=delay,
    885                       loop_indices=new_indices,
    886                       current_values=new_values)

~\anaconda3\envs\qcodes10\lib\site-packages\qcodes\actions.py in __call__(self, loop_indices, **ignore_kwargs)
    165                 out_dict[param_id] = param_out
    166 
--> 167         self.store(loop_indices, out_dict)
    168 
    169 

~\anaconda3\envs\qcodes10\lib\site-packages\qcodes\data\data_set.py in store(self, loop_indices, ids_values)
    404          """
    405         for array_id, value in ids_values.items():
--> 406             self.arrays[array_id][loop_indices] = value
    407         self.last_store = time.time()
    408         if (self.write_period is not None and

~\anaconda3\envs\qcodes10\lib\site-packages\qcodes\data\data_array.py in __setitem__(self, loop_indices, value)
    339         self._update_modified_range(min_li, max_li)
    340 
--> 341         self.ndarray.__setitem__(loop_indices, value)
    342 
    343     def __getitem__(self, loop_indices):

ValueError: could not convert string to float: 'S21'
```

astafan8, 2020-07-22:
I'm not very much familiar with that driver/instrument but i think `vna.trace` is just the particular S-parameter that you are interested in. The actual data is available in `.magnitude` and `.phase` parameters, see the example notebook here https://qcodes.github.io/Qcodes/examples/driver_examples/Qcodes%20example%20with%20Keysight%20Network%20Analyzer.html.

bugusan, 2020-07-22:
> I'm not very much familiar with that driver/instrument but i think `vna.trace` is just the particular S-parameter that you are interested in. The actual data is available in `.magnitude` and [...]

thank you for your comment. Actually, in that way, I could get data but since I am sweeping voltage from some other instruments, I have 2D loop, to be able to make all my instrument work at the same time I need VNA to work via qcodes.Measure function

astafan8, 2020-07-22:
> to be able to make all my instrument work at the same time I need VNA to work via qcodes.Measure function

i'm not sure i understand why. I wanted to suggest that instead of `data=qc.Measure(vna.trace).run(name='test')` one needs to use `data=qc.Measure(vna.magnitude).run(name='test')` because `vna.magnitude` parameter acquires the data while `vna.trace` parameter jsut allows you to select which S parameter you're interested in.

----

anyway, did this issue get resolved? if so, please close it :)

bugusan, 2020-07-27:
I agree with using  vna.magnitude. However, this time I get unhashable type: 'numpy.ndarray' error.

`data=qc.Measure(vna.magnitude).run()`

```
ypeError                                 Traceback (most recent call last)
<ipython-input-35-4a14ea72942f> in <module>
----> 1 data=qc.Measure(vna.magnitude).run()
      2 

~\anaconda3\envs\qcodes10\lib\site-packages\qcodes\measure.py in run(self, use_threads, quiet, station, **kwargs)
     75         """
     76 
---> 77         data_set = self._dummyLoop.get_data_set(**kwargs)
     78 
     79         # set the DataSet to local for now so we don't save it, since

~\anaconda3\envs\qcodes10\lib\site-packages\qcodes\loops.py in get_data_set(self, *args, **kwargs)
    656         """
    657         if self.data_set is None:
--> 658             data_set = new_data(arrays=self.containers(), *args, **kwargs)
    659             self.data_set = data_set
    660 

~\anaconda3\envs\qcodes10\lib\site-packages\qcodes\loops.py in containers(self)
    466                 # note that this supports lists (separate output arrays)
    467                 # and arrays (nested in one/each output array) of return values
--> 468                 action_arrays = self._parameter_arrays(action)
    469 
    470             else:

~\anaconda3\envs\qcodes10\lib\site-packages\qcodes\loops.py in _parameter_arrays(self, action)
    548             for j, (vij, nij, lij, uij) in enumerate(zip(sp_vi, sp_ni, sp_li, sp_ui)):
    549                 sp_def = (shape[: 1 + j], j, setpoints, vij, nij, lij, uij)
--> 550                 if sp_def not in all_setpoints:
    551                     all_setpoints[sp_def] = self._make_setpoint_array(*sp_def)
    552                     out.append(all_setpoints[sp_def])

TypeError: unhashable type: 'numpy.ndarray'
```

astafan8, 2020-07-27:
@bugusan a side question - do you absoltely need to use `qcodes.Measure`? Couldn't you use `qcodes.data_set.measurements.Measurement` context manager as described in this doc https://qcodes.github.io/Qcodes/examples/DataSet/Performing-measurements-using-qcodes-parameters-and-dataset.html ?

astafan8, 2020-07-27:
I couldn't figure out why the "unhashable type: 'numpy.ndarray'" occurs, perhaps, `Loop` and `Measure` are not smart enough about parameters which return numpy arrays of data. You could ask this question on our Slack channel, perhaps other people from our community know - do you know how to access the Slack channel?

bugusan, 2020-07-27:
would you please advise me of how to access the Slack chann?

astafan8, 2020-07-27:
@bugusan just send an email to Mikhail.Astafev(at)microsoft.com and I will invite you to the Slack channel