microsoft/Qcodes #2182: how to install interactive widgets?

sofomryu, 2020-09-21:
Steps to reproduce
1. I was following the 15 minutes to QCoDeS to use interactive widgets and I put the next code
from qcodes.interactive_widget import experiments_widget
2.and then it shows no module named 'qcodes.interactive_widget'
3.so I install widgets in anaconda promt using next code
conda install -c conda-forge ipywidgets
4. still it shows ModuleNotFoundError: No module named 'qcodes.interactive_widget'
5. I update my qcodes version to recent one and it still doesn't work.

### Expected behaviour
Success to import interactive_widget

### Actual behaviour
ModuleNotFoundError: No module named 'qcodes.interactive_widget'

### System
**WINDOW 10**

**qcodes branch**

**qcodes commit**

jenshnielsen, 2020-09-21:
The interactive widget was only added in version 0.18 that just came out. Can you check which version of qcodes that you are running

jenshnielsen, 2020-09-21:
E.g please check the exact version with something like
```
In [6]: import qcodes

In [7]: qcodes.__version__
Out[7]: '0.18.0+76.g894d08d52d'
```

sofomryu, 2020-09-21:
> E.g please check the exact version with something like
> 
> ```
> In [6]: import qcodes
> 
> In [7]: qcodes.__version__
> Out[7]: '0.18.0+76.g894d08d52d'
> ```

Thanks a lot! I check my version as 0.17.0.
So to solve this problem, I followed the updating of qcodes as below in anaconda promt.

conda update -n base conda -c defaults
conda env update

But it still ver 0.17.0. can you please give me an link how to update properly?

jenshnielsen, 2020-09-21:
That only upgrades the depdendencies of qcodes. You should be able to upgrade qcodes with `pip install --upgrade qcodes`

jenshnielsen, 2020-09-21:
Closing since this seems to be resolved. Please ping again if you have more issues

closed 2020-09-21 by jenshnielsen

sofomryu, 2020-09-21:
> That only upgrades the depdendencies of qcodes. You should be able to upgrade qcodes with `pip install --upgrade qcodes`

Thanks for the reply! But trying `pip install --upgrade qcdoes` still give me an version 0.17.0 
When I installed the qcodes, I installed it via downloading environment.yml file in the Getting Strated page, which is installing latest stable release. Is it why I can't upgrade to the 0.18.0?

astafan8, 2020-09-21:
> I installed it via downloading environment.yml file in the Getting Strated page, which is installing latest stable release. Is it why I can't upgrade to the 0.18.0?

unlikely.

another thing to try is to first ``pip uninstall qcodes`` and then ``pip install qcodes``. Even if that does not install qcodes 0.18, then try ``pip install qcodes==0.18.0``

jenshnielsen, 2020-09-21:
The environment.yml only sets up the dependencies for qcodes. It does not and never has installed qcodes it self. I just tried installing qcodes from pip and it works as expected.

```
pip install --upgrade qcodes
Collecting qcodes
  Downloading qcodes-0.18.0-py3-none-any.whl (1.1 MB)
     |████████████████████████████████| 1.1 MB 3.3 MB/s
```

Could you show us the exact command that you are running and the output it generates

sofomryu, 2020-09-21:
[image]

I found out that qcodes 0.18.0 is installed via anaconda promt. and then, I reboot.
But in the jupyter notebook, it still shows the qcodes version as 0.17.0

[image]

astafan8, 2020-09-21:
in that case it's likely that the jupyter notebook uses a different kernel, not the one where qcodes 0.18 was just installed.

sofomryu, 2020-09-21:
> in that case it's likely that the jupyter notebook uses a different kernel, not the one where qcodes 0.18 was just installed.

Thanks a lot! I'll just delete whole anaconda, jupyter and install them again.

sofomryu, 2020-09-21:
Great! it is working now. I really appreciate your help. Have a niceday.