microsoft/Qcodes #2223: Storing (camera) array data fails (ParameterWithSetpoints)

ThorstenGroh, 2020-09-28:
If I understood correctly, `ArrayParameter` is deprecated and `ParameterWithSetpoints` is now the way to describe multidimensional parameters like 2D images coming from a camera for example. However I can't find a way of storing this data neither using a `Loop` nor the `DataSaver` functionality. 

I would prefer saving it using the Loop functionality, since we plan to work with the HDF5 backend for disk storage, that as far as I can see currently is not supported by the `DataSaver`.

### Steps to reproduce
1. Create Instrument with a `ParameterWithSetpoints` field. Here I am choosing an camera like example, that provides the parameter `img` with the setpoints `x` and `y`.

```python
import numpy as np
from qcodes import Parameter, ParameterWithSetpoints, Instrument
from qcodes.utils.validators import Arrays, Numbers

class DummyImage(ParameterWithSetpoints):

    def get_raw(self):
        nx = self.root_instrument.nx()
        ny = self.root_instrument.ny()
        return np.random.randint(0, 9, size=(ny, nx))

class DummyCamera(Instrument):
    def __init__(self, name: str, nx=3, ny=4, **kwargs):
        super().__init__(name, **kwargs)
        
        self.add_parameter(name="nx",
                           initial_value=nx,
                           vals=Numbers(0),
                           get_cmd=None,
                           set_cmd=None)
        
        self.add_parameter(name="ny",
                           initial_value=ny,
                           vals=Numbers(0),
                           get_cmd=None,
                           set_cmd=None)
        
        self.add_parameter(name="x",
                           vals=Arrays(shape=(self.nx.get,)),
                           get_cmd=lambda: np.arange(self.nx()))
        
        self.add_parameter(name="y",
                           vals=Arrays(shape=(self.ny.get,)),
                           get_cmd=lambda: np.arange(self.ny()))
        
        self.add_parameter(name="img",
                           parameter_class=DummyImage,
                           setpoints=(self.y, self.x),
                           vals=Arrays(shape=(self.ny.get, self.nx.get)))

cam = DummyCamera("cam")
```

2. Create a parameter to vary in the measurement.

```python
t = Parameter(name="t", set_cmd=None)
```

3. (Option I) Start a loop and try to store the data to the automatically created dataset.

```python
from qcodes.loops import Loop

loop = Loop(t[0:10:1]).each(cam.img)
loop.run()
```

4. (Option II) Start a measurement and try to store the data using the `DataSaver`.

```python
from qcodes import Measurement

meas = Measurement()
meas.register_parameter(t)
meas.register_parameter(cam.img, setpoints=(t,))

with meas.run() as datasaver:
    for set_t in np.arange(10):
        t.set(set_t)
        get_img = cam.img.get()
        
        datasaver.add_result((t, set_t), (cam.img, get_img))
```

### Expected behaviour

Data is stored in a multidimensional array 

### Actual behaviour

For the storage attempt using the `Loop` following error is raised:

```python
TypeError: only size-1 arrays can be converted to Python scalars

The above exception was the direct cause of the following exception:

Traceback (most recent call last):

  File "E:\sciebo\QuantumMetrology\Personal\Thorsten Groh\Code\Misc\test_qcodes\issue_minimal_example.py", line 63, in <module>
    loop.run()

  File "C:\tools\Anaconda3\lib\site-packages\qcodes\loops.py", line 716, in run
    self._run_wrapper()

  File "C:\tools\Anaconda3\lib\site-packages\qcodes\loops.py", line 765, in _run_wrapper
    self._run_loop(*args, **kwargs)

  File "C:\tools\Anaconda3\lib\site-packages\qcodes\loops.py", line 855, in _run_loop
    current_values=new_values)

  File "C:\tools\Anaconda3\lib\site-packages\qcodes\actions.py", line 167, in __call__
    self.store(loop_indices, out_dict)

  File "C:\tools\Anaconda3\lib\site-packages\qcodes\data\data_set.py", line 405, in store
    self.arrays[array_id][loop_indices] = value

  File "C:\tools\Anaconda3\lib\site-packages\qcodes\data\data_array.py", line 341, in __setitem__
    self.ndarray.__setitem__(loop_indices, value)

ValueError: setting an array element with a sequence.
```

This seems to be related to the open issue #627 , where the workaround of converting the data to a python `tuple` is provided. This however is not suitable for large amounts of data like MP camera images. Is there any update on this?

For the storage attempt using the `DataSaver` following error is raised:

```python
WARNING:qcodes.dataset.measurements:An exception occured in measurement with guid: aaaaaaaa-0000-0000-0000-0174d592e780;
Traceback:
Traceback (most recent call last):
  File "C:\tools\Anaconda3\lib\site-packages\qcodes\dataset\measurements.py", line 301, in _validate_result_deps
    self._interdeps.validate_subset(list(results_dict.keys()))
  File "C:\tools\Anaconda3\lib\site-packages\qcodes\dataset\descriptions\dependencies.py", line 456, in validate_subset
    raise DependencyError(param, missing_deps)
qcodes.dataset.descriptions.dependencies.DependencyError: cam_img has the following dependencies that are missing: {'cam_x', 'cam_y'}

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "E:\sciebo\QuantumMetrology\Personal\Thorsten Groh\Code\Misc\test_qcodes\issue_minimal_example.py", line 79, in <module>
    datasaver.add_result((t, set_t), (cam.img, get_img))
  File "C:\tools\Anaconda3\lib\site-packages\qcodes\dataset\measurements.py", line 141, in add_result
    self._validate_result_deps(results_dict)
  File "C:\tools\Anaconda3\lib\site-packages\qcodes\dataset\measurements.py", line 304, in _validate_result_deps
    'are missing.') from err
ValueError: Can not add result, some required parameters are missing.

Starting experimental run with id: 4. 
Traceback (most recent call last):

  File "C:\tools\Anaconda3\lib\site-packages\qcodes\dataset\measurements.py", line 301, in _validate_result_deps
    self._interdeps.validate_subset(list(results_dict.keys()))

  File "C:\tools\Anaconda3\lib\site-packages\qcodes\dataset\descriptions\dependencies.py", line 456, in validate_subset
    raise DependencyError(param, missing_deps)

DependencyError: cam_img has the following dependencies that are missing: {'cam_x', 'cam_y'}

The above exception was the direct cause of the following exception:

Traceback (most recent call last):

  File "E:\sciebo\QuantumMetrology\Personal\Thorsten Groh\Code\Misc\test_qcodes\issue_minimal_example.py", line 79, in <module>
    datasaver.add_result((t, set_t), (cam.img, get_img))

  File "C:\tools\Anaconda3\lib\site-packages\qcodes\dataset\measurements.py", line 141, in add_result
    self._validate_result_deps(results_dict)

  File "C:\tools\Anaconda3\lib\site-packages\qcodes\dataset\measurements.py", line 304, in _validate_result_deps
    'are missing.') from err

ValueError: Can not add result, some required parameters are missing.
```

The datasaver is apparently expecting `{'cam_x', 'cam_y'}` to be set as well. In this case these however should stay constant during the run, the camera is generally not providing the image pixel by pixel. This is were `ArrayParameter` sounds to be the more suited class to take. Is there a way to flag these parameters as constant?

I wonder if anybody encountered this problem before. Is there some driver implementing this behavior already? I looked through the spectrum analyzers, oscilloscopes drivers, but couldn't find anything suited. The Andor DU401 driver in the contrib_drivers repo is using python lists to return the data, the TPS 2012 is using `ArrayParameter`. 

I would be grateful for any ideas.

P.S.: Thanks a lot for this wonderful work here. I'm really looking forward to use QCoDeS more extensively in the lab.

jenshnielsen, 2020-09-29:
Both the Array parameters and the ParameterWithSetpoints intend for you to supply setpoints. QCoDeS was mostly designed with the idea in mind that you want to have explicit dependencies on all axis. There is currently no support for specifiying non changing setpoint axis but that is a resonable request. 

The `Loop` is part of the old legacy dataset and does not support ParameterWithSetpoint. I would recommend against ever using it.

However, for now I would suggest that you simply use a Parameter with a Array validator

Something like this

```
import numpy as np
from qcodes import Parameter, ParameterWithSetpoints, Instrument
from qcodes.utils.validators import Arrays, Numbers

class DummyImage(Parameter):

    def get_raw(self):
        nx = self.root_instrument.nx()
        ny = self.root_instrument.ny()
        return np.random.randint(0, 9, size=(ny, nx))

class DummyCamera(Instrument):
    def __init__(self, name: str, nx=3, ny=4, **kwargs):
        super().__init__(name, **kwargs)
        
        self.add_parameter(name="nx",
                           initial_value=nx,
                           vals=Numbers(0),
                           get_cmd=None,
                           set_cmd=None)
        
        self.add_parameter(name="ny",
                           initial_value=ny,
                           vals=Numbers(0),
                           get_cmd=None,
                           set_cmd=None)
        
        self.add_parameter(name="img",
                           parameter_class=DummyImage,
                           vals=Arrays(shape=(self.ny.get, self.nx.get)))

cam = DummyCamera("cam1")

```

```
meas = Measurement(exp=exp)

meas.register_parameter(cam.img)
with meas.run() as datasaver:
    datasaver.add_result((cam.img, cam.img()))
```

```
datasaver.dataset.get_parameter_data()
{'cam1_img': {'cam1_img': array([[[2, 0, 3],
          [3, 7, 2],
          [2, 3, 5],
          [4, 5, 3]]])}}
```
Note however, that the build in plotting in QCoDeS does not currently know how to plot this 
due to the missing setpoints

ThorstenGroh, 2020-09-29:
Thanks, that at least stores the data now.

However if providing a setpoint parameter that is varied from shot to shot, and registering it to the measurement, the resulting dataset has a quite storage hungry format of the set values.

This
```python
import numpy as np
from qcodes import (Parameter, ParameterWithSetpoints, Instrument, Measurement,
                    load_or_create_experiment)
from qcodes.utils.validators import Arrays, Numbers
from pprint import pprint

class DummyImage(Parameter):
    def get_raw(self):
        nx = self.root_instrument.nx()
        ny = self.root_instrument.ny()
        return np.random.randint(0, 9, size=(ny, nx))

class DummyCamera(Instrument):
    def __init__(self, name: str, nx=3, ny=4, **kwargs):
        super().__init__(name, **kwargs)
        
        self.add_parameter(name="nx",
                           initial_value=nx,
                           vals=Numbers(0),
                           get_cmd=None,
                           set_cmd=None)
        
        self.add_parameter(name="ny",
                           initial_value=ny,
                           vals=Numbers(0),
                           get_cmd=None,
                           set_cmd=None)
        
        self.add_parameter(name="img",
                           parameter_class=DummyImage,
                           vals=Arrays(shape=(self.ny.get, self.nx.get)))

cam = DummyCamera("cam")
t = Parameter("t", set_cmd=None)

exp = load_or_create_experiment(experiment_name="Test", sample_name="Test") 
meas = Measurement(exp=exp)

meas.register_parameter(t)
meas.register_parameter(cam.img, (t,))

with meas.run() as datasaver:
    for t1 in np.arange(0, 5, 1):
        t.set(t1)
        datasaver.add_result((t, t1), (cam.img, cam.img()))
    
    data = datasaver.dataset

pprint(data.get_parameter_data())
```
returns
```
Starting experimental run with id: 6. 
{'cam_img': {'cam_img': array([[[8, 2, 0],
        [5, 2, 5],
        [1, 2, 3],
        [6, 4, 5]],

       [[2, 4, 6],
        [0, 4, 4],
        [6, 4, 4],
        [5, 3, 5]],

       [[2, 8, 4],
        [3, 3, 1],
        [6, 4, 5],
        [8, 2, 2]],

       [[2, 2, 2],
        [8, 7, 3],
        [6, 0, 5],
        [8, 7, 4]],

       [[1, 0, 7],
        [8, 2, 3],
        [1, 7, 7],
        [1, 2, 2]]]),
             't': array([[[0., 0., 0.],
        [0., 0., 0.],
        [0., 0., 0.],
        [0., 0., 0.]],

       [[1., 1., 1.],
        [1., 1., 1.],
        [1., 1., 1.],
        [1., 1., 1.]],

       [[2., 2., 2.],
        [2., 2., 2.],
        [2., 2., 2.],
        [2., 2., 2.]],

       [[3., 3., 3.],
        [3., 3., 3.],
        [3., 3., 3.],
        [3., 3., 3.]],

       [[4., 4., 4.],
        [4., 4., 4.],
        [4., 4., 4.],
        [4., 4., 4.]]])}}
```
Is this necessary for the logic of the dependencies? This would get a little bit inconvinient and space inefficient for large image data for example. Or is this just the returned representation and the database is storing the data more efficient?

jenshnielsen, 2020-09-30:
@ThorstenGroh Those extra values of the `t` setpoint are not actually storred within the database. They are expanded on load time by get_parameter_data

ThorstenGroh, 2020-09-30:
Okay, that is nice. I think, I can close this now then. Thanks a lot for the quick help.

closed 2020-09-30 by ThorstenGroh