microsoft/Qcodes #2231: Qcodes driver error: InstrumentChannel parameters are not the attributes to the parent instrument?

Saba-Mehshar, 2020-10-02:
Hi, I have been trying to use a qcodes driver for the Network Analyzer but getting an attribute error. My Instrument Channel has a parameter called 'magnitude', but it does not get bind to the parent instrument and giving this error when I try to measure it: 

AttributeError                            Traceback (most recent call last)

<ipython-input-4-753c436e1d0b> in <module>
      1 # Run a measurement
      2 meas = Measurement()
----> 3 meas.register_parameter(vna.magnitude)
      4 
      5 with meas.run() as datasaver:

~\.conda\envs\qcodes\lib\site-packages\qcodes\utils\helpers.py in __getattr__(self, key)
    396         raise AttributeError(
    397             "'{}' object and its delegates have no attribute '{}'".format(
--> 398                 self.__class__.__name__, key))
    399 
    400     def __dir__(self):

AttributeError: 'NetworkAnalyser_S5048' object and its delegates have no attribute 'magnitude'

However, I have no trouble when setting the parameters which are added to the parent instrument. If anyone knows how to bind the Instrument channels with the main instrument, then please respond to this issue.
Many thanks in advance.

jenshnielsen, 2020-10-02:
Hi @Saba-Mehshar 
Hi @Saba-Mehshar 

Is there a particular reason why it needs to be bould to the instrument. 
Normally I would just supply the full path to the parameter.
`vna.channel.magnitude` where channel is the name that your channel is registered under. 

A few more questions to help understand what is going on. 

* Which exact VNA is this. The R&S driver from qcodes? 
* If you just do `vna.magnitude.get()` does the that error in the same way.

Saba-Mehshar, 2020-10-02:
Hi @jenshnielsen 
Thank you very much for your quick response!

I have used the both vna.channel.magnitude and vna.magnitude.get() , yet getting the similar attribute error.

---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-6-d6aa0f5e0f28> in <module>
----> 1 vna.magnitude.get()

~\.conda\envs\qcodes\lib\site-packages\qcodes\utils\helpers.py in __getattr__(self, key)
    396         raise AttributeError(
    397             "'{}' object and its delegates have no attribute '{}'".format(
--> 398                 self.__class__.__name__, key))
    399 
    400     def __dir__(self):

AttributeError: 'NetworkAnalyser_S5048' object and its delegates have no attribute 'magnitude'

This is a copper mountain S5048 2-Port 4.8 GHz Analyzer, and I have been using a Qcodes  Keysight Network Analyzers  N5245A driver (have only made fewer changes). I can control my VNA with this driver (frequency, bandwidth and power etc.) and have never got any issue but cannot plot the results.
This is how my code looks like in jupyter notebook:

"" import logging
import matplotlib.pyplot as plt

import qcodes as qc
from qcodes.dataset.experiment_container import new_experiment, load_experiment_by_name
from qcodes.dataset.measurements import Measurement
from qcodes.dataset.plotting import plot_by_id
import qcodes.instrument_drivers.networkanalyser.NetworkAnalyser_S5048 as NetworkAnalyser_S5048
from qcodes.instrument.specialized_parameters import ElapsedTimeParameter

logger = logging.getLogger()

exp_name = 'VNA_Example'
sample_name = '2'
try:
    exp = load_experiment_by_name(exp_name, sample=sample_name)
    print('Experiment loaded. Last ID no:', exp.last_counter)
except ValueError:
    exp = new_experiment(exp_name, sample_name)
    print('Starting new experiment.')

Experiment loaded. Last ID no: 68

vna = NetworkAnalyser_S5048.NetworkAnalyser_S5048('vna', 'TCPIP0::localhost::5025::SOCKET')

Connected to: CMT S5048 (serial:14027062, firmware:19.3.0/2) in 0.10s

vna.power(-20)
vna.start(20e3)
vna.stop(2e9)
vna.points(1000)
vna.active_trace(1)
vna.if_bandwidth(2e3)
vna.averaging(True)
vna.average_factor(3)""

No error up to this point. I think my instrument channel is not added to the parent instrument. I can send you the driver code if needed.
I have been getting this error for a week but couldn't resolve it. Please give me some suggestions. Thank you very much.

jenshnielsen, 2020-10-02:
As far as I can read from the driver of the N52xx I would expect that you need to use:

`vna.traces.magnitude` 

to get the values

Saba-Mehshar, 2020-10-02:
Hi, I have got the following error

---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
<ipython-input-8-fd7c02c62372> in <module>
----> 1 vna.traces.magnitude

~\.conda\envs\qcodes\lib\site-packages\qcodes\instrument\channel.py in __getattr__(self, name)
    437         """
    438         # Check if this is a valid parameter
--> 439         if name in self._channels[0].parameters:
    440             setpoints = None
    441             setpoint_names = None

IndexError: list index out of range

jenshnielsen, 2020-10-04:
Did you add a parameter to measure with the trace like here. https://qcodes.github.io/Qcodes/examples/driver_examples/Qcodes example with Keysight Network Analyzer.html?highlight=n5245a

```
pna.trace("S21")
```

Saba-Mehshar, 2020-10-05:
Hi @jenshnielsen 

yes, I have added a parameter :  self.add_parameter('trace',
                                                         label='Trace',
                                                         get_cmd=self._Sparam,
                                                         set_cmd=self._set_Sparam)

but I am still getting the same attribute error:
AttributeError: 'NetworkAnalyser_S5048' object and its delegates have no attribute 'trace'

I think there is some problem with my Qcodes driver(I have made some changes in N5245A!). 

I am now trying to use 'QCoDeS driver for the HP 8753D Network Analyzer'. It seems to work a bit, but readable snapshot says, 'couldn't update s_parameter and display_format'. Because of this I am unable to take a trace.
Here is my output in Jupyter notebook:

%matplotlib notebook

import qcodes as qc
from qcodes.instrument_drivers.networkanalyser.VNAS504 import VNAS504
from functools import partial

VNA = VNAS504('VNA', 'TCPIP0::localhost::5025::SOCKET')

Connected to: CMT S5048 (serial:14027062, firmware:19.3.0/2) in 0.10s

VNA.reset()
VNA.print_readable_snapshot(update=True)   
  
[VNA(VNAS504)] Snapshot: Could not update parameter: s_parameter
[VNA(VNAS504)] Snapshot: Could not update parameter: display_format

VNA:
	parameter         value
--------------------------------------------------------------------------------
IDN                :	{'vendor': 'CMT', 'model': 'S5048', 'serial': '14027062'...
averaging          :	OFF 
display_format     :	None 
display_reference  :	0 
display_scale      :	10 
number_of_averages :	10 
output_power       :	0 (dBm)
s_parameter        :	None 
start_freq         :	30000 (Hz)
stop_freq          :	1e+09 (Hz)
sweep_time         :	0 (s)
timeout            :	5 (s)
trace              :	Not available 
trace_points       :	201 

# Let's get a single trace of S21 with 10 averages
# with a frequency range from 30 kHz to 1 GHz
# on a log scale
VNA.display_format('Log Mag')
VNA.s_parameter('S21')
VNA.start_freq(30e3)
VNA.stop_freq(1e9)

# and let's adjust the y-scale to fit
VNA.display_scale(0.12)
VNA.display_reference(-0.1)

# and finally enable averaging
VNA.averaging('ON')
VNA.number_of_averages(10)

# Now aquire a trace
VNA.trace.prepare_trace()  # this must be called prior to a Measurement or Loop
tracedata = qc.Measure(VNA.trace).run()  # this returns a dataset
plot = qc.MatPlot(tracedata.arrays['VNA_trace'])

[... 81 lines cut ...]

~\.conda\envs\qcodes\lib\site-packages\pyvisa\resources\messagebased.py in _read_raw(self, size)
    398                     logger.debug('%s - reading %d bytes (last status %r)',
    399                                  self._resource_name, size, status)
--> 400                     chunk, status = self.visalib.read(self.session, size)
    401                     ret.extend(chunk)
    402             except errors.VisaIOError as e:

~\.conda\envs\qcodes\lib\site-packages\pyvisa\ctwrapper\functions.py in read(library, session, count)
   1582     buffer = create_string_buffer(count)
   1583     return_count = ViUInt32()
-> 1584     ret = library.viRead(session, buffer, count, byref(return_count))
   1585     return buffer.raw[:return_count.value], ret
   1586 

~\.conda\envs\qcodes\lib\site-packages\pyvisa\ctwrapper\highlevel.py in _return_handler(self, ret_value, func, arguments)
    191 
    192         if ret_value < 0:
--> 193             raise errors.VisaIOError(ret_value)
    194 
    195         if ret_value in self.issue_warning_on:

VisaIOError: ('VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.', "asking 'S11?' to <VNAS504: VNA>", 'getting VNA_s_parameter')

Please give suggestion with this one because HP 8753D Network Analyzer is quite similar to my VNA. Thank you very much.

jenshnielsen, 2020-10-05:
I am a bit confused. Are you trying to use a driver for first a Keysight and now an HP VNA for a instrument from a completly different vendor ? That is very unlikely to work. If you need a driver for the that instrument you will most likely have to write it from scratch. You may be lucky that some of the commands are the same but in general that is very unlikely to work

Saba-Mehshar, 2020-10-05:
@jenshnielsen 

I have rewritten all the SCPI commands for Copper mountain (have written the driver from scratch!) but I have been following the script of Keysight/HP VNA Qcodes driver. Cannot I follow the script of these drivers?

jenshnielsen, 2020-10-05:
The error that you show indicates that the instrument does not understand the command send when you execute `VNA.trace.prepare_trace()` so there is some kind of issue related to whatever you send to the instrument

Saba-Mehshar, 2020-10-06:
Thanks @jenshnielsen .

Yes, there was two commands missing in my vna driver! 
I am now getting "AttributeError: module 'qcodes' has no attribute 'Measure'." This is how my output looks like:

%matplotlib notebook
import qcodes as qc
from qcodes.instrument_drivers.networkanalyser.VNAS504 import VNAS504
from functools import partial

VNA = VNAS504('VNA', 'TCPIP0::localhost::5025::SOCKET')
Connected to: CMT S5048 (serial:14027062, firmware:19.3.0/2) in 0.10s

VNA.reset()

VNA.print_readable_snapshot(update=True)

VNA:
	parameter         value
--------------------------------------------------------------------------------
IDN                :	{'vendor': 'CMT', 'model': 'S5048', 'serial': '14027062'...
averaging          :	OFF 
display_format     :	SWR 
display_reference  :	0 
display_scale      :	10 
number_of_averages :	10 
output_power       :	0 (dBm)
s_parameter        :	S22 
start_freq         :	20000 (Hz)
stop_freq          :	4.8e+09 (Hz)
sweep_time         :	0 (s)
timeout            :	5 (s)
trace              :	Not available 
trace_points       :	201 

VNA.display_format('Lin Mag')
VNA.s_parameter('S11')
VNA.start_freq(30e3)
VNA.stop_freq(1e9)
VNA.display_scale(0.12)
VNA.display_reference(-0.1)
VNA.averaging('ON')
VNA.number_of_averages(20)

VNA.print_readable_snapshot(update=True)

VNA:
	parameter         value (There is still no change in the snapshot - nothing is being updated)
--------------------------------------------------------------------------------
IDN                :	{'vendor': 'CMT', 'model': 'S5048', 'serial': '14027062'...
averaging          :	OFF 
display_format     :	SWR 
display_reference  :	0 
display_scale      :	10 
number_of_averages :	10 
output_power       :	0 (dBm)
s_parameter        :	S22 
start_freq         :	30000 (Hz)
stop_freq          :	1e+09 (Hz)
sweep_time         :	0 (s)
timeout            :	5 (s)
trace              :	Not available (dim. less)
trace_points       :	201 

VNA.trace.prepare_trace()  
tracedata = qc.Measure(VNA.trace).run()  
plot = qc.MatPlot(tracedata.arrays['VNA_trace'])

---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
[... 312 lines cut ...]

    def _display_format_getter(self) -> str:
        """
        get_cmd for the display_format parameter
        """
        val_mapping = {'MLOG': 'Log Mag',
                       'PHAS': 'Phase',
                       'GDEL': 'Group Delay',
                       'SMIT': 'Smith',
                       'POL': 'Polar',
                       'MLIN': 'Lin Mag',
                       'REAL': 'Real',
                       'IMAG': 'Imag',
                       'SWR': 'SWR'}

        # keep asking until we find the currently used format
        for cmd in val_mapping.keys():
            resp = self.ask('CALC:FORM?'.format(cmd))
            if resp in ['1', '1\n']:
                break

        return val_mapping[cmd]

Not sure where I am making mistakes. 
Sorry, this has been really long comment.

jenshnielsen, 2020-10-06:
You are using the old qcodes measurement where the import has changed. I just fixed the hp notebook so it can run

I would suggest that you worry about getting the driver working first. 

Then you should read this notebook https://qcodes.github.io/Qcodes/examples/DataSet/Performing-measurements-using-qcodes-parameters-and-dataset.html to understand how to do the measurements 

Could you make a pull request against https://github.com/qcodes/qcodes_contrib_drivers with your new driver so we can look at it in a better interface

Saba-Mehshar, 2020-10-06:
Hi @jenshnielsen 

I have pulled a new request against https://github.com/qcodes/qcodes_contrib_drivers .
Thank you very much for your consistent assistance.

jenshnielsen, 2020-10-18:
This has moved to the contrib driver pr https://github.com/QCoDeS/Qcodes_contrib_drivers/pull/78 and seems to be more of a question of how to develop a driver than an error so closing this one

closed 2020-10-18 by jenshnielsen