microsoft/Qcodes #2285: Can't open Database for old data RuntimeError: No parameters supplied

tmautry, 2020-10-14:
If you encounter a bug use the following template.
If you have a feature request feel free to freestyle.
### Steps to reproduce
1. I have three files testing.hdf5, testing.hdf5-shm, testing.hdf5-wal into which I recorded some data
2. Now I would like to load that data

import numpy as np
import matplotlib.pyplot as plt
import pandas as pd

import qcodes as qc
from qcodes.dataset.legacy_import import import_dat_file

from qcodes import initialise_database
from qcodes import load_or_create_experiment
from qcodes.dataset.measurements import Measurement
from qcodes.dataset.plotting import plot_by_id
from qcodes.dataset.data_set import load_by_id

initialise_database()

exp = load_or_create_experiment('old_data_loading', sample_name='QS-ES-14-12')

run_ids = import_dat_file('C:/Users/tmautry/Documents/Data/testing.hdf5')

3.  This gives the error RuntimeError: No parameters supplied

### Expected behaviour
Should load the data
as per the instructions found 
https://qcodes.github.io/Qcodes/examples/DataSet/Load%20old%20data.html

### Actual behaviour
RuntimeError: No parameters supplied

astafan8, 2020-10-15:
Hey @pyquest1 , a clarification question:

* the `C:/Users/tmautry/Documents/Data/testing.hdf5` file - has it been created by qcodes Loop (aka "old/legacy" dataset) or is just "some old data"?
  * Also, `testing.hdf5, testing.hdf5-shm, testing.hdf5-wal` - this looks like you're actually having an sqlite database file with the name `testing.hdf5`, i deduct that because the -wal and -shm files can exist if an sqlite is opened in so-called WAL mode (which qcodes uses by default).
  * old/legacy qcodes dataset does not store data just in a single file. as far as i remember, the folder structure matters as well, take a look at and example here https://github.com/QCoDeS/Qcodes/tree/master/qcodes/tests/dataset/fixtures/2018-01-17 (which is also used in the doc page that you are referring to)
  * i am not very knowledgable about the old qcodes dataset and the loop, hence i don't know for sure, but it seems unlikely that `import_dat_file` function would load HDF5 files.

pyquest1, 2020-10-15:
Hi Astafan8

Not sure why you are saying this is legacy?

Saving was done using 
initialise_or_create_database_at()
exp = load_or_create_experiment()

meas = Measurement(...)

meas.register_parameter(...) 
meas.register_parameter(...)  

with meas.run() as datasaver:
           data =...
                datasaver.add_result(...)

What would load it since qcodes is what saved it?

jenshnielsen, 2020-10-16:
@pyquest1 Note where you are importing  `import_dat_file` from (`qcodes.dataset.legacy_import import`)
As indicated from the location this is imported from it is indended to load legacy data (written with the qcodes loop as in these 
examples https://qcodes.github.io/Qcodes/examples/index.html#legacy-examples).

Please see here for instance of how to load a dataset from the 
https://qcodes.github.io/Qcodes/examples/DataSet/Performing-measurements-using-qcodes-parameters-and-dataset.html#Reloading-datasets for how to load data written as in your example. 

Note that the file that you are storing data in is a sqlite database and not an hdf5 file. I strongly recommend that you give it an extension that reflects this (so .db rather than .hdf5)

pyquest1, 2020-10-16:
Hi Jens,

Sorry, I didn't notice the legacy issue. I have read the documentation but it is long.  As a result I have just started searching for what I need in the documentation or just directly reading the github files.

I thought qcodes supported hdf5 for containing the "database"?  I could swear I read that in the documentation but now cannot find it...

The reloaded dataset example is nearly useless to me.  Why/how would I know beforehand what the captured_run_id is for an experiment that occurred months ago?

Thanks again for your help.

jenshnielsen, 2020-10-16:
The legacy dataset supports hdf5 that is probably what you have seen. There are various ways to load data according to the sample name and the experiment that you have storred it under. It should all be in the documentation. Could you give an example of how you would like to be able to load data? You will need to load it according to some identifier. QCoDeS stores a the run_id, an experiment name, a samle name and the date all can be used to load the data

jenshnielsen, 2020-12-15:
It seems like this is mostly a question about how to use the qcodes dataset and not a bug. Please ping if there is an actual bug to have this reopened

closed 2020-12-15 by jenshnielsen