microsoft/Qcodes #2310: Can't import dummy intrument with measurement

Danielgb23, 2020-10-19:
If you encounter a bug use the following template.
If you have a feature request feel free to freestyle.
### Steps to reproduce
import class dummy instrument with measurement

Ubuntu 16.04
python 3.6

Danielgb23, 2020-10-19:
I checked the files of qcodes in my computer and it's missing. And I just updated qcodes. If it's a new feature it shouldn't be added to the official guide until it's added in a update. Or my system is having some trouble with the updates

astafan8, 2020-10-20:
sorry, @Danielgb23 , the documentation is a bit ahead of the qcodes releases. A new release of qcodes will be published in a matter of days where the dummy instrument will be avaiable :) (if you really want to try things now, either run the tutorial in binder or install qcodes from git repo clone)

closed 2020-10-20 by astafan8