microsoft/Qcodes #2407: Error to connect keithly2400 with RS232

sofomryu, 2020-11-06:
If you encounter a bug use the following template.
If you have a feature request feel free to freestyle.
### Steps to reproduce
1.
[image]

2.
3.

### Expected behaviour
Tell us what should happen
The connection should be well operated and it worked well before.

### Actual behaviour
Tell us what happens instead
Now when I try to connect them, it shows me 'module 'pyvisa.constants' has no attribute 'BufferOperation''
I set both RS232 and keithley2400 with 
Baud: Rate 9600
Data: Bits 8
Stop: Bits 1
Parity: None
Terminal: LF

And I checked it's working well when I type command via NI MAX. 
What should I consider to solve this problem?

### System
**operating system**
windows10 64bit
**qcodes branch**
0.19.0
**qcodes commit**

jenshnielsen, 2020-11-06:
Please try upgrading to qcodes `0.19.2` You have an incopatible combination qcodes and pyvisa installed. QCoDeS 0.19.2 should have support for both pyvisa 1.10 and 1.11

sofomryu, 2020-11-09:
> Please try upgrading to qcodes `0.19.2` You have an incopatible combination qcodes and pyvisa installed. QCoDeS 0.19.2 should have support for both pyvisa 1.10 and 1.11

Thanks a lot! now it's working.

closed 2020-11-09 by sofomryu