microsoft/Qcodes #2409: Question about DataSaver

pyquest1, 2020-11-06:
I have a general question about using datasaver in a style that is slightly different from the examples.

### Steps to reproduce
1. Go to https://qcodes.github.io/Qcodes/examples/15_minutes_to_QCoDeS.html#Measurement

```

meas = Measurement(exp=exp, station=station)
meas.register_parameter(dac.ch1)  # register the first independent parameter
meas.register_parameter(dmm.v1, setpoints=(dac.ch1,))  # now register the dependent oone

meas.write_period = 2

with meas.run() as datasaver:
    for set_v in np.linspace(0, 25, 10):
        dac.ch1.set(set_v)
        get_v = dmm.v1.get()
        datasaver.add_result((dac.ch1, set_v),
                             (dmm.v1, get_v))

    dataset = datasaver.dataset  # convenient to have for plotting

```

###  More complicated Datasaver usage.
Datasaver is designed in a very nested loop fashion.  With the assumption that if I set x and measure y I get a dataset (x,y). This is well and good.  But there are more complicated situations and I am unsure how to deal with them.

Ex 1:
  Let's say I set x and measure many y i.e. y = {y1,y2,y3,y4,y5 ...}.  What is the right way to save the data in datasaver?  Is there a convenient method or would I have to build a for loop to add each y individually with the same x? Psuedo code below:

```

with meas.run() as datasaver:
    for set_v in np.linspace(0, 25, 10):
        dac.ch1.set(set_v)
        get_v = dmm.v1.get() # returns an array of voltages
  
        for v in get_v:
               datasaver.add_result((dac.ch1, set_v),
                             (dmm.v1, v))

    dataset = datasaver.dataset  # convenient to have for plotting

```

Ex 2: 

Let's say I wanted to add a finished array of results at the end of the loop to datasaver instead of at each step in datasaver.  Is there an easy way to do this? Pseudo code below.

```

with meas.run() as datasaver:
    dat_list = list()
    for set_v in np.linspace(0, 25, 10):
        dac.ch1.set(set_v)
        get_v = dmm.v1.get() # returns an array of voltages
        dat_list.append([set_v,get_v])
    dat = np.asarray(dat_list)
    datasaver.add_result(dat)

    dataset = datasaver.dataset  # convenient to have for plotting

```

FarBo, 2020-11-09:
Hello @pyquest1 
I need to be a bit certain how your question can be handled with datasaver in Measurement context manager. But, what I am certain about is that you can use qcodes doNd functions for the same reason. Please look at my PR #2306 , where Measurement is compared with doNd. Basically you only need to set values you want measure and as many parameters you want to measure over in a single line:
for example:
do1d(sweeping_parameter, 0, 10, 1, 1, measured_parameter1, measured_parameter2, ...)

Basically, doNd has Measurement and datasaver wrapped in it and you don't need to set any of those yourself.

In the mentioned PR, you see that that I used arrayed parameters for both measuring and measured parameters.

But, if you want to explicitly use Measurement context manager, I can respond to that ASAP.

pyquest1, 2020-11-20:
Hi Farbo,

I'm having some trouble accessing your file.

In any event.  I'm not trying to do a multidimensional sweep.  It's just that there are instances where multiple y values are returned.  For instance an oscilliscope might return N points and not a single point after a function call.  So I am trying to understand the right use in this situation.

FarBo, 2020-11-23:
@pyquest1 

You need to do some clicks in order to view the notebook. Anyway, I think you can find your answer here (section: Performing several measuments concurrently):
https://github.com/QCoDeS/Qcodes/blob/master/docs/examples/DataSet/Performing-measurements-using-qcodes-parameters-and-dataset.ipynb

Let me know if it solves your issue :)

closed 2021-09-02 by FarBo