microsoft/Qcodes #2764: xarray dataset created by to_xarray_dataset cannot be saved

peendebak, 2021-02-22:
### Steps to reproduce
1. Run the minimal example included below

```
import numpy as np
import os
import qcodes as qc
from qcodes import load_or_create_experiment, initialise_database, Measurement
from qcodes.tests.instrument_mocks import DummyInstrument, DummyInstrumentWithMeasurement
import tempfile

qc.logger.start_all_logging()

dac = DummyInstrument('dac', gates=['ch1', 'ch2'])
dmm = DummyInstrumentWithMeasurement('dmm', setter_instr=dac)
station = qc.Station(dmm, dac)

initialise_database()
exp = load_or_create_experiment(experiment_name='working_with_pandas',
                          sample_name="no sample")
#%%
meas = Measurement(exp)
meas.register_parameter(dac.ch1)  # register the first independent parameter
meas.register_parameter(dac.ch2)  # register the second independent parameter
meas.register_parameter(dmm.v2, setpoints=(dac.ch1, dac.ch2))  # register the dependent one

with meas.run() as datasaver:

    for v1 in np.linspace(-1, 0, 4, endpoint=False):
        for v2 in np.linspace(-1, 1, 3):
            dac.ch1(v1)
            dac.ch2(v2)
            val = dmm.v2.get()
            datasaver.add_result((dac.ch1, v1),
                                 (dac.ch2, v2),
                                 (dmm.v2, val))

dataset1 = datasaver.dataset
xaDataSet = dataset1.to_xarray_dataset()

xaDataSet.to_netcdf(os.path.join(tempfile.mkdtemp(), "saved_on_disk.nc")  )
# the problems is in the 'depends_on' field that is in the attributes

```
### Expected behaviour
The xarray dataset created by `to_xarray_dataset` should be saved by the default `xarray` methods

### Actual behaviour

On running `xaDataSet.to_netcdf('file.nc')` an exception is generated.
The problems seems to be caused by the `depends_on` field in the generated xarray structure.

### System
Windows

qcodes 0.16.0+3019.g706305583'

astafan8, 2021-02-22:
@peendebak could you paste the complete traceback, please?

jenshnielsen, 2021-02-23:
It seems like xarrays to netcdf simply does not like lists of strings in its attributes.

```python
import xarray as xr
import numpy as np
a = xr.DataArray(np.array([1,2,3]))
a.attrs = {'a': ["foobar"]}
a.to_netcdf(os.path.join(tempfile.mkdtemp(), "saved_on_disk.nc"))
```

```
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
<ipython-input-22-b4e395991fd1> in <module>
      1 a = xr.DataArray(np.array([1,2,3]))
      2 a.attrs = {'a': ["foobar"]}
----> 3 a.to_netcdf(os.path.join(tempfile.mkdtemp(), "saved_on_disk.nc")  )

c:\users\jens-work\miniconda3\envs\qcodespip\lib\site-packages\xarray\core\dataarray.py in to_netcdf(self, *args, **kwargs)
   2662             dataset = self.to_dataset()
   2663 
-> 2664         return dataset.to_netcdf(*args, **kwargs)
   2665 
   2666     def to_dict(self, data: bool = True) -> dict:

c:\users\jens-work\miniconda3\envs\qcodespip\lib\site-packages\xarray\core\dataset.py in to_netcdf(self, path, mode, format, group, engine, encoding, unlimited_dims, compute, invalid_netcdf)
   1652             unlimited_dims=unlimited_dims,
   1653             compute=compute,
-> 1654             invalid_netcdf=invalid_netcdf,
   1655         )
   1656 

c:\users\jens-work\miniconda3\envs\qcodespip\lib\site-packages\xarray\backends\api.py in to_netcdf(dataset, path_or_file, mode, format, group, engine, encoding, unlimited_dims, compute, multifile, invalid_netcdf)
   1125     finally:
   1126         if not multifile and compute:
-> 1127             store.close()
   1128 
   1129     if not compute:

c:\users\jens-work\miniconda3\envs\qcodespip\lib\site-packages\xarray\backends\scipy_.py in close(self)
    218 
    219     def close(self):
--> 220         self._manager.close()

c:\users\jens-work\miniconda3\envs\qcodespip\lib\site-packages\xarray\backends\file_manager.py in close(self, needs_lock)
    220             file = self._cache.pop(self._key, default)
    221             if file is not None:
--> 222                 file.close()
    223 
    224     def __del__(self):

c:\users\jens-work\miniconda3\envs\qcodespip\lib\site-packages\scipy\io\netcdf.py in close(self)
    294         if hasattr(self, 'fp') and not self.fp.closed:
    295             try:
--> 296                 self.flush()
    297             finally:
    298                 self.variables = OrderedDict()

c:\users\jens-work\miniconda3\envs\qcodespip\lib\site-packages\scipy\io\netcdf.py in flush(self)
    404         """
    405         if hasattr(self, 'mode') and self.mode in 'wa':
--> 406             self._write()
    407     sync = flush
    408 

c:\users\jens-work\miniconda3\envs\qcodespip\lib\site-packages\scipy\io\netcdf.py in _write(self)
    416         self._write_dim_array()
    417         self._write_gatt_array()
--> 418         self._write_var_array()
    419 
    420     def _write_numrecs(self):

c:\users\jens-work\miniconda3\envs\qcodespip\lib\site-packages\scipy\io\netcdf.py in _write_var_array(self)
    464             # Set the metadata for all variables.
    465             for name in variables:
--> 466                 self._write_var_metadata(name)
    467             # Now that we have the metadata, we know the vsize of
    468             # each record variable, so we can calculate recsize.

c:\users\jens-work\miniconda3\envs\qcodespip\lib\site-packages\scipy\io\netcdf.py in _write_var_metadata(self, name)
    485             self._pack_int(dimid)
    486 
--> 487         self._write_att_array(var._attributes)
    488 
    489         nc_type = REVERSE[var.typecode(), var.itemsize()]

c:\users\jens-work\miniconda3\envs\qcodespip\lib\site-packages\scipy\io\netcdf.py in _write_att_array(self, attributes)
    445             for name, values in attributes.items():
    446                 self._pack_string(name)
--> 447                 self._write_att_values(values)
    448         else:
    449             self.fp.write(ABSENT)

c:\users\jens-work\miniconda3\envs\qcodespip\lib\site-packages\scipy\io\netcdf.py in _write_att_values(self, values)
    557     def _write_att_values(self, values):
    558         if hasattr(values, 'dtype'):
--> 559             nc_type = REVERSE[values.dtype.char, values.dtype.itemsize]
    560         else:
    561             types = [(int, NC_INT), (float, NC_FLOAT), (str, NC_CHAR)]

KeyError: ('U', 24)
```

astafan8, 2021-02-23:
@jenshnielsen yes, that makes sense, let's add this dependency!

PR #2769 (2021-02-23): Explicit dependency on netcdf4
Fixes #2764
closed 2021-02-23 by jenshnielsen