microsoft/Qcodes #2815: problem with logging  "No package metadata was found for qcodes

erikawa-e, 2021-03-11:
I have a problem with logging.
Specifically, I received the following error: "No package metadata was found for qcodes".

This happens because 
importlib_metadata.Distribution.from_name('qcodes').requires gives None.

(importlib_metadata.Distribution.from_name('jsonschema').requires gives

['attrs (>=17.4.0)',
 'pyrsistent (>=0.14.0)',
 'setuptools',
 'six (>=1.11.0)',
 'functools32 ; python_version < "3"',
 'importlib-metadata ; python_version < "3.8"',
 "idna ; extra == 'format'",
 "jsonpointer (>1.13) ; extra == 'format'",
 "rfc3987 ; extra == 'format'",
 "strict-rfc3339 ; extra == 'format'",
 "webcolors ; extra == 'format'",
 "idna ; extra == 'format_nongpl'",
 "jsonpointer (>1.13) ; extra == 'format_nongpl'",
 "webcolors ; extra == 'format_nongpl'",
 "rfc3986-validator (>0.1.0) ; extra == 'format_nongpl'",
 "rfc3339-validator ; extra == 'format_nongpl'"]

so this problem happens only for qcodes)

I have another computer where loggin is working fine.
There importlib_metadata.Distribution.from_name('qcodes').requires gives 

['numpy>=1.15',
 'pyvisa<1.12.0,>=1.11.0',
 'h5py>=2.8.0',
 'websockets>=7.0',
 'jsonschema>=3.0.0',
 'ruamel.yaml!=0.16.6,>=0.16.0',
 'wrapt>=1.10.4',
 'pandas>=0.25.0',
 'xarray>=0.16.2',
 'tabulate>=0.8.0',
 'tqdm>=4.32.2',
 'opencensus<0.8.0,>=0.7.10',
 'opencensus-ext-azure<2.0.0,>=1.0.4',
 'matplotlib>=2.2.3',
 'requirements-parser>=0.2.0',
 'typing_extensions>=3.7.4',
 'packaging>=20.0',
 'ipywidgets>=7.5.0',
 'broadbean>=0.9.1',
 'importlib-metadata<4.0.0,>=1.0.0; python_version < "3.8"',
 'pyqtgraph>=0.11.0; extra == "qtplot"',
 'slacker>=0.9.42; extra == "slack"',
 'pytest>=6.0.0; extra == "test"',
 'PyVisa-sim>=0.4.0; extra == "test"',
 'hypothesis>=5.49.0; extra == "test"',
 'pytest-xdist>=2.0.0; extra == "test"',
 'deepdiff>=5.0.2; extra == "test"',
 'pytest-mock>=3.0.0; extra == "test"',
 'pytest-rerunfailures>=5.0.0; extra == "test"',
 'lxml>=4.3.0; extra == "test"',
 'zhinst-qcodes>=0.1.1; extra == "zurichinstruments"']

jenshnielsen, 2021-03-11:
How did you install qcodes on the first machine. I cannot reproduce the issue and would guess this is more likely to be related to a broken installation

erikawa-e, 2021-03-11:
Yeah, I agree that it should be the installation problem. On the broken one, even "import qcodes" did not work until I added the qcodes folder to PYTHONPATH. 

I followed the same installation procedure for the both machines but one works and the other one does not work... I tried the re-installation several times on the not-working PC. However, the results are always the same... On the two machines, they have the exactly same version of the Windows10.

jenshnielsen, 2021-03-11:
I would strongly recommend against ever using PYTHONPATH. Using that means that you don't really install the package so its no surprise that the metadata is wrong.

We should investigate why the original install did not work. 

A few questions here to try to pin point what the issue might be.

* Did you install qcodes from a git clone or the official release (pip install qcodes)
* Which version of pip do you have installed `pip --version`
* Could it be that you have more than one version of python on the computer such that you install into one version and try to load from the other?

erikawa-e, 2021-03-12:
Thank you so much for your inputs.

> Did you install qcodes from a git clone or the official release (pip install qcodes)
-->My installation is the following:
1. install Anaconda 
   Anaconda3-2020.11-Windows-x86_64
2. install Git
   Git-2.30.1-64-bit
3. clone GitHub repository
   In cmd, 
   "cd C:\Users\***\Desktop\Python\Qcodes_cloned"
   "git clone https://github.com/QCoDeS/Qcodes.git"
4. make a qcodes environment
   In Anaconda Prompt,
   "cd C:\Users\***\Desktop\Python\Qcodes_cloned\Qcodes"
   "conda env create -f .\environment.yml"
   wait a while and you find that the environment is created under
   "C:\Users\***\Anaconda3\envs".

> Which version of pip do you have installed pip --version
-->pip 20.2.4 from C:\Users\***\anaconda3\lib\site-packages\pip (python 3.8)

> Could it be that you have more than one version of python on the computer such that you install into one version and try to load from the other?
Probably yes... probably not...

If I "import sys; sys.path" on Jupyter without qcodes environment on the not working computer, I get 

['C:\\Users\\***',
 'C:\\Users\\***\\Desktop\\Python\\Qcodes_cloned\\Qcodes',
 'C:\\Users\\***\\anaconda3\\python38.zip',
 'C:\\Users\\***\\anaconda3\\DLLs',
 'C:\\Users\\***\\anaconda3\\lib',
 'C:\\Users\\***\\anaconda3',
 '',
 'C:\\Users\\***\\anaconda3\\lib\\site-packages',
 'C:\\Users\\***\\anaconda3\\lib\\site-packages\\win32',
 'C:\\Users\\***\\anaconda3\\lib\\site-packages\\win32\\lib',
 'C:\\Users\\***\\anaconda3\\lib\\site-packages\\Pythonwin',
 'C:\\Users\\***\\anaconda3\\lib\\site-packages\\IPython\\extensions',
 'C:\\Users\\***\\.ipython']
 
 .
 
 If I "import sys; sys.path" on Jupyter with qcodes environment on the not working computer, I get

['C:\\Users\\***',
 'C:\\Users\\***\\Desktop\\Python\\Qcodes_cloned\\Qcodes',
 'C:\\Users\\***\\anaconda3\\envs\\qcodes\\python37.zip',
 'C:\\Users\\***\\anaconda3\\envs\\qcodes\\DLLs',
 'C:\\Users\\***\\anaconda3\\envs\\qcodes\\lib',
 'C:\\Users\\***\\anaconda3\\envs\\qcodes',
 '',
 'C:\\Users\\***\\anaconda3\\envs\\qcodes\\lib\\site-packages',
 'C:\\Users\\***\\anaconda3\\envs\\qcodes\\lib\\site-packages\\win32',
 'C:\\Users\\***\\anaconda3\\envs\\qcodes\\lib\\site-packages\\win32\\lib',
 'C:\\Users\\***\\anaconda3\\envs\\qcodes\\lib\\site-packages\\Pythonwin',
 'C:\\Users\\***\\anaconda3\\envs\\qcodes\\lib\\site-packages\\IPython\\extensions',
 'C:\\Users\\***\\.ipython']
 
 .
 
 Does it mean I have Python 3.7 for qcodes and Python 3.8 installed together with Anaconda3.
 But I am wondering if it can be a problem because I get the same results with the working computer.
 
 
 If I "import sys; sys.path" on Jupyter without qcodes environment on the working computer, I get 
 
 ['C:\\Users\\*****\\Desktop\\Python\\analysis',
 'C:\\Users\\*****\\Anaconda3\\python38.zip',
 'C:\\Users\\*****\\Anaconda3\\DLLs',
 'C:\\Users\\*****\\Anaconda3\\lib',
 'C:\\Users\\*****\\Anaconda3',
 '',
 'C:\\Users\\*****\\Anaconda3\\lib\\site-packages',
 'C:\\Users\\*****\\Anaconda3\\lib\\site-packages\\win32',
 'C:\\Users\\*****\\Anaconda3\\lib\\site-packages\\win32\\lib',
 'C:\\Users\\*****\\Anaconda3\\lib\\site-packages\\Pythonwin',
 'C:\\Users\\*****\\Anaconda3\\lib\\site-packages\\IPython\\extensions',
 'C:\\Users\\*****\\.ipython']

 If I "import sys; sys.path" on Jupyter with qcodes environment on the not working computer, I get
 
['C:\\Users\\*****\\Desktop\\Python\\analysis',
 'C:\\Users\\*****\\Anaconda3\\envs\\qcodes\\python37.zip',
 'C:\\Users\\*****\\Anaconda3\\envs\\qcodes\\DLLs',
 'C:\\Users\\*****\\Anaconda3\\envs\\qcodes\\lib',
 'C:\\Users\\*****\\Anaconda3\\envs\\qcodes',
 '',
 'C:\\Users\\*****\\AppData\\Roaming\\Python\\Python37\\site-packages',
 'C:\\Users\\*****\\AppData\\Roaming\\Python\\Python37\\site-packages\\win32',
 'C:\\Users\\*****\\AppData\\Roaming\\Python\\Python37\\site-packages\\win32\\lib',
 'C:\\Users\\*****\\AppData\\Roaming\\Python\\Python37\\site-packages\\Pythonwin',
 'C:\\Users\\*****\\Anaconda3\\envs\\qcodes\\lib\\site-packages',
 'c:\\users\\*****\\desktop\\python\\qcodes_cloned\\qcodes',
 'C:\\Users\\*****\\Anaconda3\\envs\\qcodes\\lib\\site-packages\\win32',
 'C:\\Users\\*****\\Anaconda3\\envs\\qcodes\\lib\\site-packages\\win32\\lib',
 'C:\\Users\\*****\\Anaconda3\\envs\\qcodes\\lib\\site-packages\\Pythonwin',
 'C:\\Users\\*****\\Anaconda3\\envs\\qcodes\\lib\\site-packages\\IPython\\extensions',
 'C:\\Users\\*****\\.ipython']

jenshnielsen, 2021-03-12:
> 
    install Anaconda
    Anaconda3-2020.11-Windows-x86_64
    install Git
    Git-2.30.1-64-bit
    clone GitHub repository
    In cmd,
    "cd C:\Users\(Username)\Desktop\Python\Qcodes_cloned"
    "git clone https://github.com/QCoDeS/Qcodes.git"
    make a qcodes environment
    In Anaconda Prompt,
    "cd C:\Users\(Username)\Desktop\Python\Qcodes_cloned\Qcodes"
    "conda env create -f .\environment.yml"
    wait a while and you find that the environment is created under
    "C:\Users\(Username)\Anaconda3\envs".

Then you have not installed qcodes. You are missing the step where you actually install the cloned qcodes version. 

You should do 
```
pip install -e qcodes
```
To install qcodes from source after you have created the environment

erikawa-e, 2021-03-13:
Thank you...

pip install -e qcodes
ERROR: File "setup.py" not found. Directory cannot be installed in editable mode: 

I reinstalled Anaconda and qcodes. Then I even do not have C:\Users\(Username)\Anaconda3 folder...

Some problems with Anaconda installation now...

erikawa-e, 2021-03-13:
I could try 
conda install qcodes
But still not working...

erikawa-e, 2021-03-13:
After several trials of uninstallation and installation of Anaconda3,
finally
pip install qcodes
worked!!

Thank you so much for your help!!

jenshnielsen, 2021-03-15:
Great that it worked :)

Sorry it should have read:
`pip install -e .` from the directory containing setup.py

closed 2021-03-15 by jenshnielsen

erikawa-e, 2021-03-15:
Thanks. In any case, I did not have setup.py anywhere at that time due to the wrong installation.