microsoft/Qcodes #2824: wrong installation??

erikawa-e, 2021-03-15:
After this issue 
https://github.com/QCoDeS/Qcodes/issues/2815
is solved...

Sorry that I am still asking questions here...

Now I have a problem on which file is referred when qcodes are imported

On the "working" computer,
import qcodes.instrument_drivers.QDevil.QDevil_QDAC
qcodes.instrument_drivers.QDevil.QDevil_QDAC.__file__ 
gives
'c:\\users\\(Username)\\desktop\\python\\qcodes_cloned\\qcodes\\qcodes\\instrument_drivers\\QDevil\\QDevil_QDAC.py'

On the "previously not working"",
 'C:\\Users\\\(Username2)\\anaconda3\\envs\\qcodes\\lib\\site-packages\\qcodes\\instrument_drivers\\QDevil\\QDevil_QDAC.py'

Which one is correct?

Which one is installed wrongly now??

jenshnielsen, 2021-03-15:
Hi @erikawa-e 

This depends on how you install the package. 

If you do:
`pip install qcodes` or `pip install .` from the cloned qcodes source pip will copy qcodes into `site-packages` 

If you do `pip install -e .` pip will make a link to the cloned source code within the `site-packages` dir. You should see a file called `qcodes.egg-link` or something like that containing the path to your cloned source dir. 
This is known as an editable install and is really handy when you are developing the code as you don't need to reinstall every time you make a change

erikawa-e, 2021-03-15:
Oh I see..
Thank you. I am learning a lot :)

I guess I should reuninstall and reinstall qcodes again if I want to have a link to the cloned source code..
Is that correct?

jenshnielsen, 2021-03-15:
Yes. 

```
pip uninstall qcodes
pip install -e .
```
Where the last command needs to be executed from the cloned qcodes repo

erikawa-e, 2021-03-15:
I see. Thank you so much!!

closed 2021-03-15 by jenshnielsen