microsoft/Qcodes #2827: register_parameter cannot register parameters

erikawa-e, 2021-03-17:
Sorry for multiple questions again...

After uninstalling and reinstalling qcodes(https://github.com/QCoDeS/Qcodes/issues/2824),
now I think I have a link to the cloned source code.

However, now I cannot register parameter:
`measurement.register_parameter(qdac.ch02.v)`
gives 
ValueError: Can not register object of type <class 'qcodes.instrument.parameter.Parameter'>. Can only register a QCoDeS Parameter.

erikawa-e, 2021-03-17:
Somehow, after restarting computers, qcodes and all the other things, this error disappeared. 
I do not know what was the problem but sorry for this inconsiderate post.

jenshnielsen, 2021-03-17:
Hi @erikawa-e that does indeed look odd. Since the issue seems to have fixed it self I will close this. Please let us know if you see the issue again and we can have a look

closed 2021-03-17 by jenshnielsen

erikawa-e, 2021-03-17:
OK, thank you for your help!