microsoft/Qcodes #2831: validation of set value happens before applying set_parser

simonzihlmann, 2021-03-17:
I am not sure if this is a bug since I don't fully know what the set_parser function of a parameter is supposed to do. Therefore, a small example below.

### Steps to reproduce
Let's imagine we have a parameter class with `set_parser=int` and `vals=validators.Ints(min_value=0)`
```
class test_param(Parameter):
    def __init__(self, name):
        # only name is required
        super().__init__(name, label='test_parameter',
                         vals=validators.Ints(min_value=0),
                        get_parser = int,
                        set_parser = int)
        self._count = 0

    # you must provide a get method, a set method, or both.
    def get_raw(self):
        self._count += 1
        return self._count

    def set_raw(self, val):
        self._count = val
```
and that we create a parameter:

`test = test_param('t1')`

### Expected behaviour when setting an integer value, like `test(2)`
setting works as expected

### Expected behaviour when setting a float value, like `test(2.0)`
Same behaviour as for `test(2)` since the set_parsesr should ensure that 2.0 is cast into 2.

### Actual behaviour when setting a float vlaue, like 'test(2.0)'

An error is produced since the validator is not happy.
```
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-11-9027be1cd919> in <module>
----> 1 test(2.0)

~\AppData\Roaming\Python\Python38\site-packages\qcodes\instrument\parameter.py in __call__(self, *args, **kwargs)
    403         else:
    404             if self.settable:
--> 405                 self.set(*args, **kwargs)
    406                 return None
    407             else:

~\AppData\Roaming\Python\Python38\site-packages\qcodes\instrument\parameter.py in set_wrapper(value, **kwargs)
    643             except Exception as e:
    644                 e.args = e.args + (f'setting {self} to {value}',)
--> 645                 raise e
    646 
    647         return set_wrapper

~\AppData\Roaming\Python\Python38\site-packages\qcodes\instrument\parameter.py in set_wrapper(value, **kwargs)
    603                                     " that is not settable.")
    604 
--> 605                 self.validate(value)
    606 
    607                 # In some cases intermediate sweep values must be used.

~\AppData\Roaming\Python\Python38\site-packages\qcodes\instrument\parameter.py in validate(self, value)
    702             context = self.name
    703         if self.vals is not None:
--> 704             self.vals.validate(value, 'Parameter: ' + context)
    705 
    706     @property

~\AppData\Roaming\Python\Python38\site-packages\qcodes\utils\validators.py in validate(self, value, context)
    341         """
    342         if not isinstance(value, self.validtypes):
--> 343             raise TypeError(
    344                 '{} is not an int; {}'.format(repr(value), context))
    345 

TypeError: ('2.0 is not an int; Parameter: t1', 'setting t1 to 2.0')
```

I know that it is a bit a weird example. But I came across it when trying to use the built-in do1d function to sweep an integer parameter of an instrument (like number of points of a VNA trace)

As said in the beginning, I am not fully sure what to expect.

jenshnielsen, 2021-03-19:
Hmn, this is a good question. I can see it going both ways.

On one hand is it dangerous to pass a float to a parameter that expects an int. Its fine in the case of 2.0 but what if one passes 2.5 ? 

On the other hand I do see your issue trying to sweep over this using dond as a limitation. Perhaps it would be better to open dond slightly so that one can pass a list or generator of numbers in rather than generating them

simonzihlmann, 2021-03-23:
I agree that it is not clear what should happen if you try to pass 2.5 to a parameters that expects an integer.

However, my point is rather about the set_parser. What is it used for in general?

Concerning the doNd, I am not so sure how often people would like to sweep an integer value in reality. As mentioned above, the example is not so meaningful.

Akshita07, 2021-05-21:
Hello @simonzihlmann , I am here to try to answer your question `What is set_parser used for in general?` 

`set_parser` allows you to set a function to transform the input set value to an encoded value sent to the instrument. 
Let's take a hypothetical case, without involving an instrument first. Say there are 15 steps in a process, and those can be categorized as 1-9 being initialization, 10-14 being process ongoing and finally 15th is closing. And you want to have a parameter to know which category the process is on and not exactly the step number. This kind of parameter can be written very easily with a set_parser without creating a cumbersome val_mapping.

```python

def set_state(val: int) -> str:
    if val in range(1,10):
        return 'initializing'
    elif val in range(10,15):
        return 'process ongoing'
    else:
        return 'closed'
    
p = Parameter(name='test',
             set_parser=set_state,
             vals=Ints(min_value=1, max_value=15),
             set_cmd=None)

```

Within the process code this parameter can be set with integer values and when you want to know weather process is initializing, ongoing or closed you can call this param to check.

This analogy can be extended, when an instrument needs to be set to some particular states for multiple different input values. I can't think of a very concrete example involving as instrument right away but I hope you get a gist of what are the possibilities involving `set_parser`.

simonzihlmann, 2021-06-07:
Dear @Akshita07, thanks for the explanation. I see what it is used for thanks to your example. Then it is very similar to a val_mapping?
I think one can close this issue.

Akshita07, 2021-06-08:
Hello @simonzihlmann,

No set_parser is not really similar to val_mapping. Function of set_parser is to perform parsing for the input set values while that of val_mapping is to map the values which either comes from the getting or during setting of the Parameter.

Please feel free to close the issue, if you are satisfied.

simonzihlmann, 2021-08-12:
Thanks @Akshita07 for the clarification. I clos this issue.