microsoft/Qcodes #2855: Deprecate/remove UHFLI driver example based on old qcodes driver

ThorvaldLarsen, 2021-03-26:
Qcodes driver example for UHFLI is based on old qcodes driver (and qcodes loop measure functions) rather than the new ZI provided driver. [Link to example](https://qcodes.github.io/Qcodes/examples/driver_examples/Qcodes%20example%20with%20ZI%20UHF-LI.html)
Very minor issue but confused me for quite some time as I missed the deprecation warning on import of the qcodes driver and could figure why some functions didnt match other examples I had.

Maybe worth adding a large warning in the top of the example? Or maybe completely remove it so this is not the first thing that shows up when googling: "qcodes example UHFLI".

FarBo, 2021-04-06:
Since the driver is still not removed and only deprecated, I think the example should stay. On the other hand, this example is also using old QCoDeS Loop. This makes it a little complicated because one should sit and correct the example for a deprecated driver :)
I think adding a note (saying this example and its corresponding driver are deprecated in favor of zhinst implementations) in the begining of the notebook is a better solution.
I can do that if we agree on it.

jenshnielsen, 2021-04-06:
@FarBo I think remove the example is fine. No need to document a deprecated driver. Furthermore the driver has already been deprecated for a while by now

FarBo, 2021-04-06:
@jenshnielsen 

HMM. That's also true. Then we can proceed with the axing path
I will do that if there is no objection from anyone else :)

PR #2893 (2021-04-06): Removed ZI UHFLI notebook
As the driver is deprectaed and ZI has its own driver and notebook for this instrument, This notebook is removed in this PR.

closes #2855
closed 2021-04-06 by jenshnielsen