microsoft/Qcodes #2908: Disable logging on import for Sphinx documentation builds

ovaar, 2021-04-09:
### Steps to reproduce
1. Create a project with sphinx documentation
2. Add a python notebook to your documentation
3. Import qcodes inside the python notebook.
4. Build sphinx documentation
5. > now conditional import of logging is run.

### Expected behaviour

When Qcodes is imported in a python notebook which is build by sphinx, qcodes conditionally start logging.
The function `logger.running_in_test_or_tool` is evaluated after running `logger.start_logging_on_import` raising an exception.

### Actual behaviour

The following error is raised.
```
File "d:\git\qcodes\qcodes\logger\logger.py", line 365, in conditionally_start_all_logging
    if not running_in_test_or_tool() and start_logging_on_import():
File "d:\git\qcodes\qcodes\logger\logger.py", line 339, in start_logging_on_import
    config = qc.config
AttributeError: module 'qcodes' has no attribute 'config'
```

### System
**operating system**
Windows / Linux

**qcodes branch**
master

**qcodes commit**
f688fe3f1f22ce72e4b0eb0ee359cd7b807f34d9

astafan8, 2021-04-09:
do i understand correctly that "import qcodes" does not instantite "qcodes.config" when "Qcodes is imported in a python notebook which is build by sphinx"? this is very unexpected, qcodes' own docs have many executable notebooks that we build docs from and they all import qcodes, and that happens in CI (so, not in a "tool"), and some of the notebooks access qcodes.config and all works. I wonder what the difference is in our CI and docs building as compared to your project where you import qcodes...

ovaar, 2021-04-09:
It seems that this issue is related to the usage of `sphinx-autodoc-typehints` in combination with `set_type_checking_flag = True` [source](https://github.com/agronholm/sphinx-autodoc-typehints/blob/master/sphinx_autodoc_typehints.py#L479). Somehow it seems that enabling static type checking messes up populating the global `qcodes.config` parameter. I cannot explain why this would happen.

The error originates from building the following docs [page](https://quantify-quantify-scheduler.readthedocs-hosted.com/en/latest/tutorials/Tutorial%201.%20basic%20experiment.html#compilation-of-pulses-onto-physical-hardware), which shows the basic usage of Qcodes.

jenshnielsen, 2021-04-09:
That seems likely. I think its a good fix to not start the logging when running within sphinx but I do suspect that you are likely to run into a similar issue if your notebook are trying to execute qcodes functions as there are plenty of other places in qcodes that rely on the configuration singleton being instantiated

jenshnielsen, 2021-04-12:
@ovaar it would be great it you want to add a line or two about Quantify to https://github.com/QCoDeS/Qcodes/blob/master/docs/community/related_projects.rst

PR #2909 (2021-04-12): fix(logger): Disable conditionally start logging for Sphinx documenta…
Disable conditionally start logging for Sphinx documentation. Solves #2908

Fixes #2908.

Changes proposed in this pull request:
- Swapped `start_logging_on_import` and `running_in_test_or_tool` so that the tools are evaluated before starting logging on import.
- Added sphinx main to the list of tools.

**Edit** This is not really a proper fix for the issue that seems to be lying elsewhere but a work around that should enable users to build documentation in certain configurations.
closed 2021-04-12 by jenshnielsen