microsoft/Qcodes #2922: Dataset with nested metadata throws TypeError when exporting to NetCDF file

guenp, 2021-04-13:
### Steps to reproduce
1. Prepare a Dataset
2. Add metadata to the dataset where the value is a `dict`
3. Export the dataset to netCDF

### Expected behaviour
The data is exported to netCDF where values that are `dict`s are either skipped or converted to `str`

### Actual behaviour
```python
TypeError                                 Traceback (most recent call last)
<ipython-input-155-35b4f9ff4165> in <module>
----> 1 ds.export(export_type="netcdf", path=output_dir, prefix=file_prefix)

~/repos/Qcodes/qcodes/dataset/data_set.py in export(self, export_type, path, prefix)
   1711             )
   1712 
-> 1713         self._export_path = self._export_data(
   1714             export_type=export_type,
   1715             path=path,

~/repos/Qcodes/qcodes/dataset/data_set.py in _export_data(self, export_type, path, prefix)
   1675             file_name = self._export_file_name(
   1676                 prefix=prefix, export_type=DataExportType.NETCDF)
-> 1677             return self._export_as_netcdf(path=path, file_name=file_name)
   1678 
   1679         elif DataExportType.CSV == export_type:

~/repos/Qcodes/qcodes/dataset/data_set.py in _export_as_netcdf(self, path, file_name)
   1642         file_path = os.path.join(path, file_name)
   1643         xarr_dataset = self.to_xarray_dataset()
-> 1644         xarr_dataset.to_netcdf(path=file_path)
   1645         return path
   1646 

~/anaconda3/envs/bringup/lib/python3.9/site-packages/xarray/core/dataset.py in to_netcdf(self, path, mode, format, group, engine, encoding, unlimited_dims, compute, invalid_netcdf)
   1687         from ..backends.api import to_netcdf
   1688 
-> 1689         return to_netcdf(
   1690             self,
   1691             path,

~/anaconda3/envs/bringup/lib/python3.9/site-packages/xarray/backends/api.py in to_netcdf(dataset, path_or_file, mode, format, group, engine, encoding, unlimited_dims, compute, multifile, invalid_netcdf)
   1057     # validate Dataset keys, DataArray names, and attr keys/values
   1058     _validate_dataset_names(dataset)
-> 1059     _validate_attrs(dataset)
   1060 
   1061     try:

~/anaconda3/envs/bringup/lib/python3.9/site-packages/xarray/backends/api.py in _validate_attrs(dataset)
    229     # Check attrs on the dataset itself
    230     for k, v in dataset.attrs.items():
--> 231         check_attr(k, v)
    232 
    233     # Check attrs on each variable within the dataset

~/anaconda3/envs/bringup/lib/python3.9/site-packages/xarray/backends/api.py in check_attr(name, value)
    220 
    221         if not isinstance(value, (str, Number, np.ndarray, np.number, list, tuple)):
--> 222             raise TypeError(
    223                 f"Invalid value for attr {name!r}: {value!r} must be a number, "
    224                 "a string, an ndarray or a list/tuple of "

TypeError: Invalid value for attr 'extra_metadata': {
```

### System
**operating system**
Ubuntu 20.04.1 LTS on WSL2
**qcodes branch**
`master`

Tagging @jenshnielsen. What would you recommend as a best solution here - drop the nested data or convert it to `str`? We currently convert the snapshot to `str` as well so to be consistent this may be the best way to go.

jenshnielsen, 2021-04-14:
@guenp Yes similar to the snapshot I think we should serialize the extra metadata to json. netcdf4 is very limited in the types allowed in the attributes dict

guenp, 2021-04-14:
@jenshnielsen That would be great, thanks! I've implemented a quick-and-dirty fix for this such that I was able to upload my data but a more sustainable solution would be great :) The main request I have is that we add some functionality to load the data from netCDF file as well and deserialize the data back to a nested dictionary.

Another issue I ran into is that the serialized nested dicts are very verbose and take up multiple pages when using the __repr__ functionality in a jupyter notebook. I posted a question on this in the xarray gitter: https://gitter.im/pydata/xarray

jenshnielsen, 2021-04-15:
Hi @guenp looking more closely at this I am a bit confused as to how you produced this dataset.
If I do the following 

```
dataset1D.add_metadata("sometag", {1:1})
```
on some dataset I will get an unsupported type error from sqlite

```
2021-04-15 15:30:23,942 ¦ qcodes.dataset.sqlite.connection ¦ ERROR ¦ connection ¦ atomic ¦ 106 ¦ Rolling back due to unhandled exception
Traceback (most recent call last):
  File "c:\users\jens-work\source\repos\qcodes\qcodes\dataset\sqlite\connection.py", line 103, in atomic
    yield conn
  File "c:\users\jens-work\source\repos\qcodes\qcodes\dataset\sqlite\connection.py", line 159, in atomic_transaction
    c = transaction(atomic_conn, sql, *args)
  File "c:\users\jens-work\source\repos\qcodes\qcodes\dataset\sqlite\connection.py", line 134, in transaction
    c.execute(sql, args)
sqlite3.InterfaceError: Error binding parameter 0 - probably unsupported type.
2021-04-15 15:30:23,951 ¦ qcodes.dataset.sqlite.connection ¦ ERROR ¦ connection ¦ atomic ¦ 106 ¦ Rolling back due to unhandled exception
Traceback (most recent call last):
  File "c:\users\jens-work\source\repos\qcodes\qcodes\dataset\sqlite\connection.py", line 103, in atomic
    yield conn
  File "c:\users\jens-work\source\repos\qcodes\qcodes\dataset\sqlite\connection.py", line 159, in atomic_transaction
    c = transaction(atomic_conn, sql, *args)
  File "c:\users\jens-work\source\repos\qcodes\qcodes\dataset\sqlite\connection.py", line 134, in transaction
    c.execute(sql, args)
sqlite3.InterfaceError: Error binding parameter 0 - probably unsupported type.
```

So I would not expect it to be possible to produce a dataset with nested metadata. I guess I must be overlooking something?

guenp, 2021-04-15:
@jenshnielsen Ah, I think you're right. This dataset was produced with Nanotune using `json.dumps`: , see [nanotune/tests/data_savers.py#L33](https://github.com/microsoft/nanotune/blob/3d25fa29330fea9a0a9ee09ce9eada94178517af/nanotune/tests/data_savers.py#L33).

So the problem is that when qcodes exports the dataset to xarray, it stores the whole `DataSet._metadata` dict is nested under the `"extra_metadata"` key in `.attrs` rather than storing them by key/value pair.

Here's a minimal example to reproduce:
```python
from qcodes import Measurement
import json
import numpy as np

meas = Measurement()
meas.register_custom_parameter(
    "voltage", paramtype="numeric", label="voltage x", unit="V"
)
meas.register_custom_parameter(
    "current",
    paramtype="numeric",
    label="dc current",
    unit="A",
    setpoints=("voltage",),
)

voltage, current = np.random.rand(), np.random.rand()
with meas.run() as datasaver:
    datasaver.add_result(("voltage", voltage), ("current", current))
    datasaver.dataset.add_metadata("my_param1", 42)
    datasaver.dataset.add_metadata("my_param2", "other")

    nt_metadata = {
        "foo": {
            "bar": {
                "baz": "test"
            },
            "spam": [1, 2, 3],
        }
    }
    datasaver.dataset.add_metadata("foo_metadata", json.dumps(nt_metadata))

datasaver.dataset.export(export_type="netcdf", path=".", prefix="export_test_")
```

this raises the above error:
```
TypeError: Invalid value for attr 'extra_metadata': {'my_param1': 42, 'my_param2': 'other', 'foo_metadata': '{"foo": {"bar": {"baz": "test"}, "spam": [1, 2, 3]}}'} must be a number, a string, an ndarray or a list/tuple of numbers/strings for serialization to netCDF files
```

jenshnielsen, 2021-04-16:
@guenp Thanks, that we can write a test for and fix :)

PR #2934 (2021-04-20): Don't write extra metadata for xarray in a nested dict 
Since this prevents export to netcdf

Fixes #2922
closed 2021-04-20 by jenshnielsen