microsoft/Qcodes #2995: Plot_dataset for partially completed 2D

ThorvaldLarsen, 2021-05-11:
In one of the latest updates the plotting functions doesnt handle partially completed 2D dataset. It now falls back to a scatter plot instead of a partially filled map.
Ideally we would like to keep the 0.22.0 version behavior for partially completed plots.

Version 0.25.0
[image]

Version 0.22.0
[image]

jenshnielsen, 2021-05-18:
This is caused by the dataset now being plotted from the cache which uses nan as a filler for the empty elements rather than simply omitting them. Adding something like `dataarray = dataarray[~np.isnan(dataarray)]` to `flatten_1D_data_for_plot` seems to fix this but it is unclear if this will have problematic issues for datasets that really contain nans

jenshnielsen, 2021-05-18:
A better solution would therefor be to make use of the actual shape data if available

PR #3024 (2021-05-25): Improved plotting of pre-shaped 2d data 
After the predefined shape was added to the dataset non measured data is filled with nans. This updates the 2d pcolormesh plot to handle this. 

It does this by bypassing the logic that checks for shape since this is no longer needed. However Matplotlib cannot handle a pcolormesh plot where the x and y coordinates are nans so some logic is needed to ensure this is clipped correctly. There are two use cases to consider. If the data is on a rectilinear grid we can plot all rows and columns that contain non nan setpoints. For the other case of non rectilin [... cut]
closed 2021-05-25 by trevormorgan