microsoft/Qcodes #3054: N52xx.py driver bug

alebrunoeth, 2021-05-26:
### Steps to reproduce
1.Run python code from notebook and sweep over around 12500 * n points or more (n integer).
2.Wait for measurements.
3.Plot measurement results and analyse trace result.

### Expected behaviour
Result is just the expected trace response.

### Actual behaviour
Every (approximately) 12500 points, an outlier occurs composed of four points (with amplitude respectively close to 1e-45, 2, 1e-43, 1e-7). This is not from the physical measurement and we would like to not use any functions to delete this manually. (In our case the first points are number 12405 to 12408)
### System
**operating system**
Windows 10 64-bit
**qcodes version**
0.23.0

[PNABase_bug.txt](https://github.com/QCoDeS/Qcodes/files/6546965/PNABase_bug.txt)

astafan8, 2021-05-26:
@alebrunoeth do you mind attaching a plot with that data and those outliers (as an image)? for a visual aid

alebrunoeth, 2021-05-26:
dB plot over whole spectrum.
[image]
zoomed plot in absolute value
[image]

alebrunoeth, 2021-05-26:
dB plot using 60000 points (4 features appear)

[image]

jenshnielsen, 2021-05-26:
@spauka I think you originally wrote this driver. Have you ever seen anything like this? 

@alebrunoeth before doing anything else I would suggest that you ensure that the instrument is updated to the latest firmware version. It seems at least possible that this is an issue with the instrument firmware

alebrunoeth, 2021-05-27:
I already did update the firmware

jenshnielsen, 2021-05-27:
@alebrunoeth thanks for the update unfortunately I don't think the fact that the issue is not present in the gui can be used to establish where the issue is. The driver is using `CALC:DATA? FDATA` to read back the data from the instrument. It will require some debugging to establish why the data contains some wrong samples but it seems most likely that those datapoints are already present in the data read back at this stage

alebrunoeth, 2021-05-27:
Thank you, I will try and debug and try to see if the problem arises while using VisaImplementation as well

spauka, 2021-05-27:
@jenshnielsen @alebrunoeth I have indeed seen this same issue - although it was on an Infinium scope and not on a PNA. The issue arose when we tried to transfer >100000 pts from the scope in a single measurement. It seemed like the instrument was inserting some random characters in the middle of the data.

I'm just asking around to ask if we had a better solution than to break the measurement down into multiple smaller acquisitions 😆

spauka, 2021-12-15:
Hi @alebrunoeth - just wondering if you're still seeing this issue? I'm not able to reproduce it any more on my scope of DSO so it may have been fixed somewhere?

spauka, 2023-06-02:
@jenshnielsen I'm regularly taking 50001 point traces on the PNA and not seeing this issue. I suggest we can close it?

jenshnielsen, 2023-06-02:
@spauka agreed @alebrunoeth if you still can reproduce please let us know and we can reopen

closed 2023-06-02 by jenshnielsen