microsoft/Qcodes #3109: Git folder weighting more than 99.3% of the repo size

edumur, 2021-06-10:
Hi all,

I recently wondered why the repo size was so big.
First, let me say that the repo size is 3.6GB which is huge for waht we have in it (mainly text files).

I ran WinDirStat to get some statistics on the repo and I found this:

[image]

Is there any reason to keep the .git folder so big?
After running the following command:
```bash
git repack -a -d --depth=250 --window=250
git prune
git gc --aggressive
```
I got the following stat:
[image]

It saved 0.8GB which is better but not perfect...

Well at the end, I guess I just wanted to raise the question of the size of the repo which I find quite big ^^ , and maybe come up with a solution.

### System
**operating system** Windows 10

jenshnielsen, 2021-06-10:
@edumur it is indeed unfortunate that the repo is as big as it is. I did some stats at some point and it is mainly due to ipy notebooks and the images therein. However, the .git folder contains all the revisions of the code that has ever been created so it is expected to be big compared to the checkout code. It is not really possible to shrink the repository without rewriting the history and chancing every single git commit since the start. In the future we will be more strict about the number of notebooks and their revisions to try to stop the growth

edumur, 2021-06-11:
Thanks for this reply.
Do you think it is possible to have some files, maybe notebook, which do not exist today that we can erase from the git history to save some space?

jenshnielsen, 2021-06-11:
Not really, you cannot remove a file from the git history unless you are willing to rewrite the entire history changing every single commit hash and that is a very destructive thing to do

edumur, 2021-06-11:
I see...
Well, as you said, this is unfortunate ^^" .
You can close the issue or leave it opened, as you wish.

Thanks again for you reply and the work you do on this lib ;) .

jenshnielsen, 2021-06-14:
After some investigation it turns out that this was largely due to the git history being retained in the gh-pages branch that holds the docs. After deleting that a fresh clone of qcodes is ~140 MB for me.

To get the saving locally you will probably have to:
* Fetch the origin.
* If your own fork or any other fork that you have added as a remove contains a gh-pages branch you may also have to delete the branch from that fork
* Run the above commands to perform gc etc

Or alternatively checkout from a fresh clone

edumur, 2021-06-14:
~140MB, that's so cool.
I will try right away, thanks!

jenshnielsen, 2021-06-14:
closing since the size is now down to a more reasonable level. Let me know if there are other issues

closed 2021-06-14 by jenshnielsen