microsoft/Qcodes #3139: Installing issue

peitian1020, 2021-06-21:
If you encounter a bug use the following template.
If you have a feature request feel free to freestyle.
### Steps to reproduce
1. Install Qcodes
2. Install Spyder
3. Launch failed

### Expected behaviour
Install qcodes by following the instruction, and spyder(qcodes) should appear in start menu

### Actual behaviour
spyder(qcodes) is not automatically installed, I install it in anaconda navigator. But errors occur when I run spyder

### System
**operating system**
Windows
**qcodes branch**
Latest
**qcodes commit**

FarBo, 2021-06-30:
There shouldn't be a problem installing spyder.
qcodes does not install spyder by default and the user needs to install it to the qcodes environment using pip or conda.

So when you create a conda environement for qcodes, e.g. conda create -n qcodes python=3.7 (or later)
Then let's say you are using pip to install qcodes:
$ conda activate qcodes
$ pip install qcodes

Then on the same environment:
$ pip install spyder

Finally run the spyder after the installation cmpletion:
$ spyder

This should open spyder with qcodes kernel

Let me know if the above steps work on your side and if you see error, please share us the error. I am gonna update our qcodes documentation to clarify installing spyder.

FarBo, 2021-07-05:
The documentation is update in PR #3179

closed 2021-07-06 by jenshnielsen