microsoft/Qcodes #3151: Setting R&S ZNB npts fails in CW mode

mgunyho, 2021-06-24:
### Steps to reproduce
1. Load R&S ZNB driver
2. set it to CW mode
3. try to change the `npts` parameter

MWE:

```py
import qcodes as qc
import qcodes.instrument_drivers.rohde_schwarz.ZNB as ZNB

qc.Instrument.close_all()

station = qc.Station()

station.add_component(ZNB.ZNB('VNA', ..., init_s_params=False))
station.VNA.add_channel('S21')

vna = station.VNA

# this works fine
vna.S21.npts(123)
print(vna.S21.npts())

# this causes a timeout error
vna.S21.setup_cw_sweep()
vna.S21.npts(201)
print(vna.S21.npts())

# this works again
vna.S21.setup_lin_sweep()
vna.S21.npts(345)
print(vna.S21.npts())
```

### Expected behaviour
The `npts` parameter should be set correctly.

### Actual behaviour
I get the exception:
```
VisaIOError: ('VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.', "asking 'SENS1:FREQ:START?' to <ZNB: VNA>", 'getting VNA_S21_start', 'setting VNA_S21_npts to 201')
```

The problem seems to be that setting `npts` calls `self.update_lin_traces()` which in turn calls `self.start.get()` which doesn't seem to work if the VNA is in CW mode.

### System
**operating system**
Windows 7

**qcodes branch**
`v0.26.0`

jenshnielsen, 2021-06-30:
@mgunyho Thanks for the detailed description. It looks like perhaps update_lin_traces should only be called if the instrument is in a different mode. Would you be willing to make a pull request that implements this ?

mgunyho, 2021-06-30:
Yes, that seems to be the correct fix. I can try to implement it the next time we're using the VNA.

PR #3168 (2021-07-02): Fix R&S ZNB setting npts in CW mode
Fixes #3151.

I branched off of v0.26, let me know if you want me to rebase onto master.

@jenshnielsen
closed 2021-07-02 by jenshnielsen