microsoft/Qcodes #3181: HP3245A driver issue

DorianFraudet, 2021-07-06:
### _Steps to reproduce_
1. Instrument driver in attachment
2. Run the following script to import the instrument and add it to station 
```
[HP3245A.txt](https://github.com/QCoDeS/Qcodes/files/6768972/HP3245A.txt)

import qcodes as qc
from qcodes.logger.logger import start_all_logging

from HP3245A import HP3245A

I_source = HP3245A('current_source', 'GPIB::24')

start_all_logging()

# Create a station
station = qc.Station()
station.add_component(I_source)

station.snapshot()
station.components
```

### _Expected behaviour_
Connect to the instrument and print the identity of the instrument

### _Actual behaviour_
Wait for ~ 15 s and print : 
`Connected to : None current_source (serial:None, firmware:None) in 16.95s`
The connection should be way faster and the identity of the instrument is not found.

### _System_
**Windows 7 professional**

### _Qcodes version_
**0.21.0**

jenshnielsen, 2021-07-06:
How did you find this driver? As far as I can see it is not part of any qcodes release nor a release of qcodes contrib drivers

DorianFraudet, 2021-07-06:
I did this driver, but I put an issue since I think this bug is not necessarily coming from the driver itself (I can control the current source) but most likely from the initialization of the VisaInstrument base class ? It seems that in the end it connects to the instrument but can't read the infos of the intrument (name,...) and loose a lot of time trying to query it.

jenshnielsen, 2021-07-06:
Ah ok. 

With respect to the identification of the instrument. The instrument base clas driver implements get_idn. This however, requires the instrument to return a standard formatted string when you ask with `*IDN?` not all instruments does this. 
If the HP3245A does not do this you will need to overwrite get_idn to implement a way that matches this instrument. 

With respect to the slowdown. Does this happen when you instantiate the instrument class e.g. this line `
I_source = HP3245A('current_source', 'GPIB::24')` or when adding it to the station. Adding an instrument to the station will get all parameters that are not marked as snapshot_get =  false  so depending on the instrument that can take some time

DorianFraudet, 2021-07-09:
Hey, 

I owerwrote the get_idn function and it's now working fine. Thanks for your help :)

closed 2021-07-09 by DorianFraudet