microsoft/Qcodes #3220: Not compliant *IDN?

supersibas11, 2021-07-20:
Hi all,

I 'm starting to implement a driver for a TTR506 VNA from Tektronix. The *IDN? command returns me:

'TTR506A,Tektronix,B010965,Application version 2.0.1.4ead5a2e Firmware version 066185400'

Which means it is not compliant with #107 (https://github.com/QCoDeS/Qcodes/pull/107), so I suppose it would be problematic.

Do you have any ideas to solve it?

Thank you for your help.

jenshnielsen, 2021-07-20:
To handle this you will probably need to overwrite `get_idn` defined here https://github.com/QCoDeS/Qcodes/blob/master/qcodes/instrument/base.py#L466 to unpack the data format as given in your instrument. I think this will be sufficient but let me know if you have more questions.

supersibas11, 2021-07-20:
Thank you, 

I was thinking about that but I was not 100% sure. I'll try it.

closed 2021-07-20 by supersibas11