microsoft/Qcodes #3271: How can we update Qcodes?

erikawa-e, 2021-08-17:
I am sorry for this stupid question.
I am new both to Qcodes and github.

After I installed Qcodes on the local computer, how can I update it?

I do not even know how to update it fully to the latest version but is it possible to update only some parts?

https://github.com/QCoDeS/Qcodes_contrib_drivers/releases

For example, I am interested in v0.9.0 because I would like to have the M5180 driver.
What I do now is to copy and paste M5180 folder manually.
Is there any more sophisticated way to do?

Another concern is if I update the local Qcodes to the latest version, will I lose the modifications that I made locally?

Thank you in advance.

jenshnielsen, 2021-08-17:
Hi @erikawa-e 

QCoDeS and qcodes-contrib-drivers (which I guess is what you are referring to ? ) are both releases as regular python packages. 
As such the things that apply to most python packages will also apply to qcodes. 

It is not possible to update only some parts as everything in a package needs to be tested together. You can of cause copy files between different versions but that is very likely to cause problems down the line. We always recommend users to use the latest version. 

If you have local modifications I would encourage you to contribute them to the official version so that they can become part of the next release. 

We recommend that users install qcodes using pip or conda and does not install the packages locally from a git clone. That way you are getting official releases that have been tested and you can update them with pip and conda. 

If you do have local modifications and you need to update you will need to reapply them to the updated version. This can be done either manually or using git merge or rebase but is in any case a fair amount of work, which is why we would suggest that you contribute your fixes to the official version.

The qcodes docs on installation here https://qcodes.github.io/Qcodes/start/index.html may be of some use

Hope this helps in making it more clear.

erikawa-e, 2021-08-17:
Dear jenshnielsen,

Thank you so much for your kind instruction!
It is clear now.

closed 2021-08-17 by erikawa-e