microsoft/Qcodes #3653: Getting a DelegateParameter with scale fails if it is not initialized

haroldmeerwaldt, 2021-12-07:
If you create a DelegateParameter and pass a scale argument, it is not possible to get its cache before it is initialized. The scale gets applied to the None that resides in the source parameter and gives a TypeError.

### Steps to reproduce
```python
from qcodes import Parameter, DelegateParameter

p = Parameter("p")
dp = DelegateParameter("dp", p, scale=1.1)

print(dp.source.cache.get(False))  # gives back None
print(dp.cache.get(False))  # fails
```

### Expected behaviour
I expected to get None back, just as you would a plain uninitialized Parameter.

A workaround is to give nan as an initial cache value. This works:
```python
from qcodes import Parameter, DelegateParameter
import numpy as np

p = Parameter("p")
dp = DelegateParameter("dp", p, scale=1.1, initial_cache_value=np.nan)

print(dp.source.cache.get(False))
print(dp.cache.get(False))
```

### Actual behaviour
I got a TypeError:
```python
Traceback (most recent call last):

  File "D:\users\hw_sw_integration\projects\low-level-software\src\lls_input\amber\scripts\example_02.py", line 14, in <module>
    print(dp.cache.get(False))

  File "D:\users\hw_sw_integration\venv-3.9\lib\site-packages\qcodes\instrument\parameter.py", line 1541, in get
    return self._parameter._from_raw_value_to_value(

  File "D:\users\hw_sw_integration\venv-3.9\lib\site-packages\qcodes\instrument\parameter.py", line 601, in _from_raw_value_to_value
    value = value / self.scale

TypeError: unsupported operand type(s) for /: 'NoneType' and 'float'
```

It appears that `_from_raw_value_to_raw_value` applies the scale on the value even if it's None. My solution would be to put a big `if value is not None:` in the method.

### System
Windows 10
qcodes 0.30.1

astafan8, 2021-12-08:
@haroldmeerwaldt thank you for reporting the issue!

i think your suggestion is good! but let's make the value None check only for offset and the scale, so on line https://github.com/QCoDeS/Qcodes/blob/c66a8af4ac8fa21498849a24734e4e1749952304/qcodes/instrument/parameter.py#L582 and https://github.com/QCoDeS/Qcodes/blob/c66a8af4ac8fa21498849a24734e4e1749952304/qcodes/instrument/parameter.py#L598.

Would you be willing to start a PR? and also add a few tests for this in this file https://github.com/QCoDeS/Qcodes/blob/master/qcodes/tests/parameter/test_parameter_scale_offset.py ?

haroldmeerwaldt, 2021-12-09:
@astafan8 That sounds like a good plan! Allowing None for the parser and in the inverse_val_mapping dictionary makes sense. I'll start a PR with tests.

commit 7e2e059e (2021-12-14): Merge #3662

3662: Allow getting an uninitialized DelegateParameter with offset or scale r=jenshnielsen a=haroldmeerwaldt

- [x] do not apply offset or scale when getting delegate parameter and source is None (in `_from_raw_value_to_value`)
- [x] added two tests

Fixes #3653
PR #3662 (2021-12-14): Allow getting an uninitialized DelegateParameter with offset or scale
- [x] do not apply offset or scale when getting delegate parameter and source is None (in `_from_raw_value_to_value`)
- [x] added two tests

Fixes #3653
closed 2021-12-14 by bors[bot]