microsoft/Qcodes #3686: Remove pinned versions from {test_,}requirements.txt & environment.yml

spauka, 2021-12-15:
At the moment, test_requirements.txt and requirements.txt have a set of pinned dependancies that are causing a lot of noise from dependabot. This also makes PRs annoying to maintain as small frequent changes to the master branch updating requirements.txt come in.

The requirements in the requirements.txt file are also not a minimal set of requirements, as is defined in setup.cfg.

I propose the following:
 - Similar to the [numpy](https://github.com/numpy/numpy) package, don't maintain a requirements.txt file for the base. Use the base requirements in setup.cfg or environment.yml as source of requirements.
 - Where necessary to create a test environment, remove pinned versions from test_requirements.txt
 - Remove pinned versions from `environment.yml`. The versions of python, jupyter and numpy in this file are getting quite old.

This should also relax a bit of maintenance burden as maintainers do not have to manually bump versions as packages are updated. Since many users will install from pypi or conda, this will also ensure that as packages are updated, maintainers are alerted to potential issues.

Happy to implement this in a PR if maintainers agree that this a good way forward? 

@jenshnielsen

jenshnielsen, 2021-12-15:
> At the moment, test_requirements.txt and requirements.txt have a set of pinned dependancies that are causing a lot of noise from dependabot.

It is true that this creates some noise however as I will try to argue below I think the benefits significantly out way the negatives. 

> This also makes PRs annoying to maintain as small frequent changes to the master branch updating requirements.txt come in.

I am curious how? As a pr maintainer I would expect you to do exactly zero with relation to this unless you yourself upgrade dependencies. If you just mean keeping the the branch up to date with master then there is no need to do that. We ensure that when the pr is merged it will be tested against master. This happens automatically using our merge bot (bors)

> Similar to the numpy package, don't maintain a requirements.txt file for the base. Use the base requirements in setup.cfg or environment.yml as source of requirements.

This is what we used to do originally and we explicitly switched away from that because we found it unacceptably brittle. The issue is that every single update to a dependency or a transitive dependency has the risk of breaking CI. I cannot count the number of times where in stead of writing or reviewing code I had to fix the CI build by pinning some dependency since an update broke it. I would be very sad to go back to a system that imho is not good enough. Compared to numpy we have many more dependencies since we sit higher up in the stack so we are much more exposed to these kind of issues.

> This should also relax a bit of maintenance burden as maintainers do not have to manually bump versions as packages are updated. 

Again I don't really really think so since this switches work from a simple near automatic task to an unbound random task of tracking down what broke. 

> Since many users will install from pypi or conda, this will also ensure that as packages are updated, maintainers are alerted to potential issues.

I think I disagree with this in practice. Using a dependabot setup we every day get an explicit pr where it is very clear which dependency upgrade. Unpinning the dependencies we may not even notice this one a day where there are no contributions to qcodes and it is much harder to track down what went wrong. 

> Remove pinned versions from environment.yml. The versions of python, jupyter and numpy in this file are getting quite old.

This file is unused and very out of date. We have not recommended users to install qcodes in this way for a long time and the file was only kept around for backwards compatibility. QCoDeS can either be installed fully from pypi or using conda forge using the conda package of QCoDeS. I think its time to remove this file.

spauka, 2021-12-15:
Fair enough, thanks for the very detailed response @jenshnielsen. It sounds like there are detailed reasons that things are as they are and given that I don't think there's too much benefit to continuing this discussion :)

closed 2021-12-15 by spauka