microsoft/Qcodes #3896: Experiment widget ghost tab

mingyee2, 2022-02-03:
### Steps to reproduce
1. Run through the “15 minutes to QCoDeS tutorial” up to [27]
2. Click both Snapshot buttons
3. Click the *left* snapshot tab
4. Click the Close Snapshot button

### Expected behaviour
The tab closes, leaving behind the second tab created (the second one clicked)

### Actual behaviour
The *second* tab closes, leaving behind an empty “ghost” tab for the first one clicked that cannot be closed (because the tab is now empty)

### System
It would be helpful to provide such information:
Windows 10, QCoDeS 0.32.0

jenshnielsen, 2022-02-18:
Thanks for the report. I can reproduce the issue. From reading the docs of ipywidgets its not clear how one is supposed to close a tab. There is also this stack overflow https://stackoverflow.com/questions/69452318/with-ipywidgets-how-to-close-the-tab-containing-the-button-that-is-pressed without an answer

jenshnielsen, 2022-02-18:
I think the problem is that the modification of the tab.children inside the delete method does not work correctly.

e.g. here
```
tab.children = tuple(c for c in tab.children if c != output)
```

I have tried replacing this with just output.close() however while this seems to work it now breaks with another error while creating new tabs