microsoft/Qcodes #3939: SignalHound USB_SA124B

Astlaan, 2022-02-17:
I have two questions regarding the driver for SignalHound USB_SA124B:

1. I see that, unlike for example the EraSynth drivers, this driver doesn't take the COM port or any other selector as argument. How is it then possible to simultaneously connect and use two signalhounds? When I instantiate the class, it connects to one of the signalhounds connected to the PC, but I'm not aware of how the driver determines which instrument to connect to.
2. This driver specifically makes use of Windows `.dll` . Are Linux compatible drivers not available for this instrument?

jenshnielsen, 2022-02-17:
I am not all that familiar with the signal hound but it looks like 

1. OpenDevice takes a device handle which is an integer starting at 0. I would guess that if you want to connect to a second signalhound you probably need to increase that counter. We should probably let the user pass this argument to the init of the device defaulting to 0. 
2. The driver is written and tested on windows. If there is a linux version of the software from Signal hound it should not be too much work to extend the driver to also work against a linux version of the driver (is an .SO file) but that does require some work

Astlaan, 2022-02-17:
For 1., I modified the driver locally in order to pass the handle during instantiation.

Instantiating the class twice with different handle values appears to work. The order with which the instruments are connected appeared not to depend on the handle. Connections to the instruments were always established in the same order.

jenshnielsen, 2022-02-17:
Ok good. We would be very happy for a pr that allows the user to set this when creating the instrument similary to how a visa instrument takes the address