microsoft/Qcodes #4203: Qcodes cannot connect tektronix TPS2012 scope

gztony1227, 2022-05-27:
Dear Jens and others,
I already sent an email about this. Now I tried more and have it opened as issue

### Steps to reproduce
1.Having a Tektronix TPS 2012B scope
2.can connect to it with pyvisa
3. cannot connect to it with qcodes which is based on pyvisa

### Expected behaviour
I expect to connect to TPS 2012B

### Actual behaviour
I cannot connect best to see this video:
https://www.dropbox.com/t/NLpXHO8Sz4Alc9E3

### System
It would be helpful to provide such information:

**operating system**
Windows 10 

If you are using a released version of qcodes (recommended):
qcodes 0.32.0

jenshnielsen, 2022-06-01:
@gztony1227 

It looks like you can connect with pyvisa but get a time out when connecting via QCoDeS. 

Could you try to log what is going on 

E.g. something like

``
import qcodes.logger as logger
from qcodes.instrument_driver.tektronix.TPS2012 import TPS2012

with logger.console_level('DEBUG'):
    TPS2012("name", address)
```

And print the output here. That should help us figure out when the timeout happens

gztony1227, 2022-06-01:
Dear Jens,
Thank you for looking into this. I appreciate it.
I'm not sure why I'm even having issue to make the logger or logging work:
What I tried so far:

jenshnielsen, 2022-06-01:
Sorry you probably need to enable the logger first

```
from qcodes.logger import start_all_logging
start_all_logging()
```

Also see https://qcodes.github.io/Qcodes/examples/logging/logging_example.html for more details

gztony1227, 2022-06-01:
Hi Jens,
Here is the printed output:

2022-06-01 12:53:47,284 ¦ qcodes.instrument.base.com.visa ¦ INFO ¦ visa ¦ set_address ¦ 118 ¦ [tek(TPS2012)] Opening PyVISA Resource Manager with default backend.
2022-06-01 12:53:47,314 ¦ qcodes.instrument.base.com.visa ¦ INFO ¦ visa ¦ set_address ¦ 123 ¦ [tek(TPS2012)] Opening PyVISA resource at address: ASRL5::INSTR
2022-06-01 12:53:47,364 ¦ qcodes.instrument.base.com.visa ¦ DEBUG ¦ visa ¦ ask_raw ¦ 234 ¦ [tek(TPS2012)] Querying: *IDN?

any idea what went wrong? If it's not obvious, maybe we can zoom? I sent you my calendar for available times in the email.

best,
Tony

jenshnielsen, 2022-06-01:
You see above here 

```2022-06-01 12:54:07,376 ¦ qcodes.instrument.base ¦ DEBUG ¦ base ¦ get_idn ¦ 599 ¦ [tek(TPS2012)] Error getting or interpreting *IDN?: ''``` 
That when we are asking `*IDN?` to the instrument we are not getting the expected result back.

If open the instrument with pyvisa directly. Could you try sending the same command with 

```
inst.query("*IDN?")
```
where inst is the instrument created from open_resource

and see if that returns anything

gztony1227, 2022-06-01:
Hi Jens,
Yeah, with pyvisa, using following codes:
```
import pyvisa

rm = pyvisa.ResourceManager()
rm.list_resources()
inst = rm.open_resource('ASRL5::INSTR')
print(inst.query("*IDN?"))
```

I get correct output:
```
rm = pyvisa.ResourceManager()
rm.list_resources()
inst = rm.open_resource('ASRL5::INSTR')
print(inst.query("*IDN?"))
TEKTRONIX,TPS 2012B,0,CF:91.1CT FV:v11.10 
```
I have connected numerous instruments with qcodes. This is a very strange case. Do you have a local instrument with exactly model to try?

jenshnielsen, 2022-06-01:
I wonder if the terminator is set wrong. Looking at this manual it seems like the read terminator can be configured by put the instrument should take any terminator

https://download.tek.com/manual/071144104web.pdf page 90

we don’t set any terminator so you could check what the instrument is set to and then pass terminator=“\n” or whatever makes sense

jenshnielsen, 2022-06-02:
In https://github.com/QCoDeS/Qcodes/pull/4219 I changed the logic so that we use the same terminator by default as pyvisa will normally. It would be great if you could test the master branch of QCoDeS to see if this is enough or if we also need to explicitly set the terminator for the TPS2012.

Let me know if you need help installing qcodes from source

jenshnielsen, 2022-06-04:
Closed by #4219

closed 2022-06-04 by jenshnielsen