microsoft/Qcodes #4365: Typo in the name of the Agilent E8257D driver

mgunyho, 2022-07-06:
I was looking for the driver for our Agilent E8257D, and it took a while to find that the driver is called E8**52**7D in qcodes (flipped 2 and 5). Note that the [unit test](https://github.com/QCoDeS/Qcodes/blob/master/qcodes/tests/drivers/test_Agilent_E8527D.py#L21) where the IDN is checked is correct.

It should be fairly straightforward to rename the class and create a deprecated class with the old name, but if that's too much incompatibility, it would be enough to mention this in the docstring of the driver so that it would show up in a search for "8257D" (that's how I was trying to find the driver).

jenshnielsen, 2022-07-06:
@mgunyho Thanks, looking closes it seems like all the Agilent instrument names are somewhat of a mess.

I think we should take the chance to clean this up.

I suggest that we stick with a convention of the class called `{Vendor}{ModelNumber}` (CamelCase) and the file called `{Vendor}_{Modelnumber}.py` and then add backwards compatible wrappers. 

We currently have:
*  Agilent_3440A in Agilent_3440A.py
*  E8267 in E8267C.py
*  Agilent_E8527D in E8527D.py

I think these should be

*  Agilent3440A in Agilent_3440A.py
*  AgilentE8267C in Agilent_E8267C.py
*  AgilentE8257D in Agilent_E8257D.py

It's unclear to me what the difference is between the c and d models and if the drivers support both so perhaps, they should be?

*  Agilent3440A in Agilent_3440A.py
*  AgilentE8267 in Agilent_E8267.py
*  AgilentE8257 in Agilent_E8257.py

mgunyho, 2022-07-06:
Sounds like a good idea! Unfortunately I won't have enough time to submit a PR in the near future.

Another small gripe I have with the E8257 driver is that the parameter that turns RF generation on/off is called "status" (without any docstring!), I think it should be renamed to output_enabled to be consistent with other drivers (although I know that some other MW source drivers call it status as well).

I'm not that familiar with Agilent/Keysight naming conventions but by a cursory scan on the website it doesn't seem like there are similar devices with names like E`NNNN`D and E`NNNN`C, so maybe the last letter also signifies something. I think it would be best to include the letter in the driver name as well, to minimize confusion.

mgunyho, 2022-07-06:
Hmm okay, for the E8267C at least, it looks like the C signifies the "generation", the Keysight website says that it's an obsolete device and they recommend E8267D instead.. So maybe I was wrong.

jenshnielsen, 2022-07-25:
@mgunyho would you by any chance have time to test #4371

mgunyho, 2022-07-25:
I am on vacation until next Monday, I'll be able to try it then. Sorry for the delay.

jenshnielsen, 2022-07-25:
No worries thanks ;)

commit 308f556d (2022-08-01): Merge #4371

4371: Update agilent drivers to conform to docs standard r=jenshnielsen a=jenshnielsen

Updates to implement #4368  for Agilent. Closes #4365 

`@mgunyho` Could you test the new driver. I have updated the name of the variable and gotten rid of the not needed deg2rad converters and replaced them with numpy

I also suggest that we get rid of the autogenerated docs for the instruments and replace them with something like this.
I would like to add autodocsumm for these modules but am facing the same issue as in https://github.com/QCoDeS/Qcodes/pull/4273
PR #4371 (2022-08-01): Update agilent drivers to conform to docs standard
Updates to implement #4368  for Agilent. Closes #4365 

@mgunyho Could you test the new driver. I have updated the name of the variable and gotten rid of the not needed deg2rad converters and replaced them with numpy

I also suggest that we get rid of the autogenerated docs for the instruments and replace them with something like this.
I would like to add autodocsumm for these modules but am facing the same issue as in https://github.com/QCoDeS/Qcodes/pull/4273
closed 2022-08-01 by bors[bot]