microsoft/Qcodes #4593: problem with station and meas.run():

user123369, 2022-09-09:
when adding station to the measurement then using the 'with meas.run( ) as datasaver' the station does not seem to work.
further the station object does not want to return a snapshot either after running the meas.run(), even though it returned it fine earlier in the code.

Here is all the code:

import Qcodes_Keithley213Driver_2 as kd
import qcodes_fluke8842aDriver_2 as fd
import numpy as np
import matplotlib.pyplot as plt
import time

import tempfile
import os

import qcodes
from qcodes import initialise_or_create_database_at, \
    load_or_create_experiment, Measurement, Parameter
from qcodes.dataset.plotting import plot_dataset
from qcodes.tests.instrument_mocks import (
    DummyInstrument,
    DummyInstrumentWithMeasurement
)

keithSource.close()
flukeMeter.close()

keithSource = kd.Keithley213Driver('keithSource','GPIB0::9::INSTR')
flukeMeter = fd.Fluke8842aDriver('fDMMInst1','GPIB0::10::INSTR')
# dac = DummyInstrument('dac', gates=['ch1', 'ch2'])

station = qcodes.Station()
# station.add_component(dac)
# station.add_component(keithSource)
# station.add_component(flukeMeter)
# _=station.snapshot()
# print(_)                                               #returns fine here

initialise_or_create_database_at('C:\\Users\\ehb21173\\Desktop\\transistor test 2\\transistor_tests_2.db')
qcodes.config.core.db_location
exp = load_or_create_experiment(experiment_name='transistor test 1_ IV plots', sample_name='test on 2n7000_2')
meas = Measurement(exp=exp, station=station, name='transistor tests name')

meas.register_parameter(keithSource.ch1.voltage)
meas.register_parameter(keithSource.ch2.voltage)
meas.register_parameter(flukeMeter.voltageDC, setpoints=(keithSource.ch1.voltage, keithSource.ch2.voltage))

keithSource.ch1.maxV(10) # gate
keithSource.ch2.maxV(10) # source
sourceVoltage = np.linspace(0,10,2,endpoint=True)
gateVoltage = np.linspace(1,4,2,endpoint=True)

meas.write_period = 1

with meas.run() as datasaver:
    for j in range(len(gateVoltage)):
        keithSource.ch1.voltage(gateVoltage[j])  # gate

        for i in range(len(sourceVoltage)):
            keithSource.ch2.voltage(sourceVoltage[i]) # source
            time.sleep(3)
            voltage = flukeMeter.voltageDC()
            datasaver.add_result((keithSource.ch1.voltage, gateVoltage[j]),
                                 (keithSource.ch2.voltage, sourceVoltage[i]),
                                 (flukeMeter.voltageDC, voltage))

        time.sleep(3)
        keithSource.ch2.voltage(0)
        keithSource.ch1.voltage(0)
        time.sleep(3)
dataset = datasaver.dataset
[... 199 lines cut ...]
                         docstring='command for measurement of voltage DC', *args, **kwargs)

    # you must provide a get method, a set method, or both.
    def get_raw(self):
        self.root_instrument.visa_handle.write("G0")
        config = self.root_instrument.visa_handle.read_raw()
        return self.root_instrument.list_of_configs_S[int(str((int(config)))[2])] #change this to dictionary so easier to read

    def set_raw(self, speed):
        index = self.root_instrument.list_of_configs_S.index(speed)
        self.root_instrument.visa_handle.write("S"+str(index))

class Fluke8842aDriver(VisaInstrument):

    def __init__(self, name, address, *args,**kwargs):
        super().__init__(name, address, terminator="\r\n", *args,**kwargs)

        self.currentMeasConfig = 'V DC'
   

        self.list_of_configs_S = ["slow", "med","fast"]

        self.add_parameter('voltageDC', parameter_class=voltageDC,get_parser=float, unit='V')
        self.add_parameter('currentDC', parameter_class=CurrentDC,get_parser=float, unit='A')
        self.add_parameter('rate', parameter_class=rate)

astafan8, 2022-09-09:
@user123369 thanks for reporting and adding all your code for reference, that is helpful! however, it's still not easy for me to see what can go wrong. Just for the record, what python version and qcodes version and pyvisa version are you using (you can see it if you run `pip list` in your python environment)?

let's try to debug the issue. Could you:
1. could you restart the python kernel, and then initialize `flukeMeter` and then, without the station, execute `flukeMeter.print_readable_snapshot()` (a few times)? does this work? if it does not work, then there's something wrong in the driver (that i can't see from the code, but could be in the way the instrument works). 
2. if the above works and there are no issues, then could you call `flukeMeter.snapshot()` and share the content of the returned dictionary (when pasting into a github message you can wrap that in triple-tick-quotes, like \```the-snapshot-here```)?
3. after that, try to run your measurement again (or at least the `meas.run()` part) and see if you get the recursion error again.

user123369, 2022-09-09:
@astafan8 Hi, thanks for the quick repsonse.

1)I have restarted the kernel and ran print_readble_snapshot() several times it works just fine
2) the return of the flukeMeter.snapshot() before anstantiating the statin is:

{'functions': {},
 'submodules': {},
 'parameters': {'IDN': {'__class__': 'qcodes.instrument.parameter.Parameter',
   'full_name': 'fDMMInst1_IDN',
   'value': None,
   'raw_value': None,
   'ts': None,
   'inter_delay': 0,
   'unit': '',
   'label': 'IDN',
   'post_delay': 0,
   'vals': '<Anything>',
   'name': 'IDN',
   'instrument': 'qcodes_fluke8842aDriver_2.Fluke8842aDriver',
   'instrument_name': 'fDMMInst1'},
  'timeout': {'__class__': 'qcodes.instrument.parameter.Parameter',
   'full_name': 'fDMMInst1_timeout',
   'value': 10.0,
   'raw_value': 10.0,
   'ts': '2022-09-09 14:17:52',
   'inter_delay': 0,
   'unit': 's',
   'label': 'timeout',
   'post_delay': 0,
   'vals': '<MultiType: Numbers v>=0, Enum: {None}>',
   'name': 'timeout',
   'instrument': 'qcodes_fluke8842aDriver_2.Fluke8842aDriver',
   'instrument_name': 'fDMMInst1'},
  'voltageDC': {'__class__': 'qcodes_fluke8842aDriver_2.voltageDC',
   'full_name': 'fDMMInst1_voltageDC',
   'value': None,
   'raw_value': None,
   'ts': None,
   'inter_delay': 0,
   'unit': 'V',
   'label': 'voltageDC',
   'post_delay': 0,
   'name': 'voltageDC',
   'instrument': 'qcodes_fluke8842aDriver_2.Fluke8842aDriver',
   'instrument_name': 'fDMMInst1'},
  'currentDC': {'__class__': 'qcodes_fluke8842aDriver_2.CurrentDC',
   'full_name': 'fDMMInst1_currentDC',
   'value': None,
   'raw_value': None,
   'ts': None,
   'inter_delay': 0,
   'unit': 'A',
   'label': 'currentDC',
   'post_delay': 0,
   'name': 'currentDC',
   'instrument': 'qcodes_fluke8842aDriver_2.Fluke8842aDriver',
   'instrument_name': 'fDMMInst1'},
  'rate': {'__class__': 'qcodes_fluke8842aDriver_2.rate',
   'full_name': 'fDMMInst1_rate',
   'value': None,
   'raw_value': None,
   'ts': None,
   'inter_delay': 0,
   'unit': '',
   'label': 'rate',
   'post_delay': 0,
   'name': 'rate',
   'instrument': 'qcodes_fluke8842aDriver_2.Fluke8842aDriver',
   'instrument_name': 'fDMMInst1'}},
 '__class__': 'qcodes_fluke8842aDriver_2.Fluke8842aDriver',
 'name': 'fDMMInst1',
 'address': 'GPIB0::10::INSTR',
 'terminator': '\r\n',
 'read_terminator': '\r\n',
 'write_terminator': '\r\n',
 'timeout': 10.0}

3) the measurement works (as long as I don't add station to the experiment in which case it gives the error above), however when I try to run flukeMeter.snapshot() after teh meas.run() i get the following error:

AttributeError                            Traceback (most recent call last)
Input In [13], in <cell line: 1>()
----> 1 flukeMeter.snapshot()

File ~\.conda\envs\drivers\lib\site-packages\qcodes\utils\metadata.py:64, in Metadatable.snapshot(self, update)
     51 def snapshot(self, update: Optional[bool] = False) -> Dict[Any, Any]:
     52     """
     53     Decorate a snapshot dictionary with metadata.
     54     DO NOT override this method if you want metadata in the snapshot
   (...)
     61         Base snapshot.
     62     """
---> 64     snap = self.snapshot_base(update=update)
     66     if len(self.metadata):
     67         snap['metadata'] = self.metadata

File ~\.conda\envs\drivers\lib\site-packages\qcodes\instrument\visa.py:253, in VisaInstrument.snapshot_base(self, update, params_to_skip_update)
    231 def snapshot_base(self, update: Optional[bool] = True,
    232                   params_to_skip_update: Optional[Sequence[str]] = None
    233                   ) -> Dict[Any, Any]:
    234     """
    235     State of the instrument as a JSON-compatible dict (everything that
    236     the custom JSON encoder class :class:`qcodes.utils.helpers.NumpyJSONEncoder`
   (...)
    251         dict: base snapshot
    252     """
--> 253     snap = super().snapshot_base(update=update,
    254                                  params_to_skip_update=params_to_skip_update)
    256     snap["address"] = self._address
    257     snap["terminator"] = self.visa_handle.read_termination

File ~\.conda\envs\drivers\lib\site-packages\qcodes\instrument\base.py:253, in InstrumentBase.snapshot_base(self, update, params_to_skip_update)
    248 if params_to_skip_update is None:
    249     params_to_skip_update = []
    251 snap: Dict[str, Any] = {
    252     "functions": {name: func.snapshot(update=update)
--> 253                   for name, func in self.functions.items()},
    254     "submodules": {name: subm.snapshot(update=update)
    255                    for name, subm in self.submodules.items()},
    256     "parameters": {},
    257     "__class__": full_class(self)
    258 }
    260 for name, param in self.parameters.items():
    261     if param.snapshot_exclude:

File ~\.conda\envs\drivers\lib\site-packages\qcodes\utils\helpers.py:417, in DelegateAttributes.__getattr__(self, key)
    414 for name in self.delegate_attr_dicts:
    415     if key == name:
    416         # needed to prevent infinite loops!
--> 417         raise AttributeError(
    418             "dict '{}' has not been created in object '{}'".format(
    419                 key, self.__class__.__name__))
    420     try:
    421         d = getattr(self, name, None)

AttributeError: dict 'functions' has not been created in object 'Fluke8842aDriver'

jenshnielsen, 2022-09-09:
@user123369 What happens is that initially when you call snapshot the snapshot will be empty as can be seen by the default values in your snapshot above. Adding it to the station will trigger a refresh of the snapshot. 

I would bet that you can reproduce the issue by doing 

```
flukeMeter.snapshot(update=True)
```
in a new python process without any station and any measurement. 
This causes the snapshot method to call get on all parameters in the instrument which is the same that you do when the instrument is added to the station

user123369, 2022-09-09:
Hi @jenshnielsen 
I have started a new ipynb file and ran the follwoing code:

'''
import Qcodes_Keithley213Driver_2 as kd
import qcodes_fluke8842aDriver_2 as fd

keithSource = kd.Keithley213Driver('keithSource','GPIB0::9::INSTR')
flukeMeter = fd.Fluke8842aDriver('fDMMInst1','GPIB0::10::INSTR')

print(flukeMeter.snapshot(update=True))

keithSource.close()
flukeMeter.close()
'''

the return seems fine (no issues, the return is:
'''
{'functions': {}, 'submodules': {}, 'parameters': {'IDN': {'__class__': 'qcodes.instrument.parameter.Parameter', 'full_name': 'fDMMInst1_IDN', 'value': {'vendor': '+1.0071E+21', 'model': None, 'serial': None, 'firmware': None}, 'raw_value': {'vendor': '+1.0071E+21', 'model': None, 'serial': None, 'firmware': None}, 'ts': '2022-09-09 14:39:21', 'unit': '', 'inter_delay': 0, 'label': 'IDN', 'post_delay': 0, 'name': 'IDN', 'instrument': 'qcodes_fluke8842aDriver_2.Fluke8842aDriver', 'instrument_name': 'fDMMInst1', 'vals': '<Anything>'}, 'timeout': {'__class__': 'qcodes.instrument.parameter.Parameter', 'full_name': 'fDMMInst1_timeout', 'value': 10.0, 'raw_value': 10.0, 'ts': '2022-09-09 14:39:21', 'unit': 's', 'inter_delay': 0, 'label': 'timeout', 'post_delay': 0, 'name': 'timeout', 'instrument': 'qcodes_fluke8842aDriver_2.Fluke8842aDriver', 'instrument_name': 'fDMMInst1', 'vals': '<MultiType: Numbers v>=0, Enum: {None}>'}, 'voltageDC': {'__class__': 'qcodes_fluke8842aDriver_2.voltageDC', 'full_name': 'fDMMInst1_voltageDC', 'value': 5.9e-05, 'raw_value': b'+000.059E-3\r\n', 'ts': '2022-09-09 14:39:24', 'unit': 'V', 'inter_delay': 0, 'label': 'voltageDC', 'post_delay': 0, 'name': 'voltageDC', 'instrument': 'qcodes_fluke8842aDriver_2.Fluke8842aDriver', 'instrument_name': 'fDMMInst1'}, 'currentDC': {'__class__': 'qcodes_fluke8842aDriver_2.CurrentDC', 'full_name': 'fDMMInst1_currentDC', 'value': 1.0031e+21, 'raw_value': b'+1.0031E+21\r\n', 'ts': '2022-09-09 14:39:27', 'unit': 'A', 'inter_delay': 0, 'label': 'currentDC', 'post_delay': 0, 'name': 'currentDC', 'instrument': 'qcodes_fluke8842aDriver_2.Fluke8842aDriver', 'instrument_name': 'fDMMInst1'}, 'rate': {'__class__': 'qcodes_fluke8842aDriver_2.rate', 'full_name': 'fDMMInst1_rate', 'value': 'slow', 'raw_value': 'slow', 'ts': '2022-09-09 14:39:27', 'unit': '', 'inter_delay': 0, 'label': 'rate', 'post_delay': 0, 'name': 'rate', 'instrument': 'qcodes_fluke8842aDriver_2.Fluke8842aDriver', 'instrument_name': 'fDMMInst1'}}, '__class__': 'qcodes_fluke8842aDriver_2.Fluke8842aDriver', 'name': 'fDMMInst1', 'address': 'GPIB0::10::INSTR', 'terminator': '\r\n', 'read_terminator': '\r\n', 'write_terminator': '\r\n', 'timeout': 10.0}
'''

astafan8, 2022-09-09:
thanks! the snapshot looks fine except

> `'value': 1.0031e+21`

that's a very high value for a DC current. is that correct? (this may or may not be related to the issue though)

-------

> flukeMeter.close()

i see that in the original code above the instrument is being closed at the end of the measurement - calling `snapshot()` on a closed instrument should not work, so perhaps try to NOT close the instrument after the measurement?

-------

ok, the snapshot looks ok, `flukeMeter.snapshot(update=True)` works, so now let's try again to add it to the station and run the measurement (please, restart the kernel again so that we start from a clean slate)?

as an additional info: there's no need to pass the station instance to Measurement explicitly, see the docs on the default station https://qcodes.github.io/Qcodes/examples/Station.html#Default-Station .

jenshnielsen, 2022-09-09:
Note that the value of `1.00071e21` also shows up as the vendor of the instrument so it seems like there is a potential issue with the instrumets implementation of `*IDN` too

astafan8, 2022-09-09:
@user123369 as a sidenote, i'd also recommend to simplify the driver to not use visa_handle directly, and use ask/write that qcodes' VisaInstrument implements already, and also add a "meas_config" parameter to keep track of V-DC/A-DC stuff, like this:

```python
class voltageDC(Parameter):
    def __init__(self, name, *args, **kwargs):
        # only name is required
        super().__init__(name,
                         docstring='command for measurement of voltage DC', *args, **kwargs)

    def get_raw(self):
        if self.root_instrument.meas_config() != 'V DC':
            self.root_instrument.meas_config('V DC')

        return self.root_instrument.ask('?')

class CurrentDC(Parameter):
    def __init__(self, name, *args, **kwargs):
        # only name is required
        super().__init__(name,
                         docstring='command for measurement of voltage DC', *args, **kwargs)

    # you must provide a get method, a set method, or both.
    def get_raw(self):
        if self.root_instrument.meas_config() != 'A DC':
            self.root_instrument.meas_config('A DC')

        return self.instrument.ask.write('?')

class rate(Parameter):
    def __init__(self, name, *args, **kwargs):
        # only name is required
        super().__init__(name,
                         docstring='command for measurement of voltage DC', *args, **kwargs)

    # you must provide a get method, a set method, or both.
    def get_raw(self):
        config = self.instrument.ask("G0")
        return self.root_instrument.list_of_configs_S[int(str((int(config)))[2])] #change this to dictionary so easier to read

    def set_raw(self, speed):
        index = self.root_instrument.list_of_configs_S.index(speed)
        self.instrument.write("S"+str(index))

class Fluke8842aDriver(VisaInstrument):

    def __init__(self, name, address, *args,**kwargs):
        super().__init__(name, address, terminator="\r\n", *args,**kwargs)

        self.list_of_configs_S = ["slow", "med","fast"]

        self.add_parameter('meas_config', val_mapping={'A DC': 'F5',  'V DC': 'F1'})
        self.meas_config('V DC')

        self.add_parameter('voltageDC', parameter_class=voltageDC,get_parser=float, unit='V')
        self.add_parameter('currentDC', parameter_class=CurrentDC,get_parser=float, unit='A')
        self.add_parameter('rate', parameter_class=rate)
```

user123369, 2022-09-09:
Hi @jenshnielsen and @astafan8 

1) the value of dc current of 1.0031e+21 is just a default value the instrument returns when, so in the case of dc current it returns this value as I have it setup for a voltage measurement and asking for current doesn't work

2)the same thing foes for the *IDN; dince the instrument is old there is no *IDN query so it just returns a defualt value (i guess i will need to overide this but, still i think since its returning something the code chould work.

3)you are right now that i dont close the instrument in the original code the snapshot of the multimeter is returned after the mea.run() (when station is not added to the experient.

4)as for the code here is the new code i am running in a new kernel:
```
import tempfile
import os

import qcodes
from qcodes import initialise_or_create_database_at, \
    load_or_create_experiment, Measurement, Parameter
from qcodes.dataset.plotting import plot_dataset
from qcodes.tests.instrument_mocks import (
    DummyInstrument,
    DummyInstrumentWithMeasurement
)

keithSource.close()
flukeMeter.close()
import Qcodes_Keithley213Driver_2 as kd
import qcodes_fluke8842aDriver_2 as fd

keithSource = kd.Keithley213Driver('keithSource','GPIB0::9::INSTR')
flukeMeter = fd.Fluke8842aDriver('fDMMInst1','GPIB0::10::INSTR')

station = qcodes.Station(flukeMeter)
initialise_or_create_database_at('C:\\Users\\ehb21173\\Desktop\\transistor test 2\\transistor_tests_2.db')

exp = load_or_create_experiment(experiment_name='transistor test 1_ IV plots', sample_name='test on 2n7000_2')

meas = Measurement(exp=exp, name='transistor tests name')

meas.register_parameter(keithSource.ch1.voltage)
meas.register_parameter(keithSource.ch2.voltage)
meas.register_parameter(flukeMeter.voltageDC, setpoints=(keithSource.ch1.voltage, keithSource.ch2.voltage))

meas.write_period = 1

with meas.run() as datasaver:

            datasaver.add_result((keithSource.ch1.voltage, 1),
                                 (keithSource.ch2.voltage, 2),
                                 (flukeMeter.voltageDC, 3))
```
after getting to this point with no errors i get the same error

```
RecursionError                            Traceback (most recent call last)
Input In [8], in <cell line: 3>()
      1 meas.write_period = 1
----> 3 with meas.run() as datasaver:
      5             datasaver.add_result((keithSource.ch1.voltage, 1),
      6                                  (keithSource.ch2.voltage, 2),
      7                                  (flukeMeter.voltageDC, 3))

File ~\.conda\envs\drivers\lib\site-packages\qcodes\dataset\measurements.py:601, in Runner.__enter__(self)
    598 else:
    599     snapshot = {}
--> 601 self.ds.prepare(
    602     snapshot=snapshot,
    603     interdeps=self._interdependencies,
    604     write_in_background=self._write_in_background,
    605     shapes=self._shapes,
    606     parent_datasets=self._parent_datasets,
    607 )
    609 # register all subscribers
    610 if isinstance(self.ds, DataSet):

File ~\.conda\envs\drivers\lib\site-packages\qcodes\dataset\data_set.py:323, in DataSet.prepare(self, snapshot, interdeps, shapes, parent_datasets, write_in_background)
    313 def prepare(
    314     self,
    315     *,
   (...)
    320     write_in_background: bool = False,
    321 ) -> None:
--> 323     self.add_snapshot(json.dumps({"station": snapshot}, cls=NumpyJSONEncoder))
    325     if interdeps == InterDependencies_():
    326         raise RuntimeError("No parameters supplied")

File ~\.conda\envs\drivers\lib\json\__init__.py:238, in dumps(obj, skipkeys, ensure_ascii, check_circular, allow_nan, cls, indent, separators, default, sort_keys, **kw)
    232 if cls is None:
    233     cls = JSONEncoder
    234 return cls(
    235     skipkeys=skipkeys, ensure_ascii=ensure_ascii,
    236     check_circular=check_circular, allow_nan=allow_nan, indent=indent,
    237     separators=separators, default=default, sort_keys=sort_keys,
--> 238     **kw).encode(obj)

File ~\.conda\envs\drivers\lib\json\encoder.py:199, in JSONEncoder.encode(self, o)
    195         return encode_basestring(o)
    196 # This doesn't pass the iterator directly to ''.join() because the
    197 # exceptions aren't as detailed.  The list call should be roughly
    198 # equivalent to the PySequence_Fast that ''.join() would do.
--> 199 chunks = self.iterencode(o, _one_shot=True)
    200 if not isinstance(chunks, (list, tuple)):
    201     chunks = list(chunks)

File ~\.conda\envs\drivers\lib\json\encoder.py:257, in JSONEncoder.iterencode(self, o, _one_shot)
    252 else:
    253     _iterencode = _make_iterencode(
    254         markers, self.default, _encoder, self.indent, floatstr,
    255         self.key_separator, self.item_separator, self.sort_keys,
    256         self.skipkeys, _one_shot)
--> 257 return _iterencode(o, 0)

File ~\.conda\envs\drivers\lib\site-packages\qcodes\utils\helpers.py:96, in NumpyJSONEncoder.default(self, obj)
     93 elif isinstance(obj, np.ndarray):
     94     # for numpy arrays
     95     return obj.tolist()
---> 96 elif (isinstance(obj, numbers.Complex) and
     97       not isinstance(obj, numbers.Real)):
     98     return {
     99         '__dtype__': 'complex',
    100         're': float(obj.real),
    101         'im': float(obj.imag)
    102     }
    103 elif isinstance(obj, uncertainties.UFloat):

File ~\.conda\envs\drivers\lib\abc.py:119, in ABCMeta.__instancecheck__(cls, instance)
    117 def __instancecheck__(cls, instance):
    118     """Override for isinstance(instance, cls)."""
--> 119     return _abc_instancecheck(cls, instance)

RecursionError: maximum recursion depth exceeded in comparison

```

astafan8, 2022-09-09:
> i guess i will need to override this

well, since so far nothing help, perhaps you should try to override it? for the sake of testing, you can do this:
```python
    def get_idn(self):
        idparts = [None, self.name, None, None]
        return dict(zip(("vendor", "model", "serial", "firmware"), idparts))
```

astafan8, 2022-09-09:
... then i'm out of ideas/clues for the moment :(

it seems that one of the values in the snapshot is not really what it looks like, because it seems to recurse forever at the isinstance call on it, and it seems to be about Real/Complex/float types.

what are python, qcodes, pyvisa versions in the python environement that you're using?

user123369, 2022-09-09:
pyvisa 1.11.3
qcodes 0.34.1
python 3.10.4

jenshnielsen, 2022-09-09:
Could you try to see if just doing a snapshot of the station triggers the issue.

```
import qcodes_fluke8842aDriver_2 as fd
flukeMeter = fd.Fluke8842aDriver('fDMMInst1','GPIB0::10::INSTR')
station = qcodes.Station(flukeMeter)
station.snapshot()
```

or

```
import qcodes_fluke8842aDriver_2 as fd
flukeMeter = fd.Fluke8842aDriver('fDMMInst1','GPIB0::10::INSTR')
station = qcodes.Station(flukeMeter)
station.snapshot(update=True)
```

user123369, 2022-09-09:
@jenshnielsen hi will try this on Monday or tomorrow if I get in the lab.

user123369, 2022-09-12:
@jenshnielsen hi yep that works, here is the code:

 ```
 keithSource = kd.Keithley213Driver('keithSource','GPIB0::9::INSTR')
flukeMeter = fd.Fluke8842aDriver('fDMMInst1','GPIB0::10::INSTR')

station =qcodes.Station(flukeMeter)
flukeMeter.snapshot()
print(flukeMeter.snapshot(update=True))

keithSource.close()
 ```

here is the output:
 ```
{'functions': {}, 'submodules': {}, 'parameters': {'IDN': {'__class__': 'qcodes.instrument.parameter.Parameter', 'full_name': 'fDMMInst1_IDN', 'value': {'vendor': '+1.0071E+21', 'model': None, 'serial': None, 'firmware': None}, 'raw_value': {'vendor': '+1.0071E+21', 'model': None, 'serial': None, 'firmware': None}, 'ts': '2022-09-12 10:18:30', 'name': 'IDN', 'vals': '<Anything>', 'unit': '', 'inter_delay': 0, 'post_delay': 0, 'instrument': 'qcodes_fluke8842aDriver_2.Fluke8842aDriver', 'instrument_name': 'fDMMInst1', 'label': 'IDN'}, 'timeout': {'__class__': 'qcodes.instrument.parameter.Parameter', 'full_name': 'fDMMInst1_timeout', 'value': 10.0, 'raw_value': 10.0, 'ts': '2022-09-12 10:18:30', 'name': 'timeout', 'vals': '<MultiType: Numbers v>=0, Enum: {None}>', 'unit': 's', 'inter_delay': 0, 'post_delay': 0, 'instrument': 'qcodes_fluke8842aDriver_2.Fluke8842aDriver', 'instrument_name': 'fDMMInst1', 'label': 'timeout'}, 'voltageDC': {'__class__': 'qcodes_fluke8842aDriver_2.voltageDC', 'full_name': 'fDMMInst1_voltageDC', 'value': 5.7e-05, 'raw_value': b'+000.057E-3\r\n', 'ts': '2022-09-12 10:18:34', 'name': 'voltageDC', 'unit': 'V', 'inter_delay': 0, 'post_delay': 0, 'instrument': 'qcodes_fluke8842aDriver_2.Fluke8842aDriver', 'instrument_name': 'fDMMInst1', 'label': 'voltageDC'}, 'currentDC': {'__class__': 'qcodes_fluke8842aDriver_2.CurrentDC', 'full_name': 'fDMMInst1_currentDC', 'value': 1.0031e+21, 'raw_value': b'+1.0031E+21\r\n', 'ts': '2022-09-12 10:18:37', 'name': 'currentDC', 'unit': 'A', 'inter_delay': 0, 'post_delay': 0, 'instrument': 'qcodes_fluke8842aDriver_2.Fluke8842aDriver', 'instrument_name': 'fDMMInst1', 'label': 'currentDC'}, 'rate': {'__class__': 'qcodes_fluke8842aDriver_2.rate', 'full_name': 'fDMMInst1_rate', 'value': 'slow', 'raw_value': 'slow', 'ts': '2022-09-12 10:18:37', 'name': 'rate', 'unit': '', 'inter_delay': 0, 'post_delay': 0, 'instrument': 'qcodes_fluke8842aDriver_2.Fluke8842aDriver', 'instrument_name': 'fDMMInst1', 'label': 'rate'}}, '__class__': 'qcodes_fluke8842aDriver_2.Fluke8842aDriver', 'name': 'fDMMInst1', 'address': 'GPIB0::10::INSTR', 'terminator': '\r\n', 'read_terminator': '\r\n', 'write_terminator': '\r\n', 'timeout': 10.0}

 ```
flukeMeter.close()

jenshnielsen, 2022-09-12:
Please check the snapshot of the station not the instrument e.g.

```
keithSource = kd.Keithley213Driver('keithSource','GPIB0::9::INSTR')
flukeMeter = fd.Fluke8842aDriver('fDMMInst1','GPIB0::10::INSTR')

station =qcodes.Station(flukeMeter)
flukeMeter.snapshot()
print(station.snapshot(update=True))

```

user123369, 2022-09-14:
@jenshnielsen hi sorry for the late reply

here is the new code:
```
keithSource = kd.Keithley213Driver('keithSource','GPIB0::9::INSTR')
flukeMeter = fd.Fluke8842aDriver('fDMMInst1','GPIB0::10::INSTR')

station =qcodes.Station(flukeMeter)
flukeMeter.snapshot()
print(station.snapshot(update=True))
```

it seems to work fine here is the response
```
{'instruments': {'fDMMInst1': {'functions': {}, 'submodules': {}, 'parameters': {'IDN': {'__class__': 'qcodes.instrument.parameter.Parameter', 'full_name': 'fDMMInst1_IDN', 'value': {'vendor': '', 'model': '', 'serial': '', 'firmware': ''}, 'raw_value': {'vendor': '', 'model': '', 'serial': '', 'firmware': ''}, 'ts': '2022-09-14 12:15:59', 'name': 'IDN', 'vals': '<Anything>', 'instrument': 'qcodes_fluke8842aDriver_2.Fluke8842aDriver', 'instrument_name': 'fDMMInst1', 'post_delay': 0, 'unit': '', 'inter_delay': 0, 'label': 'IDN'}, 'timeout': {'__class__': 'qcodes.instrument.parameter.Parameter', 'full_name': 'fDMMInst1_timeout', 'value': 10.0, 'raw_value': 10.0, 'ts': '2022-09-14 12:15:59', 'name': 'timeout', 'vals': '<MultiType: Numbers v>=0, Enum: {None}>', 'instrument': 'qcodes_fluke8842aDriver_2.Fluke8842aDriver', 'instrument_name': 'fDMMInst1', 'post_delay': 0, 'unit': 's', 'inter_delay': 0, 'label': 'timeout'}, 'voltageDC': {'__class__': 'qcodes_fluke8842aDriver_2.voltageDC', 'full_name': 'fDMMInst1_voltageDC', 'value': 5e-05, 'raw_value': b'+000.050E-3\r\n', 'ts': '2022-09-14 12:15:59', 'name': 'voltageDC', 'instrument': 'qcodes_fluke8842aDriver_2.Fluke8842aDriver', 'instrument_name': 'fDMMInst1', 'post_delay': 0, 'unit': 'V', 'inter_delay': 0, 'label': 'voltageDC'}, 'currentDC': {'__class__': 'qcodes_fluke8842aDriver_2.CurrentDC', 'full_name': 'fDMMInst1_currentDC', 'value': 1.0031e+21, 'raw_value': b'+1.0031E+21\r\n', 'ts': '2022-09-14 12:16:02', 'name': 'currentDC', 'instrument': 'qcodes_fluke8842aDriver_2.Fluke8842aDriver', 'instrument_name': 'fDMMInst1', 'post_delay': 0, 'unit': 'A', 'inter_delay': 0, 'label': 'currentDC'}, 'rate': {'__class__': 'qcodes_fluke8842aDriver_2.rate', 'full_name': 'fDMMInst1_rate', 'value': 'slow', 'raw_value': 'slow', 'ts': '2022-09-14 12:16:02', 'name': 'rate', 'instrument': 'qcodes_fluke8842aDriver_2.Fluke8842aDriver', 'instrument_name': 'fDMMInst1', 'post_delay': 0, 'unit': '', 'inter_delay': 0, 'label': 'rate'}}, '__class__': 'qcodes_fluke8842aDriver_2.Fluke8842aDriver', 'name': 'fDMMInst1', 'address': 'GPIB0::10::INSTR', 'terminator': '\r\n', 'read_terminator': '\r\n', 'write_terminator': '\r\n', 'timeout': 10.0}}, 'parameters': {}, 'components': {}, 'config': None}
```

astafan8, 2022-09-14:
ok, and after overwriting the idn, did you also try to add the instrument to the station and take a snapshot via station, and also then run the measurement?

----

> sorry if its a silly question but is there some way to omit certain parameters when calling for the station snapshot?

not silliy at all! there are multiple ways of doing that, see the ``snapshot_*`` arguments to the `Parameter` here https://qcodes.github.io/Qcodes/api/parameters/index.html#qcodes.parameters.Parameter.

user123369, 2022-09-14:
@astafan8 yep still the same problem.

user123369, 2022-09-14:
hi @astafan8  so I am now running the code using the legacy loop methods here is the the code:

```
import Qcodes_Keithley213Driver_2 as kd
import qcodes_fluke8842aDriver_2 as fd
import numpy as np
import matplotlib.pyplot as plt
import time

import tempfile
import os

import qcodes as qc
from qcodes import initialise_or_create_database_at, \
    load_or_create_experiment, Measurement, Parameter
from qcodes.dataset.plotting import plot_dataset
from qcodes.tests.instrument_mocks import (
    DummyInstrument,
    DummyInstrumentWithMeasurement
)

from qcodes.loops import Loop
from qcodes.data.data_set import DataSet
from qcodes.measure import Measure as Measure

# keithSource.close()
# flukeMeter.close()
keithSource = kd.Keithley213Driver('keithSource','GPIB0::9::INSTR')
flukeMeter = fd.Fluke8842aDriver('fDMMInst1','GPIB0::10::INSTR')
# dac = DummyInstrument('dac', gates=['ch1', 'ch2'])

keithSource.ch1.maxV(10) # gate
keithSource.ch2.maxV(10) # source

loop=qc.loops.Loop(keithSource.ch1.voltage.sweep(1,4,1),delay=1).loop(keithSource.ch2.voltage.sweep(0,5,0.5),delay=1).each(flukeMeter.voltageDC)
data = loop.get_data_set(name='data_to_plot')
loop.run()

print(data.fDMMInst1_voltageDC)
```

and the output 

```
DataArray[4,11]: fDMMInst1_voltageDC
array([[4.70000e-05, 4.90000e-05, 4.70000e-05, 4.90000e-05, 4.70000e-05,
        4.80000e-05, 4.90000e-05, 4.50000e-05, 5.50000e-05, 4.60000e-05,
        5.70000e-05],
       [8.20000e-05, 7.52200e-03, 1.34010e-02, 1.67230e-02, 1.86340e-02,
        1.97220e-02, 2.03940e-02, 2.08050e-02, 2.10980e-02, 2.13300e-02,
        2.15060e-02],
       [2.41700e-02, 5.60880e-01, 1.61579e+00, 2.30040e+00, 2.68780e+00,
        2.89510e+00, 3.00640e+00, 3.06680e+00, 3.10000e+00, 3.11900e+00,
        3.13060e+00],
       [1.34417e+00, 1.35471e+00, 2.36860e+00, 3.49890e+00, 4.87710e+00,
        6.11710e+00, 6.87260e+00, 7.27890e+00, 7.49620e+00, 7.61320e+00,
        7.67670e+00]])
```
 so i think ill just stick to this legacy loop method, do you think there will be any problem with this?

astafan8, 2022-09-14:
> seems to work fine. i think ill just stick to this legacy loop method, do you think there will be any problem with this?

well, it's a legacy feature that is not being improved or supported; but it seems to work for you in this particular case, nothing can beat that :)

it likely works because you didn't add the instrument to the station :) could you try adding the instrument to the station and run the legacy loop?

anyways, i'd recommend to still use `meas.run()` measurement context manager, not the legacy loop, but just don't add the problematic instrument to the station (but add all other isntruments).

user123369, 2022-09-14:
@astafan8 yep nevermind, same problem after adding the instruments to the station (its both of them btw).

jenshnielsen, 2022-09-14:
Are we absolutely sure that the fluke is the issue?

E.g. does

```
keithSource = kd.Keithley213Driver('keithSource','GPIB0::9::INSTR')
flukeMeter = fd.Fluke8842aDriver('fDMMInst1','GPIB0::10::INSTR')

station =qcodes.Station(flukeMeter, keithSource)
print(station.snapshot(update=True))
```

work

user123369, 2022-09-14:
@jenshnielsen yep the above works, here is the output for just the keithley (note ive set all the snapshot_get = False for the parameter to see if it changes anything in the keithley driver (it doesnt):
```
station =qcodes.Station( keithSource)
print(station.snapshot(update=True))
```
```
{'instruments': {'keithSource': {'functions': {}, 'submodules': {'ch1': {'functions': {}, 'submodules': {}, 'parameters': {'waveform': {'__class__': 'qcodes.instrument.parameter.Parameter', 'full_name': 'keithSource_ch1_waveform', 'value': 0, 'raw_value': 0, 'ts': '2022-09-14 13:00:20', 'name': 'waveform', 'inter_delay': 0, 'post_delay': 0, 'unit': 'V', 'instrument': 'Qcodes_Keithley213Driver_2.KeithleyChannel', 'instrument_name': 'keithSource_ch1', 'label': 'waveform'}, 'waveformStepTime': {'__class__': 'qcodes.instrument.parameter.Parameter', 'full_name': 'keithSource_ch1_waveformStepTime', 'value': 0, 'raw_value': 0, 'ts': '2022-09-14 13:00:20', 'name': 'waveformStepTime', 'inter_delay': 0, 'post_delay': 0, 'unit': 'ms', 'instrument': 'Qcodes_Keithley213Driver_2.KeithleyChannel', 'instrument_name': 'keithSource_ch1', 'label': 'waveformStepTime'}, 'startAdress': {'__class__': 'qcodes.instrument.parameter.Parameter', 'full_name': 'keithSource_ch1_startAdress', 'value': 0, 'raw_value': 0, 'ts': '2022-09-14 13:00:20', 'name': 'startAdress', 'inter_delay': 0, 'post_delay': 0, 'unit': '', 'instrument': 'Qcodes_Keithley213Driver_2.KeithleyChannel', 'instrument_name': 'keithSource_ch1', 'label': 'startAdress'}, 'cycles': {'__class__': 'qcodes.instrument.parameter.Parameter', 'full_name': 'keithSource_ch1_cycles', 'value': None, 'raw_value': None, 'ts': None, 'name': 'cycles', 'inter_delay': 0, 'post_delay': 0, 'unit': '', 'instrument': 'Qcodes_Keithley213Driver_2.KeithleyChannel', 'instrument_name': 'keithSource_ch1', 'label': 'cycles'}, 'voltage': {'__class__': 'Qcodes_Keithley213Driver_2.voltage', 'full_name': 'keithSource_ch1_voltage', 'value': 0, 'raw_value': 0, 'ts': '2022-09-14 13:01:52', 'name': 'voltage', 'inter_delay': 0, 'post_delay': 0, 'unit': 'V', 'instrument': 'Qcodes_Keithley213Driver_2.KeithleyChannel', 'instrument_name': 'keithSource_ch1', 'label': 'voltage'}}, '__class__': 'Qcodes_Keithley213Driver_2.KeithleyChannel', 'name': 'keithSource_ch1'}, 'ch2': {'functions': {}, 'submodules': {}, 'parameters': {'waveform': {'__class__': 'qcodes.instrument.parameter.Parameter', 'full_name': 'keithSource_ch2_waveform', 'value': 0, 'raw_value': 0, 'ts': '2022-09-14 13:00:20', 'name': 'waveform', 'inter_delay': 0, 'post_delay': 0, 'unit': 'V', 'instrument': 'Qcodes_Keithley213Driver_2.KeithleyChannel', 'instrument_name': 'keithSource_ch2', 'label': 'waveform'}, 'waveformStepTime': {'__class__': 'qcodes.instrument.parameter.Parameter', 'full_name': 'keithSource_ch2_waveformStepTime', 'value': 0, 'raw_value': 0, 'ts': '2022-09-14 13:00:20', 'name': 'waveformStepTime', 'inter_delay': 0, 'post_delay': 0, 'unit': 'ms', 'instrument': 'Qcodes_Keithley213Driver_2.KeithleyChannel', 'instrument_name': 'keithSource_ch2', 'label': 'waveformStepTime'}, 'startAdress': {'__class__': 'qcodes.instrument.parameter.Parameter', 'full_name': 'keithSource_ch2_startAdress', 'value': 0, 'raw_value': 0, 'ts': '2022-09-14 13:00:20', 'name': 'startAdress', 'inter_delay': 0, 'post_delay': 0, 'unit': '', 'instrument': 'Qcodes_Keithley213Driver_2.KeithleyChannel', 'instrument_name': 'keithSource_ch2', 'label': 'startAdress'}, 'cycles': {'__class__': 'qcodes.instrument.parameter.Parameter', 'full_name': 'keithSource_ch2_cycles', 'value': None, 'raw_value': None, 'ts': None, 'name': 'cycles', 'inter_delay': 0, 'post_delay': 0, 'unit': '', 'instrument': 'Qcodes_Keithley213Driver_2.KeithleyChannel', 'instrument_name': 'keithSource_ch2', 'label': 'cycles'}, 'voltage': {'__class__': 'Qcodes_Keithley213Driver_2.voltage', 'full_name': 'keithSource_ch2_voltage', 'value': 0, 'raw_value': 0, 'ts': '2022-09-14 13:01:52', 'name': 'voltage', 'inter_delay': 0, 'post_delay': 0, 'unit': 'V', 'instrument': 'Qcodes_Keithley213Driver_2.KeithleyChannel', 'instrument_name': 'keithSource_ch2', 'label': 'voltage'}}, '__class__': 'Qcodes_Keithley213Driver_2.KeithleyChannel', 'name': 'keithSource_ch2'}, 'ch3': {'functions': {}, 'submodules': {}, 'parameters': {'waveform': {'__class__': 'qcodes.instrument.parameter.Parameter', 'full_name': 'keithSource_ch3_waveform', 'value': 0, 'raw_value': 0, 'ts': '2022-09-14 13:00:20', 'name': 'waveform', 'inter_delay': 0, 'post_delay': 0, 'unit': 'V', 'instrument': 'Qcodes_Keithley213Driver_2.KeithleyChannel', 'instrument_name': 'keithSource_ch3', 'label': 'waveform'}, 'waveformStepTime': {'__class__': 'qcodes.instrument.parameter.Parameter', 'full_name': 'keithSource_ch3_waveformStepTime', 'value': 0, 'raw_value': 0, 'ts': '2022-09-14 13:00:20', 'name': 'waveformStepTime', 'inter_delay': 0, 'post_delay': 0, 'unit': 'ms', 'instrument': 'Qcodes_Keithley213Driver_2.KeithleyChannel', 'instrument_name': 'keithSource_ch3', 'label': 'waveformStepTime'}, 'startAdress': {'__class__': 'qcodes.instrument.parameter.Parameter', 'full_name': 'keithSource_ch3_startAdress', 'value': 0, 'raw_value': 0, 'ts': '2022-09-14 13:00:20', 'name': 'startAdress', 'inter_delay': 0, 'post_delay': 0, 'unit': '', 'instrument': 'Qcodes_Keithley213Driver_2.KeithleyChannel', 'instrument_name': 'keithSource_ch3', 'label': 'startAdress'}, 'cycles': {'__class__': 'qcodes.instrument.parameter.Parameter', 'full_name': 'keithSource_ch3_cycles', 'value': None, 'raw_valu [... cut ...] ', 'name': 'timeout', 'inter_delay': 0, 'post_delay': 0, 'unit': 's', 'instrument': 'Qcodes_Keithley213Driver_2.Keithley213Driver', 'instrument_name': 'keithSource', 'label': 'timeout', 'vals': '<MultiType: Numbers v>=0, Enum: {None}>'}}, '__class__': 'Qcodes_Keithley213Driver_2.Keithley213Driver', 'name': 'keithSource', 'address': 'GPIB0::9::INSTR', 'terminator': None, 'read_terminator': None, 'write_terminator': '\r\n', 'timeout': 10.0}}, 'parameters': {}, 'components': {}, 'config': None}
```

user123369, 2022-09-14:
@jenshnielsen same problem 'maximum recursion depth exceeded in comparison'.

jenshnielsen, 2022-09-14:
Could you test if 

```
import json
from qcodes.utils.helpers import NumpyJSONEncoder

keithSource = kd.Keithley213Driver('keithSource','GPIB0::9::INSTR')
flukeMeter = fd.Fluke8842aDriver('fDMMInst1','GPIB0::10::INSTR')

station =qcodes.Station(flukeMeter)
snapshot = station.snapshot(update=True)
json.dumps({"station": snapshot}, cls=NumpyJSONEncoder)
```
or 

```
import json
from qcodes.utils.helpers import NumpyJSONEncoder

flukeMeter = fd.Fluke8842aDriver('fDMMInst1','GPIB0::10::INSTR')

snapshot = flukeMeter.snapshot(update=True)
json.dumps({"station": snapshot}, cls=NumpyJSONEncoder)

```

also has the issue

jenshnielsen, 2022-09-14:
Could you try json encoding the snapshot of each of the parameters
Something like
```python
for param_name, param in flukemeter.parameters.items()
    print(f"encoding {param_name}")
    json.dumps(param.shapshot(update=True))
```

user123369, 2022-09-14:
@jenshnielsen here is the code:
```
flukeMeter = fd.Fluke8842aDriver('flukeMeter','GPIB0::10::INSTR')

for param_name, param in flukeMeter.parameters.items():
    print(f"encoding {param_name}")
    json.dumps(param.snapshot(update=True))
```
and the output:

```
encoding IDN
encoding timeout
encoding voltageDC
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
Input In [23], in <cell line: 3>()
      3 for param_name, param in flukeMeter.parameters.items():
      4     print(f"encoding {param_name}")
----> 5     json.dumps(param.snapshot(update=True))

File ~\.conda\envs\drivers\lib\json\__init__.py:231, in dumps(obj, skipkeys, ensure_ascii, check_circular, allow_nan, cls, indent, separators, default, sort_keys, **kw)
    226 # cached encoder
    227 if (not skipkeys and ensure_ascii and
    228     check_circular and allow_nan and
    229     cls is None and indent is None and separators is None and
    230     default is None and not sort_keys and not kw):
--> 231     return _default_encoder.encode(obj)
    232 if cls is None:
    233     cls = JSONEncoder

File ~\.conda\envs\drivers\lib\json\encoder.py:199, in JSONEncoder.encode(self, o)
    195         return encode_basestring(o)
    196 # This doesn't pass the iterator directly to ''.join() because the
    197 # exceptions aren't as detailed.  The list call should be roughly
    198 # equivalent to the PySequence_Fast that ''.join() would do.
--> 199 chunks = self.iterencode(o, _one_shot=True)
    200 if not isinstance(chunks, (list, tuple)):
    201     chunks = list(chunks)

File ~\.conda\envs\drivers\lib\json\encoder.py:257, in JSONEncoder.iterencode(self, o, _one_shot)
    252 else:
    253     _iterencode = _make_iterencode(
    254         markers, self.default, _encoder, self.indent, floatstr,
    255         self.key_separator, self.item_separator, self.sort_keys,
    256         self.skipkeys, _one_shot)
--> 257 return _iterencode(o, 0)

File ~\.conda\envs\drivers\lib\json\encoder.py:179, in JSONEncoder.default(self, o)
    160 def default(self, o):
    161     """Implement this method in a subclass such that it returns
    162     a serializable object for ``o``, or calls the base implementation
    163     (to raise a ``TypeError``).
   (...)
    177 
    178     """
--> 179     raise TypeError(f'Object of type {o.__class__.__name__} '
    180                     f'is not JSON serializable')

TypeError: Object of type bytes is not JSON serializable
```

jenshnielsen, 2022-09-14:
Sorry I think there is a typo in my code. Should have been (with the NumpyJSONEncoder)

```python
for param_name, param in flukemeter.parameters.items()
    print(f"encoding {param_name}")
    json.dumps(param.shapshot(update=True), cls=NumpyJSONEncoder)
```

user123369, 2022-09-14:
@jenshnielsen 
ok this time the response is the usual:
```
encoding IDN
encoding timeout
encoding voltageDC
---------------------------------------------------------------------------
RecursionError                            Traceback (most recent call last)
Input In [10], in <cell line: 3>()
      3 for param_name, param in flukeMeter.parameters.items():
      4     print(f"encoding {param_name}")
----> 5     json.dumps(param.snapshot(update=True), cls=NumpyJSONEncoder)

File ~\.conda\envs\drivers\lib\json\__init__.py:238, in dumps(obj, skipkeys, ensure_ascii, check_circular, allow_nan, cls, indent, separators, default, sort_keys, **kw)
    232 if cls is None:
    233     cls = JSONEncoder
    234 return cls(
    235     skipkeys=skipkeys, ensure_ascii=ensure_ascii,
    236     check_circular=check_circular, allow_nan=allow_nan, indent=indent,
    237     separators=separators, default=default, sort_keys=sort_keys,
--> 238     **kw).encode(obj)

File ~\.conda\envs\drivers\lib\json\encoder.py:199, in JSONEncoder.encode(self, o)
    195         return encode_basestring(o)
    196 # This doesn't pass the iterator directly to ''.join() because the
    197 # exceptions aren't as detailed.  The list call should be roughly
    198 # equivalent to the PySequence_Fast that ''.join() would do.
--> 199 chunks = self.iterencode(o, _one_shot=True)
    200 if not isinstance(chunks, (list, tuple)):
    201     chunks = list(chunks)

File ~\.conda\envs\drivers\lib\json\encoder.py:257, in JSONEncoder.iterencode(self, o, _one_shot)
    252 else:
    253     _iterencode = _make_iterencode(
    254         markers, self.default, _encoder, self.indent, floatstr,
    255         self.key_separator, self.item_separator, self.sort_keys,
    256         self.skipkeys, _one_shot)
--> 257 return _iterencode(o, 0)

File ~\.conda\envs\drivers\lib\site-packages\qcodes\utils\helpers.py:86, in NumpyJSONEncoder.default(self, obj)
     59 """
     60 List of conversions that this encoder performs:
     61 
   (...)
     81   string representation (using the ``str`` function).
     82 """
     83 with warnings.catch_warnings():
     84     # this context manager can be removed when uncertainties
     85     # no longer triggers deprecation warnings
---> 86     warnings.simplefilter("ignore", category=DeprecationWarning)
     87     import uncertainties
     89 if isinstance(obj, np.generic) \
     90         and not isinstance(obj, np.complexfloating):
     91     # for numpy scalars

File ~\.conda\envs\drivers\lib\warnings.py:179, in simplefilter(action, category, lineno, append)
    175 assert action in ("error", "ignore", "always", "default", "module",
    176                   "once"), "invalid action: %r" % (action,)
    177 assert isinstance(lineno, int) and lineno >= 0, \
    178        "lineno must be an int >= 0"
--> 179 _add_filter(action, None, category, None, lineno, append=append)

File ~\.conda\envs\drivers\lib\warnings.py:186, in _add_filter(append, *item)
    184 if not append:
    185     try:
--> 186         filters.remove(item)
    187     except ValueError:
    188         pass

RecursionError: maximum recursion depth exceeded in comparison

```

jenshnielsen, 2022-09-14:
Ok that tells us the problem is with the snapshot of `voltageDC` it self and not some interaction between parameters

jenshnielsen, 2022-09-15:
I can reproduce this now

```
import json
from qcodes.utils.helpers import NumpyJSONEncoder

a = {        "voltageDC": {
            "__class__": "qcodes_fluke8842aDriver_2.voltageDC",
            "full_name": "fDMMInst1_voltageDC",
            "value": 5.7e-05,
            "ts": "2022-09-12 10:18:34",
            "raw_value": b"+000.057E-3\r\n",
            "name": "voltageDC",
            "unit": "V",
            "inter_delay": 0,
            "post_delay": 0,
            "instrument": "qcodes_fluke8842aDriver_2.Fluke8842aDriver",
            "instrument_name": "fDMMInst1",
            "label": "voltageDC",
        }}
json.dumps(a,  cls=NumpyJSONEncoder)
```
Triggers the issue.

The reason is the bytestring in the raw_value. The reason why you are seeing a bytestring here (note the b in from of the string)
is that you use visa_handle.read_raw. As @astafan8 suggested above using `instrument.ask` is a lot better or at the very least `instrument.visa_handle.read` and or `instrument.visa_handle.query` `read_raw` should really only be used when you handle the decoding or the raw value in your get_raw your self for strange dataformats that cannot be automatically decoded.

That being said qcodes should handle bytestrings in snapshots better.

jenshnielsen, 2022-09-15:
Tracking improvments to the JSONEncoder in https://github.com/QCoDeS/Qcodes/issues/4620

jenshnielsen, 2022-09-16:
@user123369 Would it be ok to close this now. QCoDeS now has a fix to not error on bytestrings and there is a clear path to fix your driver to not return bytestrings

user123369, 2022-09-16:
@jenshnielsen sure, thanks for al the help!

closed 2022-09-16 by user123369