microsoft/Qcodes #4637: Non snapshottable parameter in AMI430 driver (without heater)

jenshnielsen, 2022-09-20:
Snapshotting the AMI430 driver without a heater will show you (with debug logging)

```
2022-09-20 14:33:40,772 ¦ qcodes.instrument.base ¦ WARNING ¦ base ¦ snapshot_base ¦ 272 ¦ [magnet_x_SwitchHeater(AMI430SwitchHeater)] Snapshot: Could not update parameter: state
2022-09-20 14:33:40,773 ¦ qcodes.instrument.base ¦ INFO ¦ base ¦ snapshot_base ¦ 273 ¦ [magnet_x_SwitchHeater(AMI430SwitchHeater)] Details for Snapshot:
Traceback (most recent call last):
  File "c:\Users\qt3usr\Miniconda3\envs\qcodes\lib\site-packages\qcodes\instrument\base.py", line 268, in snapshot_base
    snap['parameters'][name] = param.snapshot(update=update_par)
  File "c:\Users\qt3usr\Miniconda3\envs\qcodes\lib\site-packages\qcodes\utils\metadata.py", line 64, in snapshot
    snap = self.snapshot_base(update=update)
  File "c:\Users\qt3usr\Miniconda3\envs\qcodes\lib\site-packages\qcodes\instrument\parameter.py", line 509, in snapshot_base
    state['value'] = self.get()
  File "c:\Users\qt3usr\Miniconda3\envs\qcodes\lib\site-packages\qcodes\instrument\parameter.py", line 661, in get_wrapper
    raise e
  File "c:\Users\qt3usr\Miniconda3\envs\qcodes\lib\site-packages\qcodes\instrument\parameter.py", line 648, in get_wrapper
    raw_value = get_function(*args, **kwargs)
  File "c:\Users\qt3usr\Miniconda3\envs\qcodes\lib\site-packages\qcodes\utils\command.py", line 203, in __call__
    return self.exec_function(*args)  # type: ignore[operator]
  File "c:\Users\qt3usr\Miniconda3\envs\qcodes\lib\site-packages\qcodes\instrument_drivers\american_magnetics\AMI430_visa.py", line 53, in check_enabled_decorator
    raise AMI430Exception("Switch not enabled")
qcodes.instrument_drivers.american_magnetics.AMI430_visa.AMI430Exception: ('Switch not enabled', 'getting magnet_x_SwitchHeater_state')
```

This should be fixed so there are no errors on snapshoting

jenshnielsen, 2022-09-21:
added a failing test in https://github.com/jenshnielsen/Qcodes/tree/driver/ami430_fix_warning but need to look at the manual to figure out what can be done to detect if the heater is installed or not

simonzihlmann, 2022-11-15:
The command `PSwitch:INSTalled?`, returning 0 or 1, should do the job I guess. See page 137 of the manual here: http://www.americanmagnetics.com/support/manuals/mn-430.pdf

Then adding an if-statement on line 260 in https://github.com/QCoDeS/Qcodes/blob/master/qcodes/instrument_drivers/american_magnetics/AMI430.py would allow to add a switch_heater submodule based on the configuration of the controller.

jenshnielsen, 2024-10-08:
This has been fixed

closed 2024-10-08 by jenshnielsen