microsoft/Qcodes #4695: dond to support time-dependence measurements (perhaps with ElapsedTimeParameter)

astafan8, 2022-10-05:
first of all, ElapsedTimeParameter does not have a set method, so it can't be used in dond as an independent parameter. Secondly, the ask is to run a measurement until elapsed time is more than a specified value (e.g. until elapsed time is more than 10 minutes).

Originally requested by @guenp @samantha-ho

guenp, 2022-10-05:
Thanks @astafan8 for creating this issue!
The reason why I initially thought it would be great for `dond` to support time as a parameter is because in the users' mind, time is just another dimension along which we want to measure a set of parameters or rasters. I'm not fixated on using `dond` for this if it is not within the package's design principles, but the convenience of being able to set up a measurement with `dond`, then decide "What if I monitor this thing over time"? and only having to make a small change will be great to be able to cover somehow.

astafan8, 2023-01-03:
Another request for this just came in: for every value of an outer parameter, measure some other parameter as a function of time by acquiring a value every 10 seconds for 5 minutes. I suggested the following solution (i'll come back with updates on how this worked):

```python
elapsed_time_param = ElapsedTimeParameter(...)

some_outer_sweep = LinSweep(
    ...,
    # reset the elapsed time parameter
    # for every new value of this sweep
    post_actions = (elapsed_time.reset,),
)

time_sweep = LinSweep(
    elapsed_time_param,
    start=0,  # value does not matter
    stop=5,  # 5 minutes, value does not matter
    num_points=5*60/10,  # 5 mins / 10 seconds, number of points matter
    delay=10,  # seconds; this is what "sweeps" the time
    get_after_set=True,  # thanks to #4759 
)

dond(some_outer_sweep, time_sweep, measure_something_param, ...)
```

guenp, 2023-09-15:
@astafan8, this issue just came up again. This is the workaround that @ThorvaldLarsen came up with:
```python
time_param = ElapsedTimeParameter('time')
time_param._settable = True
time_param.set = lambda x: None
time_sweep = LinSweep(time_param, 1, 1000, 1000, 0, get_after_set=True)

time_param.reset_clock()
ds, _, _ = dond(time_sweep, *params_to_measure)
```
This solution is not ideal because we have to pick the number of repetitions based on how long we want the measurement to run, instead of just specifying "please measure for 10 minutes".

We were wondering if it might make sense to implement something like this that behaves more like a `while` loop:
```python
time_param = ElapsedTimeParameter('time')
time_sweep = TimeSweep(time_param, start=0, stop=10*60, delay=0)
ds, _, _ = dond(time_sweep , *params_to_measure)
```

samantha-ho, 2023-09-22:
Now that they're released, it makes more sense to use the `datasaver_builder` and `dond_into` extension than try to make this work in the base dond method.

In rough pseudocode, it would look like this:
```python
time_param = ElapsedTimeParameter('time')
with datasaver_builder(dataset_definition) as datasavers:
  while time_param < start_time + duration:
    dond_into(datasavers[0], *params_to_meaure, additional_setpoints=(time_param,))
    time.sleep(delay)
  ds = datasavers[0].dataset
```