microsoft/Qcodes #5007: Parsing version string error with Keysight_34465A

MaJaISTA, 2023-02-13:
When I try to connect a Keysight_34465A digital multimeter with 

"
from qcodes.instrument_drivers.Keysight.Keysight_34465A_submodules import Keysight_34465A
dmm = Keysight_34465A('dmm', 'TCPIP0::10.21.64.130::inst0::INSTR')
"

I get "InvalidVersion: Invalid version '65.02.17-02.40-02.17-00.52-04-01'", whereas the actual firmware version is 'A.02.17-02.40-02.17-00.52-04-01':

"
---------------------------------------------------------------------------
InvalidVersion                            Traceback (most recent call last)
Cell In[1], line 2
      1 from qcodes.instrument_drivers.Keysight.Keysight_34465A_submodules import Keysight_34465A
----> 2 dmm = Keysight_34465A('dmm', 'TCPIP0::10.21.64.130::inst0::INSTR')

File ~\.conda\envs\qcodes\lib\site-packages\qcodes\instrument\instrument_meta.py:36, in InstrumentMeta.__call__(cls, *args, **kwargs)
     31 def __call__(cls, *args: Any, **kwargs: Any) -> Any:
     32     """
     33     Overloads `type.__call__` to add code that runs only if __init__ completes
     34     successfully.
     35     """
---> 36     new_inst = super().__call__(*args, **kwargs)
     37     is_abstract = new_inst._is_abstract()
     38     if is_abstract:

File ~\.conda\envs\qcodes\lib\site-packages\qcodes\instrument_drivers\Keysight\Keysight_34465A_submodules.py:12, in Keysight34465A.__init__(self, name, address, silent, **kwargs)
     10 def __init__(self, name: str, address: str, silent: bool = False,
     11              **kwargs: Any):
---> 12     super().__init__(name, address, silent, **kwargs)

File ~\.conda\envs\qcodes\lib\site-packages\qcodes\instrument_drivers\Keysight\private\Keysight_344xxA_submodules.py:470, in _Keysight_344xxA.__init__(self, name, address, silent, **kwargs)
    463 ####################################
    464 # Instrument specifications
    466 options = self._options()
    467 self.has_DIG = self.is_34465A_34470A and (
    468     "DIG" in options
    469     or version.parse(convert_legacy_version_to_supported_version("A.03"))
--> 470     <= version.parse(
    471         convert_legacy_version_to_supported_version(idn["firmware"])
    472     )
    473 )
    474 # Note that the firmware version check is still needed because
    475 # ``_options`` (the ``*OPT?`` command) returns 'DIG' option for
    476 # firmware 3.0 only if it has been purchased before
    477 self.has_MEM = self.is_34465A_34470A and 'MEM' in options

File ~\.conda\envs\qcodes\lib\site-packages\packaging\version.py:52, in parse(version)
     43 def parse(version: str) -> "Version":
     44     """Parse the given version string.
     45 
     46     >>> parse('1.0.dev1')
   (...)
     50     :raises InvalidVersion: When the version string is not a valid version.
     51     """
---> 52     return Version(version)

File ~\.conda\envs\qcodes\lib\site-packages\packaging\version.py:197, in Version.__init__(self, version)
    195 match = self._regex.search(version)
    196 if not match:
--> 197     raise InvalidVersion(f"Invalid version: '{version}'")
    199 # Store the parsed out pieces of the version
    200 self._version = _Version(
    201     epoch=int(match.group("epoch")) if match.group("epoch") else 0,
    202     release=tuple(int(i) for i in match.group("release").split(".")),
   (...)
    208     local=_parse_local_version(match.group("local")),
    209 )
"

OS Windows 11
qcodes 0.37.0
Pyvisa 1.13.0
pyserial 3.5

jenshnielsen, 2023-02-13:
@MaJaISTA Thanks for the report. This is due to the version parsing code in packaging having become more strict. 

As a work around you should be able to downgrade to packaging 21.3

I should have a fix soon

jenshnielsen, 2023-02-13:
@MaJaISTA Great. I have reopened to ensure that we get a proper fix in via #5007 that works with the newest packaging and has a test

commit 2720dce4 (2023-02-14): Merge pull request #5008 from jenshnielsen/fix_5007

Fix #5007 Handle version str with extra info
PR #5008 (2023-02-14): Fix #5007 Handle version str with extra info 
Packaging 22 dropped LegacyVersion. We mostly handled this by converting any letters in the version str to a number using ord. However, the version str for keysight also contained extra info and looks something like this "A.02.17-02.40-02.17-00.52-04-01"

From packaging 22 and upwards this is an error. We fix that by splitting on `-` and only using the first part. The converter is only used for Keysight versions so adding it there seems safe. 

This was not caught in the Keysight 34xxx driver tests since the test only runs for a simulated instrumen [... cut]
closed 2023-02-14 by jenshnielsen