microsoft/Qcodes #5089: Disabling auto_sweep in Keysight N52xx driver broken

spauka, 2023-03-28:
The `auto_sweep` function in the Keysight N52xx driver is broken. This feature allows data to be taken directly off screen without running a scan when disabled, which is necessary when measuring multiple parameters at once.

It looks like the check for auto_sweep was accidentally disabled in commit 78a989f

## Expected Behaviour
When running the below snippet the PNA should only take a single measurement
```python3
pna.auto_sweep(False)
qc.dataset.do0d(pna.magnitude, pna.phase)
```

## Actual
The pna will run the trace twice, once for magnitude and once for phase.

jenshnielsen, 2023-03-28:
Thanks  @spauka Not that it matters but I think the commit that brok it was https://github.com/jenshnielsen/Qcodes/commit/3c12a2811cc6c523b50798449f1ff60997bf5386

spauka, 2023-03-28:
Haha right - no it doesn't matter and it's an easy fix. Thanks for looking at it so quickly :)

PR #5090 (2023-03-28): Fix auto_sweep in Keysight N52xx Driver
This fixes #5089 and reenables auto_sweep.

I also fix the units for polar which should be a ratio rather than a voltage.
closed 2023-03-28 by jenshnielsen