microsoft/Qcodes #5210: module and package missing for AWG 5014C to use broadbean

gztony1227, 2023-06-15:
If you encounter a bug, use the following template.
If you have a feature request feel free to freestyle or use ideas section in the discussions
[here](https://github.com/QCoDeS/Qcodes/discussions).
If you have a question or general opinion, open your query on a relevant discussion section.

Bug report template:

###
Hi all,
The trouble is that I'm missing a mysterious package called "lomenus". It's not anywhere in Qcodes github or anywhere in internet I know of. not on pip, not on conda. It's probably an internal module used? Without it, it's not possible to use convert broadbean sequence into .awg file that 5014C uses.
Would you be able to upload the module to the github? thanks. @jenshnielsen Send you a similar email but I figure it would be more official to have it here to be resolved.

### Steps to reproduce
1.
[image]

2.
3.

### Expected behaviour
What did you expect to happen?

### Actual behaviour
What did happen instead?

### System
It would be helpful to provide such information:

**operating system**

If you are using a released version of qcodes (recommended):
**qcodes version**

If you are using the development version of qcodes:
**qcodes branch**
**qcodes commit**

jenshnielsen, 2023-06-15:
@gztony1227 Thanks for the report.  Lomentum was an old prototype for a rewrite of broadbean that was never released. The only thing it is used for is the function `make_send_and_load_awg_file_from_forged_sequence` for what you want to do I think you should be able to use `make_send_and_load_awg_file`. Could you check if that is the case?

gztony1227, 2023-06-20:
@jenshnielsen Sorry it took a bit of time for me to look into this. I followed your lead and also this example link:
[https://qcodes.github.io/Qcodes/examples/driver_examples/Qcodes%20example%20with%20Tektronix%20AWG5014C.html](url)

This funtion and comment within broadbean saved the day (not something obvious and I had to investigate enough to come to this conclusion):

```
def outputForAWGFile(self):
        """
        Returns a sliceable object with items matching the call
        signature of the 'make_*_awg_file' functions of the QCoDeS
        AWG5014 driver. One may then construct an awg file as follows
        (assuming that seq is the sequence object):

        .. code:: python

            package = seq.outputForAWGFile()
            make_awg_file(*package[:], **kwargs)

```
        """

Even with the example and helpful comment, I needed the 5014C driver, broadbean.py, and all to piece together how to use the function to make it work. It's now working that I believe I can send the broadbean-made sequence into the format (aka .awg file) 5014C wants and it shows up in its native win7 software. AWG Tek 5014C is also particular that if the sequence compose script doesn't set all the channel offsets , amplitudes, to be reasonable values, it would generate error and a relatively new user won't be able to tell whether the error is a problem of functions or missing to set offsets/amplitudes. *package[:] in an argument is also tricky to read for inexperienced python user. I learned it now and come to understand that 
*package[:] = (waveforms, m1s, m2s, nreps, trig_waits, goto_states, jump_tos, channels=[1, 3])

I definitely suggest removing anything mentioning Lomentum or include it as a function so that the function make_send_and_load_awg_file_from_forged_sequence can be useful. The example is not complete and Improving the example in the link will also be helpful. However, I also know that slowly 5014C is going out of fashion and will become more and more like a legendary instrument.
Let me know how you want to proceed?

jenshnielsen, 2023-06-20:
@Dominik-Vogel What you say. Should we simply remove the lomentum specific code.

@gztony1227 I am not actively using the 5014c at the moment so I am unlikely to make improvements to the documentation. But we are very happy to review any improvements if you contribute them

Dominik-Vogel, 2023-06-20:
Yes, lets remove it.

PR #5224 (2023-06-20): remove support for never released package lomentum
Closed #5210
closed 2023-06-20 by jenshnielsen