microsoft/Qcodes #5323: Non-gridded data export to netcdf

paul-lehmann-unibas, 2023-08-21:
### Description

When exporting non-gridded data to netcdf, the "missing" grid points are filled with NaNs. This causes such datasets to not roundtrip and can drastically increase file size.

### Steps to reproduce
```python
import numpy as np
import qcodes as qc
from qcodes.parameters import ManualParameter

num_points = 1000

indep_1 = ManualParameter("indep_1")
indep_2 = ManualParameter("indep_2")
dep_1 = ManualParameter("dep_1")
dep_2 = ManualParameter("dep_2")
meas = qc.Measurement(name="test")
meas.register_parameter(indep_1)
meas.register_parameter(indep_2)
meas.register_parameter(dep_1, setpoints=(indep_1, indep_2))
meas.register_parameter(dep_2, setpoints=(indep_1, indep_2))

with meas.run() as datasaver:
    datasaver.add_result((indep_1, np.linspace(0, 1, num_points)),
                         (indep_2, np.linspace(0, 1, num_points)),
                         (dep_1, np.linspace(0, 1, num_points)),
                         (dep_2, np.linspace(0, 1, num_points)))

ds = datasaver.dataset

ds.export("netcdf", ".")
```

### Expected behaviour
When exporting non-gridded data to the netcdf format, I expect the resulting file to accurately represent the data without introducing NaN values to fill in "missing" grid points. For the example code, I would expect the file size to be several 10 kilobytes.

### Actual behaviour
The exported netcdf file contains NaN values in place of "missing" grid points within the non-gridded data. Additionally, the exported file size is 16 megabytes (MB), which is significantly larger than expected.

### Possible Solutions:
The underlying problem seems to be the conversion from a pandas Multiindex Dataframe to an xarray Dataset. During this conversion the NaNs are added. Technically, xarray would allow creating a sparse dataset. However, xarray does not support exporting sparse datasets to netcdf, yet.

I added this hacky code to qcodes.dataset.exporters.export_to_xarray:
```python
def _load_to_xarray_dataarray_dict_no_metadata(
    dataset: DataSetProtocol,
    datadict: Mapping[str, Mapping[str,
                                   np.ndarray]]) -> dict[str, xr.DataArray]:
    import xarray as xr

    data_xrdarray_dict: dict[str, xr.DataArray] = {}

    for name, subdict in datadict.items():
        index = _generate_pandas_index(subdict)
        if index is not None and len(index.unique()) != len(index):
            for _name in subdict:
                data_xrdarray_dict[_name] = _data_to_dataframe(
                    subdict, index).reset_index().to_xarray()[_name]
        else:
            # Original code
            # xrdarray: xr.DataArray = (_data_to_dataframe(
            #     subdict,
            #     index).to_xarray().get(name, xr.DataArray()))
            # Hack to fix file size
            xrdarray: xr.DataArray = (_data_to_dataframe(
                subdict,
                index).reset_index().to_xarray().get(name, xr.DataArray()))
            data_xrdarray_dict[name] = xrdarray

    return data_xrdarray_dict
```
This solves the file size problem. But I guess it will also break roundtripping for gridded data.

### System
**operating system**
Operating system: Windows 10, MacOS 11.7.3

**qcodes version**
QCoDeS version: 0.39.1

**Other packages**
Python version: 3.10.5
xarray version: 2023.8.0
h5netcdf version: 1.0.1

Thank you in advance for any help on this matter.
Also please let me know if I can provide additional information or improve this issue in a different way.
And lastly, thanks for making qcodes! I really enjoy using it.

jenshnielsen, 2023-08-21:
@paul-lehmann-unibas Thanks for the report. I am working on a more general solution that will hopefully land soon

paul-lehmann-unibas, 2023-08-21:
@jenshnielsen thats great to hear! Thank you very much!

jenshnielsen, 2023-09-26:
closed by #5334

closed 2023-09-26 by jenshnielsen