microsoft/Qcodes #5369: Lakeshore 336 Driver with USB interface not connecting

filmartinelli, 2023-09-15:
If you encounter a bug, use the following template.
If you have a feature request feel free to freestyle or use ideas section in the discussions
[here](https://github.com/QCoDeS/Qcodes/discussions).
If you have a question or general opinion, open your query on a relevant discussion section.

Bug report template:

###
The driver for the lakeshore 336 does not connect when using USB interface

### Steps to reproduce
1. Connect the device via USB and initialize the class LakeshoreModel336

### Expected behaviour
The device is connected

### Actual behaviour
The device runs in timeout when waiting for the *IDN?\n

### System
It would be helpful to provide such information:

Windows 10

If you are using a released version of qcodes (recommended):
latest release

jenshnielsen, 2023-09-15:
Hi @filmartinelli I am not sure it the driver has been tested with usb connection but normally it should work. Could you show the exact command you use to connect.

jenshnielsen, 2023-09-15:
Also its usually a good idea to test if you can connect with the utility program supplied from the visa library vendor. Ni Max if you use the NI drivers etc.

filmartinelli, 2023-09-18:
Hi @jenshnielsen .

The line that I use to connect to the instrument is the following 
`from qcodes.instrument_drivers.Lakeshore import LakeshoreModel336 `
`tc = LakeshoreModel336("lake", "ASRL4::INSTR")`

ASRL4::INSTR is the VISA resource name that I can find on NI MAX or with PyVisa Resource manager.

I tried to connect the instrument via NI MAX and it works perfectly. However, in order to succesfully establish the connection, I need to specify proper parameters in terms of baud rate, bit number and parity. These parameters are different from the default values.

In QCodes I did not find a way to specify the parameters of the serial communication, so I guessed that the problem could come from the driver establishing a communication with default parameters which are wrong. 

Serial com parameters are as follow

- Baud Rate = 57900
- Data bits = 7
- Parity Odd
- Flow control none

Thank you!

jenshnielsen, 2023-09-18:
I am a bit surprised to see that it's a serial instrument and not a usb instrument https://pyvisa.readthedocs.io/en/latest/introduction/names.html

Perhaps its also possible to connect as a usb instrument. 

We typically have not needed to expose these kinds of settings but if it is required, we are happy to review a pr that makes it possible to set then on the Lakeshore class.

filmartinelli, 2023-09-18:
I was surprised as well, however according to the manual the USB interface just emulates a serial port and it does not mention any way to use it as a USB instrument. 

Here a screenshot of the [manual](https://www.lakeshore.com/docs/default-source/product-downloads/336_manual0ebc9b06cbbb456491c65cf1337983e4.pdf?sfvrsn=2e8633a3_1) 

[image]

To my understanding, it seems that the only possible way to establish connection via USB as Serial. 
Also, I do not think it is necessary to expose the setting of the baud rate etc., but just to make these number as default when calling this class.

frankhoff, 2023-10-09:
We had that issue a while ago. The default settings for the serial connection are set to 9600 as far as I know. A workaround would be to directly use pyvisa's ResourceManager, to open the connection to the device and then create the Lakeshore336 instance.

Here's some code (in this case the port was "COM3"):
```Python
from pyvisa import ResourceManager, constants
rm = ResourceManager()
LS366 = rm.open_resource("COM3", baud_rate=57600, parity=constants.Parity.odd, data_bits=7)
lake = Model_336("lake", "COM3")
```
Cheers

jenshnielsen, 2023-11-06:
@frankhoff I am not completely sure that would always be safe since it relies on pyvisa always reusing the resouce. 
@filmartinelli I would be happy to review a pr that sets these things correctly by default.

In the mean time I have opened https://github.com/QCoDeS/Qcodes/pull/5485 to allow these kwargs to be passed to open_resource

frankhoff, 2023-11-13:
Thanks for the reply!
If this procedure is unsafe, wouldn't it be good to catch this in some way? As user, I would expect qcodes to throw an error, if the device which should be used has already been opened.

Passing kwargs to open_resource for serial devices sounds like a good idea to me :+1:

ggggggggg, 2024-04-02:
I solved this by overloading `_open_resource` for the instrument that needs it, in my case a Lakeshore370.

``` python
   def _open_resource(self, address: str, visalib: str | None) -> tuple[MessageBasedResource, str, ResourceManager]:
        # first call the existing _open_resource method
        resource, visabackend, resource_manager = super()._open_resource(address, visalib)
        # then set the odd communication settings that can't be passed by argument with the current api
        from pyvisa import constants
        resource.parity = constants.Parity.odd
        resource.data_bits = 7
        return resource, visabackend, resource_manager
```