microsoft/Qcodes #5418: Random error in test_logger

jenshnielsen, 2023-10-04:
Observed in CI. Windows python 3.9

Looks like an instrument from a pre existing test was not cleaned up? Which is strange since the new autocontext manger should close all instruments at test exit

```
=========================== short test summary info ===========================
FAILED qcodes/tests/test_logger.py::test_instrument_connect_message - assert 'Closing VISA...instrument.\n' == "[awg_sim(AWG...re': '0.1'}\n"
  - [awg_sim(AWG5208)] Connected to instrument: {'vendor': 'QCoDeS', 'model': 'AWG5208', 'serial': '1000', 'firmware': '0.1'}
  + Closing VISA handle to lakeshore_372 as there are no non weak references to the instrument.
= 1 failed, 2632 passed, 233 skipped, 6 xfailed, 1 xpassed, 3 warnings in 950.19s (0:15:50) =
Error: Process completed with exit code 1.
```

jenshnielsen, 2023-10-04:
```
================================== FAILURES ===================================
_______________________ test_instrument_connect_message _______________________
[gw1] win32 -- Python 3.9.13 C:\hostedtoolcache\windows\Python\3.9.13\x[64](https://github.com/QCoDeS/Qcodes/actions/runs/6402218105/job/17378489286?pr=5416#step:9:65)\python.exe

    @pytest.mark.usefixtures("remove_root_handlers", "awg5208")
    def test_instrument_connect_message() -> None:
        """
        Test that the connect_message method logs as expected
    
        This test kind of belongs both here and in the tests for the instrument
        code, but it is more conveniently written here
        """
    
        with open(logger.get_log_file_name(), encoding="utf-8") as f:
            lines = f.readlines()
    
        con_mssg_log_line = lines[-1]
    
        sep = logger.logger.LOGGING_SEPARATOR
    
        con_mss = con_mssg_log_line.split(sep)[-1]
        idn = {"vendor": "QCoDeS", "model": "AWG5208",
               "serial": "1000", "firmware": "0.1"}
        expected_con_mssg = ("[awg_sim(AWG5208)] Connected to instrument: "
                             f"{idn}\n")
    
>       assert con_mss == expected_con_mssg
E       assert 'Closing VISA...instrument.\n' == "[awg_sim(AWG...re': '0.1'}\n"
E         - [awg_sim(AWG5208)] Connected to instrument: {'vendor': 'QCoDeS', 'model': 'AWG5208', 'serial': '1000', 'firmware': '0.1'}
E         + Closing VISA handle to lakeshore_3[72](https://github.com/QCoDeS/Qcodes/actions/runs/6402218105/job/17378489286?pr=5416#step:9:73) as there are no non weak references to the instrument.

qcodes\tests\test_logger.py:311: AssertionError
---------------------------- Captured stdout setup ----------------------------
Qcodes Logfile : C:\Users\runneradmin\.qcodes\logs\231004-3060-qcodes.log
Connected to: QCoDeS AWG5208 (serial:1000, firmware:0.1) in 0.06s
------------------------------ Captured log call ------------------------------
INFO     qcodes.instrument.visa:visa.py:30 Closing VISA handle to x as there are no non weak references to the instrument.
INFO     qcodes.instrument.visa:visa.py:30 Closing VISA handle to x as there are no non weak references to the instrument.
INFO     qcodes.instrument.visa:visa.py:30 Closing VISA handle to y as there are no non weak references to the instrument.
INFO     qcodes.instrument.visa:visa.py:30 Closing VISA handle to z as there are no non weak references to the instrument.
INFO     qcodes.instrument.visa:visa.py:30 Closing VISA handle to lakeshore_372 as there are no non weak references to the instrument.
INFO     qcodes.instrument.visa:visa.py:30 Closing VISA handle to lakeshore_372 as there are no non weak references to the instrument.
---------------------------- Captured log teardown ----------------------------
INFO     qcodes.instrument.instrument:instrument.py:1[86](https://github.com/QCoDeS/Qcodes/actions/runs/6402218105/job/17378489286?pr=5416#step:9:87) Closing all registered instruments
```

astafan8, 2023-10-04:
maybe gc gests collected at a not-expected moment?

maybe the test could be rewritten to make sure the connect message is among the messages instead of expecting it to be the last message? alternatively, the test could be rewritten to use instrument's log instance explicitly with a well-defined message, and then perhaps it would be more stable against changes to what is being logged in various places in qcodes

jenshnielsen, 2023-10-04:
I don't think the problem is that other messages are injected. Looking at the detailed output the message is just not there. In the attached pr I reworked it to use caplog and reworked the hander reset fixture to not modify pytest handlers

closed 2023-10-04 by jenshnielsen