microsoft/Qcodes #5484: Agilent E8257D Instrument Driver Bug in on() and off() Methods

jiin-so, 2023-11-06:
###
Agilent E8257D Instrument Driver Bug in on() and off() Methods

### Steps to reproduce
1. Import instrument driver and create instance
2. Call instrument.on() or instrument.off()

### Expected behaviour
The instrument should turn on or off.

### Actual behaviour
Script throws a "status" error. 

--> When the on() and off() methods were changed, the issue was fixed. I believe there is a syntax error when declaring the methods:

Original, throws error
```
    def on(self) -> None:
        self.set('status', 'on')

    def off(self) -> None:
        self.set('status', 'off')
```

Changed, does not throw error
```
    def on(self) -> None:
        self.status('on')

    def off(self) -> None:
        self.status('off')
```

 

### System
It would be helpful to provide such information:
**operating system** - Windows

If you are using a released version of qcodes (recommended):
**qcodes version** - 0.41.1

Linking the error txt file:
[
[Agilent E8257D Error Code.txt](https://github.com/QCoDeS/Qcodes/files/13262144/Agilent.E8257D.Error.Code.txt)
](url)

jenshnielsen, 2023-11-06:
Thanks @jiin-so as far as I can see there is no status parameter on the instrument so I am kind of surprised that any of the 2 ways you describe would work.

jiin-so, 2023-11-06:
Hello, what then would be the correct way to address the instrument to turn it on/off?

jenshnielsen, 2023-11-06:
I am not familiar with this instrument so I would suggest consulting the manual to see what the command is. Perhaps the intention is that this should switch output on off. In that case you should use output_enabled https://github.com/QCoDeS/Qcodes/blob/main/src/qcodes/instrument_drivers/agilent/Agilent_E8257D.py#L125

jenshnielsen, 2023-11-08:
@jiin-so Thanks could you have a look at https://github.com/QCoDeS/Qcodes/pull/5496 that should fix on and off to call output_enabled

closed 2023-11-09 by jenshnielsen