microsoft/Qcodes #5551: Random failure in do_nd test

jenshnielsen, 2023-11-24:
Looks like some test may be leaking a non closed awg instrument

```
=================================== FAILURES ===================================
______________________________ test_datasaver_1d _______________________________
[gw1] linux -- Python 3.10.13 /opt/hostedtoolcache/Python/3.10.13/x64/bin/python

experiment = test-experiment#test-sample#1@/tmp/pytest-of-runner/pytest-0/popen-gw1/test_datasaver_1d0/temp.db
--------------------...1-dummy_dac_ch1,dummy_dmm_v1-31
12-results-12-dummy_dac_ch1,dummy_dmm_v1-[76](https://github.com/QCoDeS/Qcodes/actions/runs/6974993548/job/18987846779?pr=5547#step:9:77)
13-results-13-dummy_dac_ch1,dummy_dmm_v1-76
DAC = <DummyInstrument: dummy_dac>, DMM = <DummyInstrument: dummy_dmm>
caplog = <_pytest.logging.LogCaptureFixture object at 0x7f1ce0b3beb0>
n_points = 76

    @given(n_points=hst.integers(min_value=1, max_value=[100](https://github.com/QCoDeS/Qcodes/actions/runs/6974993548/job/18987846779?pr=5547#step:9:101)))
    @example(n_points=5)
    @settings(deadline=None, suppress_health_check=(HealthCheck.function_scoped_fixture,))
    def test_datasaver_1d(
        experiment, DAC, DMM, caplog: LogCaptureFixture, n_points
    ) -> None:
        meas = Measurement()
        meas.register_parameter(DAC.ch1)
        meas.register_parameter(DMM.v1, setpoints=(DAC.ch1,))
    
        n_points_expected = 5
    
        meas.set_shapes({DMM.v1.full_name: (n_points_expected,)})
    
        with meas.run() as datasaver:
    
            for set_v in np.linspace(0, 1, n_points):
                DAC.ch1()
                datasaver.add_result((DAC.ch1, set_v),
                                     (DMM.v1, DMM.v1()))
    
        ds = datasaver.dataset
        caplog.clear()
        data = ds.get_parameter_data()
    
        for dataarray in data[DMM.v1.full_name].values():
            assert dataarray.shape == (n_points,)
    
        if n_points == n_points_expected:
            assert len(caplog.record_tuples) == 0
        elif n_points > n_points_expected:
>           assert len(caplog.record_tuples) == 2
E           AssertionError: assert 4 == 2
E            +  where 4 = len([('qcodes.instrument.visa', 20, 'Closing VISA handle to awg_sim as there are no non weak references to the instrument....a shape for dummy_dac_ch1 in dataset dummy_dmm_v1 from metadata when loading but found inconsistent lengths 76 and 5')])
E            +    where [('qcodes.instrument.visa', 20, 'Closing VISA handle to awg_sim as there are no non weak references to the instrument....a shape for dummy_dac_ch1 in dataset dummy_dmm_v1 from metadata when loading but found inconsistent lengths 76 and 5')] = <_pytest.logging.LogCaptureFixture object at 0x7f1ce0b3beb0>.record_tuples

tests/dataset/measurement/test_shapes.py:42: AssertionError

The above exception was the direct cause of the following exception:
```

jenshnielsen, 2023-11-30:
similar error here

```
______________________________ test_data______________________________ test_datasaver_1d _______________________________
[gw0] linux -- Python 3.12.0 /opt/hostedtoolcache/Python/3.12.0/x64/bin/python

experiment = test-experiment#test-sample#1@/tmp/pytest-of-runner/pytest-0/popen-gw0/test_datasaver_1d0/temp.db
--------------------...5-dummy_dac_ch1,dummy_dmm_v1-33
16-results-16-dummy_dac_ch1,dummy_dmm_v1-[70](https://github.com/QCoDeS/Qcodes/actions/runs/7042049661/job/19165580590?pr=5564#step:9:71)
17-results-17-dummy_dac_ch1,dummy_dmm_v1-70
DAC = <DummyInstrument: dummy_dac>, DMM = <DummyInstrument: dummy_dmm>
caplog = <_pytest.logging.LogCaptureFixture object at 0x7fe13262f830>
n_points = 70

    @given(n_points=hst.integers(min_value=1, max_value=100))
    @example(n_points=5)
    @settings(deadline=None, suppress_health_check=(HealthCheck.function_scoped_fixture,))
    def test_datasaver_1d(
        experiment, DAC, DMM, caplog: LogCaptureFixture, n_points
    ) -> None:
        meas = Measurement()
        meas.register_parameter(DAC.ch1)
        meas.register_parameter(DMM.v1, setpoints=(DAC.ch1,))
    
        n_points_expected = 5
    
        meas.set_shapes({DMM.v1.full_name: (n_points_expected,)})
    
        with meas.run() as datasaver:
    
            for set_v in np.linspace(0, 1, n_points):
                DAC.ch1()
                datasaver.add_result((DAC.ch1, set_v),
                                     (DMM.v1, DMM.v1()))
    
        ds = datasaver.dataset
        caplog.clear()
        data = ds.get_parameter_data()
    
        for dataarray in data[DMM.v1.full_name].values():
            assert dataarray.shape == (n_points,)
    
        if n_points == n_points_expected:
            assert len(caplog.record_tuples) == 0
        elif n_points > n_points_expected:
>           assert len(caplog.record_tuples) == 2
E           AssertionError: assert 4 == 2
E            +  where 4 = len([('qcodes.instrument.visa', 20, 'Closing VISA handle to n9030B_sim as there are no non weak references to the instrume...a shape for dummy_dac_ch1 in dataset dummy_dmm_v1 from metadata when loading but found inconsistent lengths 70 and 5')])
E            +    where [('qcodes.instrument.visa', 20, 'Closing VISA handle to n9030B_sim as there are no non weak references to the instrume...a shape for dummy_dac_ch1 in dataset dummy_dmm_v1 from metadata when loading but found inconsistent lengths 70 and 5')] = <_pytest.logging.LogCaptureFixture object at 0x7fe13262f830>.record_tuples

tests/dataset/measurement/test_shapes.py:42: AssertionError

The above exception was the direct cause of the following exception:

experiment = test-experiment#test-sample#1@/tmp/pytest-of-runner/pytest-0/popen-gw0/test_datasaver_1d0/temp.db
--------------------...5-dummy_dac_ch1,dummy_dmm_v1-33
16-results-16-dummy_dac_ch1,dummy_dmm_v1-70
17-results-17-dummy_dac_ch1,dummy_dmm_v1-70
DAC = <DummyInstrument: dummy_dac>, DMM = <DummyInstrument: dummy_dmm>
caplog = <_pytest.logging.LogCaptureFixture object at 0x7fe13262f830>

    @given(n_points=hst.integers(min_value=1, max_value=100))
>   @example(n_points=5)

tests/dataset/measurement/test_shapes.py:12: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = <hypothesis.core.StateForActualGivenExecution object at 0x7fe10b2ef260>
data = ConjectureData(VALID, 1 bytes, frozen)

[... 1390 lines cut ...]
E                                                (DMM.v1, DMM.v1()))
E               
E                   ds = datasaver.dataset
E                   caplog.clear()
E                   data = ds.get_parameter_data()
E               
E                   for dataarray in data[DMM.v1.full_name].values():
E                       assert dataarray.shape == (n_points,)
E               
E                   if n_points == n_points_expected:
E                       assert len(caplog.record_tuples) == 0
E                   elif n_points > n_points_expected:
E           >           assert len(caplog.record_tuples) == 2
E           E           AssertionError: assert 4 == 2
E           E            +  where 4 = len([('qcodes.instrument.visa', 20, 'Closing VISA handle to n9030B_sim as there are no non weak references to the instrume...a shape for dummy_dac_ch1 in dataset dummy_dmm_v1 from metadata when loading but found inconsistent lengths 70 and 5')])
E           E            +    where [('qcodes.instrument.visa', 20, 'Closing VISA handle to n9030B_sim as there are no non weak references to the instrume...a shape for dummy_dac_ch1 in dataset dummy_dmm_v1 from metadata when loading but found inconsistent lengths 70 and 5')] = <_pytest.logging.LogCaptureFixture object at 0x7fe13262f830>.record_tuples
E           
E           tests/dataset/measurement/test_shapes.py:42: AssertionError
E           
E           
E           You can reproduce this example by temporarily adding @reproduce_failure('6.91.0', b'AEU=') as a decorator on your test case

/opt/hostedtoolcache/Python/3.12.0/x64/lib/python3.12/site-packages/hypothesis/core.py:[89](https://github.com/QCoDeS/Qcodes/actions/runs/7042049661/job/19165580590?pr=5564#step:9:90)2: Flaky
----------------------------- Captured stdout call -----------------------------
```

a-n-n-a-l-e-e, 2024-02-14:
we've seen similar failures  in the unit tests
[full log](https://cache.nixos.org/log/xq0cp80zffciidvk2n5b3ya2ndbbyzlz-python3.11-qcodes-0.42.1.drv)
```console
=========================== short test summary info ============================
FAILED tests/drivers/test_keithley_s46.py::test_query_close_once_at_init - assert 2 == 1
FAILED tests/parameter/test_parameter_ramp.py::test_step_ramp - assert 4 == 1
```
[full log](https://logs.ofborg.org/?attempt_id=237e6cca-42be-4a1a-98f1-21529787f4eb&key=nixos%2Fnixpkgs.288404)
```console
=========================== short test summary info ============================
FAILED tests/parameter/test_parameter_ramp.py::test_step_ramp - assert 2 == 1
```

PR #5914 (2024-04-02): Fix #5551
Avoid logging info about closing instrument when the instrument is already closed
This causes random log messages at gc time even if the instrument had already been closed
Fixes #5551
commit 04a68acd (2024-04-02): Fix #5551

Avoid logging info about closing instrument when the instrument is already closed

This causes random log messages at gc time even if the instrument had already been closed

Fixes #5551
commit 1abae7da (2024-04-02): Merge pull request #5914 from jenshnielsen/fix_flaky_log_test

Fix #5551
closed 2024-04-02 by jenshnielsen