microsoft/Qcodes #5561: Cannot exit ipython with running qcodes.Monitor()

michaelc12qe, 2023-11-28:
### Steps to reproduce
iPython version 8.10, qcodes version 0.42, Windows 10 Pro 22H2

1. open ipython
2.from qcodes import Monitor
monitor = Monitor()
3. exit ipython

It hangs instead of closing iPython. The terminal must be terminated.

jenshnielsen, 2023-11-28:
I can reproduce this. Not sure what is going on but perhaps related to https://github.com/QCoDeS/Qcodes/issues/3814 

It would be nice to investigate if changing the monitor thread to be a deamon thread would also resolve this issue

jenshnielsen, 2023-11-29:
Note that until this is resolved it can be mitigated by explicitly calling join on the monitor thread

```
❯ ipython
Python 3.11.5 | packaged by Anaconda, Inc. | (main, Sep 11 2023, 13:26:23) [MSC v.1916 64 bit (AMD64)]
Type 'copyright', 'credits' or 'license' for more information
IPython 8.18.1 -- An enhanced Interactive Python. Type '?' for help.

In [1]: from qcodes import Monitor
In [2]: m = Monitor()
In [3]: m.join()
In [4]: exit

~ took 29s
❯
```

michaelc12qe, 2023-11-29:
I've tested turning the monitor thread into a daemon thread as you suggested, by passing the keyword argument `daemon=True` into the `Monitor` initialization. It appears to resolve this issue. I've also checked that the behaviour of the monitor remains unchanged by add QDAC2 voltage channels to the monitor, and it seems fine.

I don't know if there's any more testing required, but happy to raise a PR for this if you like.

jenshnielsen, 2023-11-29:
@michaelc12qe In the long term I would like to come up with something slightly cleaner than deamon mode but I agree that in the short term using deamon is the best solution.

PR #5565 (2023-11-30): Turn Monitor thread into daemon
Resolves #5561

This PR changes the `qcodes.Monitor` Thread into a daemon thread. When the Thread was not daemonized, it prevented both python and ipython from closing unless the thread was already joined.
closed 2023-11-30 by jenshnielsen