microsoft/Qcodes #5634: Add redirects to new documentation?

rsokolewicz, 2024-01-09:
Not sure when the move was made, but I see now a lot of 404s when referencing QCoDeS pages. 

e.g. https://qcodes.github.io/Qcodes/api/instrument/index.html gives a 404

[image]

and it looks like everything is moved to microsoft.github.io/...

Could we maybe add redirects from qcodes.github.io/Qcodes to microsoft.github.io/Qcodes?

jenshnielsen, 2024-01-11:
Unfortunately a straightforward redirect based on CNAME etc is not possible with github pages however each html page should now be redirected using a Jekyll layout. The code doing the redirect lives in 

https://github.com/QCoDeS/qcodes.github.io/blob/master/_layouts/redirect.html
https://github.com/QCoDeS/qcodes.github.io/blob/master/_config.yml 
and 
a copy of the qcodes source where each page html page is replaced by a blank page containing only a reference to the redirect layout. This will unfortunately not handle intersphinx objects.inv files but I think this is the bese we can do given the limitations of ghpages 

For reference the blank pages were generated with 

```
from pathlib import Path

p = Path("original")
myglob = p.glob("**/*.html")

for file in myglob:
    print(file)

    new_parts = ("blank_docs",) + file.parts[1:]

    new_file = Path("blank_docs", *file.parts[1:])
    print(new_file)
    new_file.parent.mkdir(exist_ok=True, parents=True)
    with new_file.open("w") as myf:
        myf.write(
            """---
layout: redirect
---
"""
```

from a copy of the existing qcodes docs

closed 2024-01-11 by jenshnielsen

rsokolewicz, 2024-01-11:
amazing. thanks! :)