microsoft/Qcodes #5645: Error when running `Infiniium.py` driver file

paniash, 2024-01-15:
Bug report template:

### `type` object error when running driver file `Infiniium.py`

### Steps to reproduce
1. Open `Infiniium.py`
2. Run the file in python i.e. `python Infiniium.py`
3. Error pops up as: `'type' object is not subscriptable`

### Expected behaviour
Running the file should yield no output.

### Actual behaviour
```
Traceback (most recent call last):
  File "Infiniium.py", line 82, in <module>
    class DSOTraceParam(ParameterWithSetpoints):
  File "Infiniium.py", line 87, in DSOTraceParam
    UNIT_MAP: ClassVar[dict[int, str]] = {
TypeError: 'type' object is not subscriptable
```

### System

**operating system** Linux

If you are using a released version of qcodes (recommended):
**qcodes version** 0.43.0

jenshnielsen, 2024-01-15:
@paniash Which version of python are you using. QCoDeS 0.43 (and newer only supports 3.9+) this looks like you are using an older version of python

paniash, 2024-01-15:
You're right. Running with a later version of python fixes the issue! Thanks.

closed 2024-01-15 by paniash

paniash, 2024-01-16:
@jenshnielsen Just opened #5646. :)